Smoothieware/Smoothieware #575: Printing from USB drive?

whosawhatsis, 2015-01-09:
Assuming a female USB A port (like this one: http://www.datapro.net/products/usb-panel-mount-to-5-pin-socket.html) was added, how difficult would it be to make Smoothie capable of printing from a USB flash drive instead of the SD card? Even with Smoothie's ability to mount the SD card over USB, I still find that the transfers are slower than I'd like, and moving around a smoothieboard's SD card every print isn't a great idea since it stores the configuration. What I'd like to do is leave the firmware stuff on the SD, but give users the ability to put gcode on a USB drive to plug into the machine, like some of the commercial printers have started to do. If it can be done fast enough, it would be nice to copy the gcode from the USB drive to the printer's internal storage (SD card) before starting a print.

arthurwolf, 2015-01-09:
It is possible, with some coding, and you would loose all other USB
functionality ( essentially you have to choose between using USB as we do
now, or as you want to ).
I'm not sure how much speed would be gained.

Note : if you don't care much about speed, but care much about not having
to remove the SD card, note Smoothie supports using two SD cards, one for
config and one for your G-code files, and you can remove that one.

wolfmanjm, 2015-01-09:
Also you can currently mount smoothie as a thumb drive on any host pc and copying files that way is almost as fast as copying them to a thumb drive and sneaker net it to the smoothie.
If USB host mode were supported as suggested it would need a special cable to provide power to the thumb drive.

wolfmanjm, 2015-01-09:
enabling the external sd card is probably the way to go for this, it is already supported and solves the same problem. Printing from sdcard is not a speed issue anyway, we can read gcode off a sdcard faster than we consume it for printing.

whosawhatsis, 2015-01-09:
In my experience, copying to smoothie using the mounted USB drive, while faster than using the gcode method with Marlin, is still at least an order of magnitude slower than copying to an SD card mounted in the computer's own SD card reader.

I didn't know about the option for a second SD card. What hardware needs to be added for that? I know a lot of people would still prefer to use a USB drive instead, but if that removes connecting to a PC over USB as an option, it's less desirable. Still, I'm sure at some point someone will want to release a "just click print" machine with that capability. I think the 3DS Cube printers, for instance, have a USB port for a flash drive, and none for connecting to a PC. They likely run on a similar chip.

wolfmanjm, 2015-01-09:
https://www.sparkfun.com/products/544 will work, or any LCD panel that has an sdcard slot (like viki2 or rrd glcd).
Setup is here...  http://smoothieware.org/panel#toc13

wolfmanjm, 2015-01-09:
You can't have host mode and device mode on the same USB port at the same time, so it is either a thumb drive OR connecting to a host, not both. I'll definitely look into adding the ability to be in host mode, but it will probably need to be a separate binary, and the diodes on the 5v line need to be removed so 5v power can be applied to the thumb drive (or a y cable). Look at any android phone, you need the special cable to be in USB OTG mode to connect a thumb drive.

closed 2015-01-09 by wolfmanjm

whosawhatsis, 2015-01-09:
If it could be added as a configuration or compile-time option without too much trouble, that would be great. Thanks.

triffid, 2015-01-10:
It would actually require someone to write an entire host-side USB stack
which, I assure you, is substantially more complex than the client side
stack we have at the moment.

I await your pull request ;)

whosawhatsis, 2015-01-10:
I assumed since the hardware supports it, there would be a library available somewhere. If not, it's probably not worth doing.

triffid, 2015-01-10:
It looks like mbed has a library for it, but I think it might take quite a
bit of work to shoehorn into smoothie.

Technically smoothie uses mbed libraries, but we branched off gcc4mbed ages
ago before mbed officially open sourced their stuff so they're quite
divergent by now

braypp, 2016-05-04:
Old topic but maybe this link will help...
http://siwawi.bauing.uni-kl.de/avr_projects/arm_projects/arm_memcards/index.html#chanfat_lpc_cm3