Smoothieware/Smoothieware #576: ABC are not the same az XYZ after abort

avrmp, 2015-01-11:
In the Cartesian coordinate robot if we trigger an abort, after issuing M999 the A, B and C coordinates are not the same as X, Y, and Z. 
So what I did was I added this to M999 in GcodeDispatch.cpp :

THEKERNEL->robot->reset_position_from_current_actuator_position();
 
and I also had to include robot.h. 
and so far it solved the problem :D but wait! That's only if I'm not messing something else up by doing this... So is this okay? Or should I be ready if it smoke?!  :D

wolfmanjm, 2015-01-11:
sorry abort does correct it, kill does not and cannot.

closed 2015-01-11 by wolfmanjm

avrmp, 2015-01-11:
so even though ABC are not the same as XYZ after kill, the ABC are still the wrong coordinates?! I kind of did a little experiment and ABC were dead on all the time?! Then I'm just dreaming :(
I have been trying to add a joystick for jogging and the way I do it now is I have an external processor connected to the joystick and also the smoothie's serial port and when ever some one trys to jog the processor send smoothie X10 commands with delays and when they let the joystick go i trigger the kill button and then I issue and M999 and I added that code for M999 but what I'm doing is not smart :( I have to figure out another way :( any suggestions?

wolfmanjm, 2015-01-11:
Yes ABC is unlikely to be correct after a kill (like hitting the endstops). If it is correct it is just luck.
What I suggest is a new Mxxx command that sets XYZ to ABC after FK (ie call THEKERNEL->robot->reset_position_from_current_actuator_position();) that way you can reset it if you prefer to not home and feel lucky :)
If you look closely at the abort code you will see I jump through hoops to get it to stop in a way that will most likely keep ABC mostly correct, it is still not perfect but works around 90% of the time. I can do that because abort does not need to stop immediately. however KILL does need to stop immediately so the same procedure will not work.
Another reason that KILL is not recoverable without homing is it turns off the power to the motors, this will lose microstepping and hence be inaccurate. it may by out 0.1mm to 1mm depending on your steps/mm but even 0.1mm is bad,.

wolfmanjm, 2015-01-11:
Once you queue commands you cannot really unqueue them reliably, you need to do a controlled decelerate, this is very difficult, @triffid did some work in that direction and I copied some of it, and used it in the abort command. 

I think your approach of using KILL to stop the joystick is a really bad one :) KILL and M999 are commands of last resort and should never be used for control. (FYI KILL does an ON_HALT event).

you need to be able to control the G codes you are generating and queuing from the joystick, that maybe hard to do if you feed them via the serial port. Off hand I can;t think of a good way to do that. You would have more control if you wrote a module to do the control.

avrmp, 2015-01-11:
Oh boy :( That's not good! I'm lucky you told me that, or I'm lucky I have smoothie lol
So I will try the abort then :), just is case: is there any way to trigger abort with a pin through the switch module?

wolfmanjm, 2015-01-11:
abort only works when you are printing from sdcard, so you can't actually use that. I don't think there is a way to do what you want. 

I think you are on the wrong track, you should not be sending commands that you do not want to execute. Once you send a command you are committed to executing that command, you can;t then decode you want to not execute that command. What you need  to do is figure out how to only send the commands you want to actually execute.

avrmp, 2015-01-11:
Well that is exactly my problem, I don't know When I want to  let the joystick go, because every time it's going to be different
How about this:
Have a Gcode file in sd card for different jogging direction (lets say I will put something like "G91 G0 X10000000 G90" and when ever I want to jog I play that file and when even I let the joy stick go I abort it?!

avrmp, 2015-01-11:
How about writing a Joystick module?!