Smoothieware/Smoothieware #579: Laser ignores some gcode cuts

qharley, 2015-01-19:
Find attached the g code simulation, and actual cut images sent to the machine.  
Do you experience the same?

[image]
[image]

Gcode as follows: 
G21
G92 X0 Y0
G0 F20000
G1 S0.8
G1 F800
G00 Y48
M3
G4 P200
G01 X223
G02 X225 Y46 I0 J-2
G01 Y4
G02 X223 Y2 I-2 J0
G01 X175
Y0
X270
Y50
X0
Y48
M5
G00 Y60
M3
G4 P200
G01 X270
Y110
X177
G03 X175 Y108 I0 J-2
G01 Y64
G02 X173 Y62 I-2 J0
G01 X0
Y60
M5
G00 X14.291 Y132.609
M3
G4 P200
G03 X11.109 Y135.791 I-1.591 J1.591
G03 X14.291 Y132.609 I1.591 J-1.591
M5
G00 Y150.609
M3
G4 P200
G03 X11.109 Y153.791 I-1.591 J1.591
G03 X14.291 Y150.609 I1.591 J-1.591
M5
G00 X181.891
M3
G4 P200
G03 X178.709 Y153.791 I-1.591 J1.591
G03 X181.891 Y150.609 I1.591 J-1.591
M5
G00 Y132.609
M3
G4 P200
G03 X178.709 Y135.791 I-1.591 J1.591
G03 X181.891 Y132.609 I1.591 J-1.591
M5
G00 X212.46 Y150.079
M3
G4 P200
G03 X208.218 Y154.321 I-2.121 J2.121
G03 X212.46 Y150.079 I2.121 J-2.121
M5
G00 X0 Y120
M3
G4 P200
G01 X270
Y170
X178
G03 X175 Y167 I0 J-3
G02 X172 Y164 I-3 J0
G01 X2
G03 X0 Y162 I0 J-2
G01 Y120
M5
M2

wolfmanjm, 2015-01-19:
there is a known G3 bug  possible duplicate of #500

qharley, 2015-01-19:
I see...

Where is G3 implemented?

qharley, 2015-01-19:
Additionally some of the cuts ignored are not G3 cuts.  Short lengths straight lines.
The holes are cut using G3, and they came out fine.

Bouni, 2015-01-19:
Just simulated the G-code here: http://microtechstelladata.com/CncPlot.aspx
I think i know the problem, at least partially :)

You have lines without a leading G01/G00! That lines are ignored.

Bouni, 2015-01-19:
And the at the **G01 X270** line it seems that the laser was off at the beginning and then switched on in the middle of the move!?

wolfmanjm, 2015-01-19:
oh yea that is not gcode that smoothie can use, you need Go or G1 on each line OR a space at the start of the line to make it use the last G0 or G1 specified. Basically you are feeding it illegal gcode.

closed 2015-01-19 by wolfmanjm

qharley, 2015-01-19:
Thanks, Yes this makes sense...

I will check with my CAM scripts

nophead, 2015-01-19:
@wolfmanjm,
Are you sure not having a space is illegal G code? The example program here: http://en.wikipedia.org/wiki/G-code doesn't have spaces before X commands on their own line and in general white space is ignored.

qharley, 2015-01-19:
I managed to disable the default behaviour on my CAM, so I am not affected any more... 
I however agree with Chris - Anyone using smoothie for CNC work will have to disable the default setup first.  Not ideal.

nophead, 2015-01-19:
I just checked LinuxCNC and that does accept X commands without leading spaces.

arthurwolf, 2015-01-19:
Yeah, this is illegal G-code, -to smoothie-

We support G-code from Slic3r and most reprap G-code generators, Cambam,
Pycam, and a few others.
But there are so many flavours of G-code out there ...