Smoothieware/Smoothieware #584: Stricter limits for TemperatureControl MINTEMP protection

PetteriAimonen, 2015-01-23:
Sometimes thermistors break, which could cause the NTC to read very low temperature and cause TemperatureControl to overpower the heater. Currently Smoothie detects this condition only when ADC value after median filtering is 0 or 4095.

However, there are a few cases when this protection is not enough:
1) Thermistors do not always break completely, which could cause very wrong temperatures but not fully saturated.
2) Noise can cause the ADC value to be not fully saturated after median filtering. This happens especially when using oversampling.
3) When messing up with PID values (yeah, my own fault..) I have had it overshoot to 400 C. Some protection against user errors would be a nice side-effect.

I would suggest to either:
1) Have hardcoded limits of e.g. -20 C to +350 C.
2) Have configurable min & max limits. This would require 8 bytes storage per TemperatureControl channel.
3) Have just configurable min limit. This would require 4 bytes per TemperatureControl channel.

I would also suggest to do a full ON_HALT / M999 stop (like limit switches) instead of just turning of the heater. This avoids trying to push filament to a malfunctioning extruder.

Would a patch implementing one of these options be considered for merge? Which one would be preferred?

qharley, 2015-01-23:
I have a min limit that I can pull request...

wolfmanjm, 2015-12-13:
Most of this has been done

closed 2015-12-13 by wolfmanjm