Smoothieware/Smoothieware #588: Allow to use limit switches for deltas

minad, 2015-02-02:
At first thx for the great work. I am a novice 3d reprap printer user and with smoothieboard everything went very smooth right from the start :)

Now the issue:

I am using a delta printer and I would like to have a failsafe that forbids to drive the arms through the endstops and also through the ground plate (using the height gamma_max).

However the documentation says "Do not enable limit switch options if you are using a linear_delta arm solution" in http://smoothieware.org/endstops

What is the current reason to prevent the use of limits for delta printers? Would it be possible to implement this maybe as a separate case if linear_delta is used?

I would also like to know if it is currently possible to use only endstops on one side and soft limits on the other side. Assuming you only connect the min end switches and configure \*_max. Is it then also possible to use the \*_max values as soft limits on the max side? This relates to #461 and should apply to cartesians/deltas/*

wolfmanjm, 2015-02-02:
The reason is limit switches make no sense for use on a delta. They can only trigger when the arms are at the top of the towers and that is a very rare circumstance.
You should always follow a G28/home on a delta with a G0 Z50 to move the carriage off the top so you can safely jog.
Limits for deltas are very hard to do as there is no simple rectangular area that a delta can print in.

closed 2015-02-02 by wolfmanjm

minad, 2015-02-02:
I disagree. When the end switches are hit at some point during a G0 command this should be interpreted as a failure. The print is so or so damaged/incomplete in this case. The machine should stop.

I don't want the area to be limited but I want the arms limited in such a way that you cannot drive through the endstops and damage the machine. This you misunderstood. It should be very well possible to implement that. It should also be possible to limit with gamma max to avoid hitting the ground plate.

And you also didn't answer half of my questions :-(

minad, 2015-02-02:
@wolfmanjm Maybe you could help clarifying this. If I interpret the limit implementation correctly at
https://github.com/Smoothieware/Smoothieware/blob/610ac8be9bee2a111cf8f7affe00a762e275f785/src/modules/tools/endstops/Endstops.cpp#L234 it could already do what I want, namely protecting the machine from driving through the endstops.

The only open question is then if soft limits could be added as a feature as suggested above. But this is maybe a separate issue then.

minad, 2015-02-03:
@wolfmanjm @arthurwolf I actually tested that end limits work perfectly for deltas to avoid driving througgh the endstops. So why does the documentation say that one should not activate the limits for deltas? See http://smoothieware.org/endstops Maybe the warning should be removed there?

wolfmanjm, 2015-02-03:
No it is there for a reason and in fact does not work. You are lucky.

minad, 2015-02-03:
Could you please be more specific? What does that mean? Why is it not working in general? 

 It would be really helpful to know... Thx in advance :-)

wolfmanjm, 2015-02-03:
The problem for deltas is that you have to move off the endstops after homing, there is a procedure for that that works for cartesians but will not always work for deltas. If you have trims set to a big enough value that they move off the endstops far enough then it will probably work. https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/endstops/Endstops.cpp#L287

minad, 2015-02-03:
Thx very much for explaining! Yes I adjusted the trims in such a way that the endstops are not triggered after homing. I would like to add that explanation to the docs if you agree.

minad, 2015-02-04:
I added a few sentences to the documentation