Smoothieware/Smoothieware #590: Bug:  resume command does not work from LCD panel custom menu

rmcniff, 2015-02-04:
It does not appear that the resume command works from the connected Viki 2 LCD panel when configured as a custom command.  The suspend commands seems to work fine, but when trying to resume, the print remains suspended.  

The following lines were set in the config to set up these custom commands:

> custom_menu.suspend.enable                   true              #
> custom_menu.suspend.name                     Suspend_Print     #
> custom_menu.suspend.command                  suspend           #
> 
> custom_menu.resume.enable                    true              #
> custom_menu.resume.name                      Resume_Print      #
> custom_menu.resume.command                   resume            #
> 
> #Configuration for suspend/resume
> leave_heaters_on_suspend 		     true	       #Leave the heaters on when suspended
> after_suspend_gcode                          G91_G0Z10_G90_G0X-50Y-50        #The gcode to run right after suspend command is issued
> before_resume_gcode	                     G91_G1E1_G90      #The gcode to run right before the print resumes

wolfmanjm, 2015-02-06:
are you printing from sdcard?
can you issue resume from pronterface and see what happens?

rmcniff, 2015-02-12:
Yes, this is while printing from the SD card.  I tried what you suggested using pronterface to issue the resume command after suspending from the Viki 2.  Only then did I see a message about waiting for the heaters.

It turns out that resume works fine using the custom menu, and the real issue is that the "leave_heaters_on_suspend" setting in the config does not appear to work.  In my config I have the following settings:  

> #Configuration for suspend/resume
> leave_heaters_on_suspend 		     true
> after_suspend_gcode                          G91_G0Z10_G90_G0X5Y5
> before_resume_gcode	                     G91_G1E1_G90
  
Yet, while suspended, the heaters shut down even with leave_heaters_on_suspend set to true.  Then when I did the resume command it only appeared that resume was not working.  I just didn't expect to wait for everything to heat up again, and concluded that the resume had failed and stopped everything prematurely.

wolfmanjm, 2015-02-12:
Ahh ok so the bug is leave_heaters_on is broken but resume in custom command is OK?
Can you change the title if so?

rmcniff, 2015-02-13:
Yes, that's right, I've determined that resume in the custom commands works fine, but even with leave_heaters_on_suspend set to true, the heaters still cool down during suspend.  I have updated the title.

wolfmanjm, 2015-03-11:
This does not seem to be an issue anymore?

closed 2015-03-11 by wolfmanjm