Smoothieware/Smoothieware #592: Plan9 network filesystem support

minad, 2015-02-05:
This is the simplest network filesystem protocol I know of. There are a lot of [implementations](http://9p.cat-v.org/implementations) and even a [very small one for arduinos](https://github.com/echoline/NinePea) which could be easily adapted and C++ified. 9p is supported by the [linux kernel]( https://www.kernel.org/doc/Documentation/filesystems/9p.txt) with the tcp transport and thus would make it very easy to browse around on the smoothie compared to this crude sftp solution that we are having right now. I would love to see this and I also offer to implement that if you are interested.

arthurwolf, 2015-02-05:
Hey.

That sounds like a very interresting feature.
I'll let others jump in if it's not the case, but I think if you want to implement this it would get merged.
Being able to access the Smoothie file system over the network does sound pretty cool :)

Cheers !

minad, 2015-02-05:
Great!  I also dropped you an email concerning that. Since I am very enthusiastic with my 3d printer right now I might find time later today to hack on it. Expect something soon :grinning:

mhackney, 2015-02-05:
Yes! And count me in if you need help/testers. I’m on a mac and struggle with USB. I am now (as of last night) actually printing (!) and connecting via eth to upload images/configs and gcode using the python script. It works but a little inelegant. I was thinking this morning, “a simple get/put over tcp would be a nice entry for the smoothie contest. You beat me to it. I’ve implemented network transfer protocols.

regards,
Michael

arthurwolf, 2015-02-05:
I can't find much info about plan9. Do you have some info/tutorial about how to use it on a Linux system ? It sounds a tad obscure, compared to things like samba and nfs.

minad, 2015-02-05:
I added a link in the above https://www.kernel.org/doc/Documentation/filesystems/9p.txt

Plan9 is obscure but in fact well supported. For example KVM uses it to expose the host filesystem to the VMs. It uses the virtio transport for that. We will use the tcp transport.

minad, 2015-02-05:
It is also very simple I have to add, so you can easily write a client in python or whatever. So the sitatution is at least as good as with this sftp upload thing that we have right now.

minad, 2015-02-05:
@mhackney What about 9p support on macosx?

mhackney, 2015-02-05:
See http://9p.cat-v.org/implementations <http://9p.cat-v.org/implementations>
 and http://9p.cat-v.org/applications <http://9p.cat-v.org/applications> lists an OSX 9p application (programmers editor. There should be enough there to figure it out.

minad, 2015-02-05:
Ok sounds not as great as on Linux. On Linux it is already included in the kernel. But maybe there are also drivers for windows/mac available somewhere?

mhackney, 2015-02-05:
I’ll check. We only need client side on mac though correct.

mhackney, 2015-02-05:
Looks like maybe so: https://code.google.com/p/mac9p/ <https://code.google.com/p/mac9p/>

It is a couple of years old and not active but 9p is pretty simple so maybe not much work to maintain. I’ll try to compile and install it later today.

wolfmanjm, 2015-02-06:
My opinion is that we have sftp, it is more than adequate, it is simple and uses virtually no memory. Anything more complex is going to be very difficult to get working within the very limited environment we currently have. SO we do not need two ways to do it, it is a waste of flash which is also getting scarce.
So my vote is a big NO.

wolfmanjm, 2015-02-06:
Additionally this will make the sdcard corruption issue much worse. the file system will not support concurrent access to the sdcard, this is what currently causes a lot of cards to get corrupted. this will just make that even worse. So I would not recommend you spend too much time on it unless you can figure out how to fix the concurrency and corruption issues as well.

triffid, 2015-02-06:
On 6 February 2015 at 14:35, Jim Morris <notifications@github.com> wrote:
>
> Additionally this will make the sdcard corruption issue much worse.
>

Only if plan9 exposes the raw disk instead of providing file access via
smoothie's fatfs layer

minad, 2015-02-06:
@wolfmanjm I think this is a good idea. It would be much better than this half hearted sftp implementation that we have right now. SFTP is not more adequate. Nobody uses SFTP. This is a personal opinion. And yours is different than mine. 9p is very simple and well supported at least under linux.  There are a lot of client libraries available for C, python, ruby etc which would also allow something like a smoothie-upload.py. I agree that the sd card issues have to be fixed at some point. But this doesn't have to do anything with this issue.

In general I have to complain a bit about your attitude. You seem to be only interested in issues that are important for you. You obviously like closing issues immediately without explaining while they are still somehow open. For example #591 is still an issue. Only because you don't want to work on it doesn't mean it is not there anymore. Maybe someone else wants to work on it. What is the current policy here? 

At the same time there is something announced like the smoothiecontest to get new developers in the boat, however you behave as if you are the dictator of the project? I would like to hear  @arthurwolf and @triffid's opinion on this. This whole thing is frustrating if you start thinking about contributing and someone like you @wolfmanjm is immediately very negative with a big NO despite previous positive comments by @mhackney and @arthurwolf. 

What do you have to lose? If you motivate people, they might contribute something useful. You can still complain afterwards and critize the implementation, the relevance, the code quality whatever.
Maybe you don't want to put it in the core but maybe it can be provided as a separate build. You don't even have to maintain that part of the code. Maybe I would maintain it if I contributed it. And if at some point I stop doing that, you can still remove it.

I also think that such a project should be done in such a way to fit different people, not only you. And this might mean that code is included that you don't use or don't like. I think the main idea of smoothieboard is to fit for different purposes, different uses, different people, isn't it?

minad, 2015-02-06:
@triffid 9p exposes the file layer, not the block layer. It works like NFS/Samba but is much simpler. The implementation will be about 1000 lines of code.

wolfmanjm, 2015-02-06:
@minad this section is for comments, I commented, that is my opinion. I really don't give a damn what you think to be honest. I comment based on my experience having written all that code. You can take it or leave it, it is your choice what you spend you time on. It is an open source project, you can fork, branch do whatever you like.
FWIW I closed that other issue because you said you were going to work on this instead.
If you want to be gatekeepr of this project go ahead. It takes up a lot of your time I hope you have a lot of free time.

minad, 2015-02-06:
Hi @wolfmanjm,

please don't misunderstand me. I have seen how much you did for this project and this is great work - I mean I am using it right now!! I don't want to diminish that and no one can challenge the fact that you did a lot. But there are two things: Your work and how you integrate other people into the project. I only critized the latter and I hope that you see my point. It would just feel much better for newcomers if their ideas are taken a bit positively. However I also think that it is fair if you directly say if an idea is nonsense if it will never be merged. But if there is a slight chance even if it is not in line with what you want to do, I think there is not much to lose. Think about it. Your opinion matters but the opinion of others does to. You work a lot to keep this project maintained and I understand that you want to keep it that way. Probably I won't be able to contribute as much but maybe I could add some bits to it to make it better, not only for me but also for others. This is the reason I don't want to fork or even keep patches to myself. So let's not fight and have fun - I mean this is a fun project after all!

Now from the technical side: I already started working a bit on it (also the reason why my point above was maybe a bit harsh). Since the architecture is quite modular this changes only about five lines in the Network.cpp to add the service similar to telnet. Then there is this submodule which is encapsulated. There shouldn't be any more changes to other modules. If we get memory problems they should be easy to tackle by selectively disabling parts of the modules. But this can be solved when we are there. The sdcard problems are worrisome, but at first I have to come to the point where I see them (Maybe through my work on 9p support??). What do you think about creating a github issue where you document the problem, steps to reproduce, debugging efforts etc?

Daniel

mhackney, 2015-02-06:
Jim, on the concurrent access to the sd issue, I was thinking something like this would help minimize that problem, but I don’t understand the code at that level yet. You may recall that I had my share of sd issues last June. My latest efforts have been focused on ethernet connectivity to Smoothie only. That allows me to leave the USB disconnected and yet do everything I need to do (in theory, there is an issue I’m having with tcp streaming from Pronterface timing out) and dodge the Mac’s auto-mounting the sd (I just did more searching this week and can’t find a way to disable this behavior).

Can you profile the issues quickly? Like, do you expect problems with only ethernet connection, only USB (and not mounted on the host), etc. Framing the problem at this level might help with the solution.

On a related note, the Pronterface build for mac (2/3/20156) allows uploading a gcode file via tcp and that seems to be working well for me. The smoothie-upload.py also worked well for uploading the firmware and config (I haven’t tried gcode but it should be fine too) after I increased the timeout value. I have a fast (gigabit) ethernet here with no wifi on any of my work machines (3 machines and the Smoothieboard). The Smoothie is on that network. I don’t know what is contributing to the timeouts but even doing a direct ethernet cable between smoothing and my host, changing cables, changing ports on my gigabit switch, etc have no effect. This indicates to me an issue on the smoothie (mine is likely haunted). With only 1 smoothieboard I can’t isolate problems as being mine, configuration/network, smoothie firmware/hardware. Are you aware of any network timeout issues others are having? Could this in any way shape or form be related to a slow/corrupt/inferior sd card or card reader? (my card reader had a short in it if you recall from last year and I had to cut the solder bridge. that was only one of a number of hardware issues).

Thanks!
Michael

minad, 2015-02-10:
I close this since #597 was merged. Please test it on your platform! It works quite well for me on linux (doesn't mean it is bug free ;). I will fix upcoming issues. @mhackney Did you have some success with the macosx driver? Maybe we could also test alternative clients for other platforms. This would be also useful when we write documentation in the smoothie wiki about 9p support for other platforms.

closed 2015-02-10 by minad