Smoothieware/Smoothieware #596: When changed the readings_per_second, PID autotuner does not work properly

arms22, 2015-02-09:
When readings_per_second = 100,
does not reach the target temperature

When the target temperature is 0 , I think because TemperatureControl is to turn off the heater pin .
```
uint32_t TemperatureControl::thermistor_read_tick(uint32_t dummy)
{
    ...
    if (target_temperature > 0) {
        ...
    } else {
        heater_pin.set((this->o = 0));
    }
    ...
}
```

arms22, 2015-02-10:
By be modified as follows, and is now reaching the target temperature.
Stability of PID control was also increased.

https://github.com/arms22/Smoothie/commit/a02b7079e93aee11317accbc7b88bb684979162d

wolfmanjm, 2015-02-22:
I don't think  these changes fixes anything that I can see, setting the temp to a negative value does not really do anything and is illegal anyway. Also this patch no longer applies to the current edge code as there is no on_gcode_execute.
If you still think there is a bug then submit a pull request against the current edge code.

closed 2015-02-22 by wolfmanjm

wolfmanjm, 2015-08-09:
I stand corrected some of this was a bug and is now corrected. I did not fully understand the issue.
Thanks

arms22, 2015-08-17:
I apologize I cannot express that well.
Thank you to fix the issue.