Smoothieware/Smoothieware #598: Library/Smoothie code separation

minad, 2015-02-10:
During my work on #597 I went a bit through the code base and I have some problems distinguishing between external libraries and smoothie code. Maybe smoothie patches to the libraries could be contributed upstream. At the same time we could try to make the separation of the code a bit cleaner.

stv0g, 2015-02-12:
I totally agree with you. I think third-party libs shouldn't be part of the repo at all.
They may be downloaded be the Makefile or by using Git's submodule feature.

Project specific changes may be applied with patches.
But I've heard that the mbed libraries are outdated anyway.
Thats mostly due to the fact, that they are part of this repo.

stv0g, 2015-02-12:
By looking at the source code, I notice a lot of similarities to @adamgreen 's [gcc4mbed](https://github.com/adamgreen/gcc4mbed) repo.

Fortunately this repo also seems reasonable up-to-date and has an improved separation of project and external code.

wolfmanjm, 2015-02-12:
we already use his latest released gcc4mbed.

stv0g, 2015-02-12:
His gcc4mbed repo uses mbed's revision 92. Smoothie uses 30?
And he also has a cleaner directory layout. Clearly separating external libs.

adamgreen, 2015-02-12:
@stv0g I just thought I would give some history on why gcc4mbed and Smoothie both include a snapshot of the mbed SDK rather than a separate download or utilize the git submodule feature.
* Initially the mbed SDK was downloaded from the mbed site as a zip archive and extracted as part of the installation process.  This was the thing in gcc4mbed that failed most often for people.  There would be a networking issue or something weird about the user's environment would cause the download or extract to fail (made more difficult that we were supporting Windows, OS X, and Linux.)  I got sick of dealing with these failures and just pulled the source code into the project.
* I personally like git submodules and would use it for gcc4mbed's snapshot of the mbed SDK but not everyone who downloads gcc4mbed actually uses git.  Many just download the latest version as an archive from github and use it that way.

I think you could switch Smoothie to something like git submodules if most of the people downloading the Smoothie sources are now experienced developers and less experienced people are just pulling the pre-built binaries.

logxen, 2015-02-12:
I've been trying to talk Arthur into using submodules for other reasons too. I think if we did that we'd also need to make a nice git tutorial page on the wiki that explained a simple workflow for getting everything and staying up to date.

stv0g, 2015-02-12:
@adamgreen Thanks for the infos.

I understand that Git submodules make things more complicated. And as long the project uses recent versions of the library, I'm totally fine with using a copy inside the repo.

I only noticed some larger diffrences between your gcc4mbed repo and Smoothie's.

wolfmanjm, 2015-02-12:
First off we barely use MBed anymore, and I think the plans are to stop using it altogether.
Second it would be a total nightmare to have some people using old Mbed and some people using new mbed.
Stability means using what you know works. I see no reason to use a newer version of mbed as what we have works, and there is no need to make use of newer version. We also have partially fixed several of the libs we do use, and again it would be a nightmare to try to get those upstream. I think at one point triffid hunter tried to get his fixes to the Mbed USB accepted upstream, and I believe they were not accepted. Which is why all the USB stuff is broken out into the libs/USB.

Git modules are fine for STABLE upstream libraries. However none of the libraries we used are stable upstream. (except maybe adams ;)

closed 2016-06-08 by wolfmanjm