Smoothieware/Smoothieware #600: Explain the code structure

minad, 2015-02-10:
There are src/modules and src/libs. How are these separated in general? For example the network part telnet/sftp/plan9 is in libs but belongs probably to modules while the uip network stack probably belongs to libs? This is also related to #598 since smoothie and library code seems to be somehow intertwined.

Maybe you can explain that or we might want to change this a bit. Probably it would be good just to have some section in the README which explains or defines how the code should be organized.

minad, 2015-02-10:
I am pretty sure that Network should go to modules. @wolfmanjm What do you think?

wolfmanjm, 2015-02-10:
nope it stays where it is. thanks.

arthurwolf, 2015-02-10:
@minad Hey. So essentially, what goes into modules/ is stuff that registers for events : http://smoothieware.org/moduleexample
The stuff in libs/ is lower level and is stuff that modules use, or that provide basic functionality ( USB, Ethernet, etc )

I think what's going on here is : we use mBed as a HAL, but it doesn't provide everything we want, so some low level stuff is in libs/ instead of in mBed. We've been talking about adding a HAL layer to Smoothie, it'd work ( a bit ) like this : mBed ( or other HAL ) -> Smoothie HAL -> libs/ -> modules/
I think if things were setup like that, what you are talking about would then be translated into moving part of Network out of libs/, and into the Smoothie HAL.
But we don't have that abstraction atm, so things should stay the way they are.

Cheers :)

minad, 2015-02-10:
thx for the explanation

arthurwolf, 2015-06-21:
Closing this for now. I do plan on documenting the structure of the code, but that'll probably come after the V2 port.

closed 2015-06-21 by arthurwolf