Smoothieware/Smoothieware #605: Allow compilation with Debian's arm-none-eabi toolchain

stv0g, 2015-02-12:
I dont like to install an additional arm-none-eabi toolchain to my system, just for Smoothie.
I'm already using the `arm-none-eabi` packages (including newlib) from Debian Jessie.

Most of the code (including mbed libs) compiles well.
I only get some linker errors for the the standard libraries which are postfixed by `_s`.
After removing those postfixes from [build/common.mk](https://github.com/Smoothieware/Smoothieware/blob/edge/build/common.mk#L145).

```
Linking ../LPC1768/main.elf
../LPC1768/./modules/utils/simpleshell/SimpleShell.o: In function `std::string::_Rep::_M_dispose(std::allocator<char> const&)':
/usr/include/newlib/c++/4.8/bits/basic_string.h:249: undefined reference to `__malloc_free_list'
```

My installed Jessie packages:
```
ii  binutils-arm-none-eabi                2.24.51.20140604-3+5                   amd64        GNU assembler, linker and binary utilities for ARM Cortex-A/R/M processors
ii  gcc-arm-none-eabi                     4.8.3-9+11                             amd64        GCC cross compiler for ARM Cortex-A/R/M processors
ii  gdb-arm-none-eabi                     7.7.1+dfsg-1+6                         amd64        GNU debugger for ARM Cortex-A/R/M processors
ii  libnewlib-arm-none-eabi               2.1.0+git20140818.1a8323b-2            all          C library and math library compiled for bare metal using Cortex A/R/M
ii  libstdc++-arm-none-eabi-newlib        4.8.3-9+4                              all          GNU Standard C++ Library v3 for ARM Cortex-A/R/M processors (newlib)
```

stv0g, 2015-02-12:
See PR #607 for a fix.

wolfmanjm, 2015-02-12:
№ we need people to use the toolchain we provide otherwise things get way out of hand having to support people using different toolchains.
You can use whatever toolchain you are happy with. However the supported approach is to use the toolchain we provide.

closed 2015-02-12 by wolfmanjm

minad, 2015-02-12:
But what about a patch which lets you keep your toolchain and also makes it compatible with the supposedly "more standard way" according to @stv0g?

wolfmanjm, 2015-02-12:
I should explain a bit. Adding a new compiler can introduce very unexpected and hard to debug bugs into the code, which can show up at very bad times during a print. These are really hard to test for and fix.
Once we find a compiler version that produces stable code for printing we stick with it for as long as we can. Stable means a great number of people have used the code for a while without any significant issues. I like to keep edge in a state that it is stable and usable for everyday printing.

FWIW we did not upgrade to the current compiler for well over a year or so, and then we only upgraded because I started to use a lot more STL and the old compiler had major bugs with STL.

minad, 2015-02-12:
I propose a different approach:

1. Allow the code to compile with different compilers.
2. Everyone should use what they like.
3. They find bugs for you.

The huge mass will use the prebuild binaries that you provide which are build with your preferred compiler to ensure stability.

minad, 2015-02-12:
And if you also provide the toolchain everything is perfect! The courageous or crazy can then still use their own :)