Smoothieware/Smoothieware #612: Ethernet activity causes printing hangup

cederom, 2015-02-14:
I have noticed that Ethernet activity causes printing to fail in the middle of work. Ethernet network broadcasts and similar seems to hang Smoothie print process.

Problem can be easily reproduced when using web interface and/or uploading file while printing.

Quick fixes may include:
* disabling Ethernet port while printing.
* dropping all unknown Ethernet packets.

Still, stability issues persist even then Ethernet port disconnected, but Ethernet seems to be additional instability source, which is really annoying for long time prints..

wolfmanjm, 2015-02-14:
Yea this is unavoidable due to memory limitations and CPU power. Sorry. Simply do not try to access while printing.

closed 2015-02-14 by wolfmanjm

cederom, 2015-02-14:
Please note that ANY kind of Ethernet activity seems to cause problems, even when I do not use WebGUI. There are different devices on LAN which may produce broadcasts, which WILL hang the Smoothie. If not possible to fix Ethernet parser, please disable Ethernet when print starts, then enable after print ends.. That should do the job, and it will be more elegant than re-pluggin cable each time..

wolfmanjm, 2015-02-14:
there is really no way to do that. feel free to look at the code and see if you can figure out a way to do it. I wrote it and cannot see a way do do that

wolfmanjm, 2015-02-14:
FWIW not everyone has this problem. Some people use ethernet all the time with no problems.

nophead, 2015-02-15:
>Yea this is unavoidable due to memory limitations and CPU power.
I have run my machines with Ethernet using a 16 bit MCU running at 25Mhz with only 8K RAM and 64K flash since 2007 and they are not affected by network traffic. You have a much more powerful processor.

Unless you run in promiscuous mode the hardware should filter out packets not directed to your MAC address unless they are broadcasts. Also if you are connected with a switch rather than a hub (which are rare these days) the switch filters everything except your traffic and broadcasts.

Broadcast for protocols you are not interested can simply be ignored, possibly in hardware. In my case only ARP broadcasts are processed and that is not time consuming. If you use Ethernet interrupts they should be lower priority than the stepper interrupt.

What on earth do you do with Ethernet packets that causes a lock up?

wolfmanjm, 2015-02-15:
@nophead we do, packets are filtered out. FWIW the OP was not using a stock Smoothie but a heavily patched fork from somewhere else. AFAIK no one else been having issues with networks slowing things down. The comment I made was regarding accessing the sdcard (ie a web request for a new page) while printing. Sdcard accesses can slow the system down enough to cause pauses. We do not have a multithreaded system here.

minad, 2015-02-15:
@wolfmanjm Please keep this open! I would like to investigate this also in relation to the 9p fs. I promise.

minad, 2015-02-15:
This is an important stability issue.