Smoothieware/Smoothieware #613: SD/FAT filesystem corruption

cederom, 2015-02-14:
SD storage seems to be vulnerable to filesystem corruption. Corruption shows up when uploading files via USB/UMS and/or Ethernet webgui. I have noticed other prints to become damaged after time of several file upload/delete cycles.

Please consider msdosfsck on boot.

Using Zmorph3D printer.

wolfmanjm, 2015-02-14:
this is well documented in the wiki. Sdcard MUST be disabled or unmounted when not specifically copying files. Concurrent access to the sdcard is not supported.

closed 2015-02-14 by wolfmanjm

cederom, 2015-02-14:
I ONLY use WebGUI for upload as UMS caused problems. There is no concurrent access. Still the FS gets corrupted.

wolfmanjm, 2015-02-14:
Then you either have a bad sdcard (there have been several reported cards that don;t work well is SPI mode), or you are powering off or resetting while writing to the card.
Even if it were possible to code and run msdosfsck on smoothie it would not help with this issue.

wolfmanjm, 2015-02-14:
Even having the USB plugged in the host (especially macs) will randomly access the card. This will cause corruption. Are you saying you have no USB even plugged in?

cederom, 2015-02-14:
Correct. No USB connected. Only Ethernet upload. I have tried several SD cards.

Maybe using built-in microSD slot for firmware then enabling SD on Front Panel and using pre-formatted card each time with new GCODE would help..

I have already asked Zmorph3D for theit SPI connections for the FrontPanel. I will try and report back :-)

wolfmanjm, 2015-02-14:
I thought Zmorph 3d have their own fork of Smoothie? They made significant changes to the code base. If so and if you are using their fork of smoothie then I suggest you ask for support from them.

cederom, 2015-02-14:
They stold me change is Serial to HID, nothing more. I will switch to pure Smoothieware and report back then :-) Any easy way to build Smoothieware on OSX/MacPorts or Linux is the fastest way? :-)

wolfmanjm, 2015-02-14:
They were not entirely correct then, looking at their fork it has extensive changes to a large part of the code. It also does not appear to be using the the current edge code but was forked sometime ago. Obviously we do not support third part forks of the code.
Yes you can easily compile from Mac but there are also several prebuilt binaries which i recommend you use. details, as always, are here ...http://smoothieware.org/ and http://smoothieware.org/compiling-smoothie

minad, 2015-02-15:
Also please keep this one open. I would also like to investigate. You don't have to care about these issues but if you keep issues then you show that you, we or I (how you define it) are working on it and care about it. It is also to collect information. Then people can just post there problems here and don't have to find it in the wiki.

arthurwolf, 2015-02-15:
@minad I don't think we should accept tickets for un-supported forks. If this issue is with a fork, the issue should be opened in that fork's issue tracker.

minad, 2015-02-15:
@arthurwolf @wolfmanjm Why are those zmorph guys not contributing back? Did someone ask them? Or are there totally different goals?