Smoothieware/Smoothieware #614: Not all gcodes are document in the wiki

minad, 2015-02-15:
For example M20 is missing. Is there are way to generate this or why is it missing after all?

I also wonder about the format of the output. Are not all of the codes well defined? Marlin obviously uses a different format: http://reprap.org/wiki/G-code#M20:_List_SD_card

arthurwolf, 2015-02-15:
I've done a lot of work recently trying to keep the wiki up to date.
If you see anything missing, don't hesitate to add it, it's a wiki.

minad, 2015-02-15:
I know. I just wanted to point it out for people (including me) to work on it. I usually keep issues open for awareness (if they are not completely resolved).

minad, 2015-02-15:
Also the reprap wiki should be kept up to date concerning smoothie. But I would prefer to automatize this :stuck_out_tongue_closed_eyes:

arthurwolf, 2015-02-15:
I don't think with the way the code is, we can automate it. That'd be nice but I don't think it can be done.

minad, 2015-02-15:
well one can regexp-"parse" for all these ->has_m ->m constructs and get a good heuristical result. This can be done in a few minutes if you don't expect 100% accuracy. I mean at the end such a check-gcode-script could just print missing ones. False positives are no problem. Perfect syncing is not a good idea, I agree. Except if we might add some kind of stupid docstrings to the c++ code i.e. @gcode M20 ... ;)

arthurwolf, 2015-06-21:
There has been a lot of work on this a while back, closing this issue for now. Please feel free to re-open or open a new one if you notice anything.

closed 2015-06-21 by arthurwolf

sorki, 2015-06-30:
This would be really nice to have as there are undocumented g-codes in code and g-codes for which documentation doesn't match. Implementation can be as simple as parsing comments from source files that would use some simple format like:

    # G30 F<mm/s> Z<mm> probe at current point
    #   F - probing feedrate
    #   Z - height to return to after probing

arthurwolf, 2015-06-30:
Unless jim has an idea of why this is a bad idea, I -think- this is a good idea :) But I often miss stuff.

So, yes, if somebody was to add this to the code source, it would be a nice addition, and I could make a script that updates the wiki.

It's a lot of work though, not sure who will do this.

The exact same thing could be done for configuration options ...

wolfmanjm, 2015-06-30:
It's not a bad idea, but who is going to edit 100's of files? and then write the script to do it?
Not me :)

arthurwolf, 2015-06-30:
I'll write the script, but I most definitely won't edit the files.