Smoothieware/Smoothieware #615: SD/FAT filesystem corruption

minad, 2015-02-15:
Follup to #613 

There are SD card corruption issues. All over the smoothie codebase are hacks or warnings to keep files only open for a short time to avoid those. The goal is to get rid of those and make the SD layer rock solid :)

The purpose of this issue:
- Document your problems
- Please document reproducable situations
- Point out where the issues could be located (code review)

minad, 2015-02-15:
First problem:

Debugging is hindered since the fs layer is very opaque. There are a lot of weird layers like fopen -> ... -> FileHandle -> ... -> FATFS (like in a full stack OS :grin:) @wolfmanjm and I think this should be cleaned up or at least documented. This also has to do something with how and that mbed is used because mbed introduces most of those layers.

nophead, 2015-02-15:
@wolfmanjm recently stated "We do not have a multithreaded system here." but also "Concurrent access to the sdcard is not supported."

How can you get concurrent access if you don't have threads?

minad, 2015-02-15:
We have protothreads which means cooperative multithreading. This is realized using a main loop and events.

Protothreads provide a nice syntax for that. I used it in plan9.cpp. You can take a look at the receive and send methods if you are interested.

nophead, 2015-02-15:
So there is only one real thread, so you don't have to worry about reentrancy. If the FATFS supports more than one file open at a time it should not get corrupted provided different protothreads don't share file descriptors. It should be exactly the same as a single thread opening more than one file.

minad, 2015-02-15:
Reentrancy is not a problem, you are right. Maybe fopen fopen write write fclose fclose is a problem.

I assume that by design FATFS supports multiple descriptors (Would be very stupid otherwise). And it is also not written for multithreading (reentrancy), since that would introduce a lot of complexity and locking issues.

minad, 2015-02-15:
What @wolfmanjm says #613 somehow confirms the problem with multiple open FILEs: "Sdcard MUST be disabled or unmounted when not specifically copying files. Concurrent access to the sdcard is not supported."

nophead, 2015-02-15:
Yes that is where I quoted from but concurrent access is impossible with only one thread. If it means simply the same thread opening two different files at the same time then FATFS must be a rubbish design as the whole point of a file descriptor structure per open file is to allow more than one to be open. You wouldn't be able to copy one file to another if that was the case.

Obviously opening the same file on two descriptors at the same time would cause problems if both write to it.

If writing to two different files at the same causes corruption it would point to a bug around modifying the FAT.

minad, 2015-02-15:
Yes. There are bugs. It is also not a good idea to files open for long. There a hacks to close/reopen files in the webserver and in sftp.

wolfmanjm, 2015-02-15:
@nophead concurrent access can occur when there is a MSD mounted volume, the host OS access the sdcard in raw mode block level, and this is what causes the corruption. multiple open files using fopen are supported and no corruption issues have been noted using that method.

wolfmanjm, 2015-02-15:
The bug with fwrite() is that if you write more than 400 bytes without closing the file and reopening then every other 512 byte block is either 0 or garbage. closing the file and reopening for append every 400 bytes is a workaround for this bug. The bug is somewhere in the fatlib and/or sdcard libraries.

nophead, 2015-02-15:
I see. Bit of a design flaw having two OSes accessing the same device at block level. I don't think that is ever going to work.

minad, 2015-02-15:
Yes, the block level thing cannot be resolved. therefore I am using 9p :)

@wolfmanjm thx for the explanation!

triffid, 2015-02-16:
@minad if you're planning an SD/FAT overhaul, please check out what I've done in https://github.com/triffid/silken-lpc17xx where I've constructed a POC/experimental non-blocking & DMA-backed SD/FAT layer

minad, 2015-02-16:
@triffid Very interesting! How much is this faster with DMA? I have this vaguely planned but this issues is here to collect information at first.

triffid, 2015-02-16:
I haven't profiled it but I expect there is some moderate speed boost.

The major performance hits for SD access are using SPI mode, and not using multi-block read/write commands where appropriate.

That's not the main advantage though, the main advantage is that we're no longer busy-looping in an opaque 3rd party library while the block transfer occurs, so other things can happen at the same time and smoothie wouldn't appear to stall during the transfer

triffid, 2015-02-16:
Given that a 512-byte block transfer takes some time (the SD card has to set up, then we have to transfer when it's ready), consider how many block transfers it takes to simply read the first line of a file...

We have to read the FS header to find the start of the root directory block (1 block read). Then we have to evaluate the first few entries (2 block reads). If the file isn't in the first few entries, we have to consult the FAT (3 block reads) and read the next cluster (4 block reads). And that's just f_open.

Then, when f_read is called, we have to read the first block of the file (5 block reads). When we reach the end of that block, we have to consult the FAT again (6 block reads) and grab the next cluster (7 block reads).

Now consider that all this time is consumed in a classic f_open(); f_read(); pair of functions, and while this happens we can't be reading the serial or crunching gcode or anything...

Switching to DMA but still using f_open/f_read wouldn't change much at all, we still have to read all those blocks and wait for them to complete before returning from f_open/f_read.

Thus, the non-blocking part of my work is the most interesting - with non-blocking, we can issue a f_open work task, then go and do other things while the DMA transfers happen in the background, periodically picking up the read blocks, processing them and either issuing another DMA transfer or flagging that the task is complete.

It also has the potential to speed up the USBMSD significantly, the host tends to read large chunks (8k or more) at once, so we could issue a multi-read and then just DMA almost directly from the SD card to USB while performing other tasks. The SD/play feature would of course have to buffer a couple of blocks to avoid having to interrupt the multi-read, but my system does have some crude task priority stuff.

The same DMA crosslink would be extremely helpful with ethernet, hopefully allowing us to load files off the SD card without interrupting running prints

cederom, 2015-02-16:
From the user perspective SD/FAT access is terribly slow both on Ethernet and USB.

Concurrent access can occur when SD is mounted through USB (automount in case of OSX maybe Linux) and file upload from WebGUI, I guess. Maybe others.

Maybe it would be wise to implement some kind of central arbiter for SD/FAT that would have exclusive access to the hardware. There we could have some kind of mutex/semaphore..

FATFSCK would be also nice to have..

nophead, 2015-02-16:
It's not as simple as a mutex. You would need to mount and un-mount the file system to switch between two OSes accessing it at block level.

minad, 2015-02-16:
@triffid Thx for the details. Nonblocking file access would be a way to go but would require a lot of modifications to the code base if everything is changed to use some kind of protothreads. At first it would make sense to change the file system intensive tasks like sftp, webserver etc. How far is this what you wrote from being ready to integrate? Are there many open issues?

@nophead @cederom there is no way to do ensure block level integrity with only a mutex. You have to mount/unmount all the time. The only way is to intercept the block level accesses and install some kind of FAT access interpreter which translates the block accesses to filesystem accesses :D but this would be kinda stupid. I rather prefer to use then a protocol on the FS layer like 9p. Other hardware (phones, ebook readers etc) is usually changing to USB storage mode and disables everything else when it allows block level access.

triffid, 2015-02-16:
@minad yes it's a daunting list of changes, and what I wrote is just barely enough code to satisfy the requirement of proving that my idea works - it's barely a proof-of-concept, definitely not anywhere near ready for merging!

@cederom mutex won't cut it, if we're doing things safely then smoothie must shut down the MSD entirely before accessing the SD card, and that requires a usb reset which will abort any running prints, and also the MSD driver apparently doesn't like being dynamically allocated at the moment.

In theory we could emulate a usb hub with separate serial and MSD devices plugged into it, but that's a lot of work and I haven't even checked if it's feasible to do on the smoothie platform.

closed 2017-01-28 by minad