Smoothieware/Smoothieware #617: Bug: Jitter / Missing Steps

rfelten, 2015-02-17:
We stumbled on strange noise from the steppers for some particular GCode files generated by the KISSlicer. With the help of a logic analyzer we tracked down the root cause the noise to an error in the step generation (See attached image).

Further we developed an minimal example to trigger the step generation error :
```
; Start at X/Y Zero
G0 X0 Y0
; Make a regular move with an arbitrary  speed
G1 X200 Y200 F9000 
; Move one axis a long distance, another axis a short distance AND change the speed.
G1 X190 Y0 F24000
; Move back to Zero
G0 X0 Y0
```

The error was tested in the current edge branch (b11df1d1943a9df48a8a6388d4da3a6871c341d4) on a original Smoothieboard with the default configuration without any hardware/steppers attached.

Logic analyzer trace see: http://i.imgur.com/efPxGXk.png

 - Signal0: Step of Y axis (moving a long distance)
 - Signal1: Direction of Y axis
 - Signal2: Enable of Y axis
 - Signal3: n/c
 - Signal4: Step of X axis (moving a short distance)
 - Signal5: Direction of X axis
 - Signal6: Enable of X axis
 - Signal7: n/c

The number of "gaps" in the step generation can be increased by set ```mm_per_line_segment ``` an lower value (e.g. 0.5 instead of 5).

A workaround (maybe with side effects [see #10]) is to disable (comment)  ```mm_per_line_segment```  in the config file.

wolfmanjm, 2015-02-17:
Thank you for the analysis. It is perfectly safe to set  mm_per_line_segment to 0, and in fact it should be zero by default in the config, not sure why it is still set to 5. The Gap is caused by the Y axis waiting for the X axis to catch up. This was improved somewhat end of december, but seems it still needs to be fixed for extreme cases like this.

wolfmanjm, 2015-02-17:
in order to replicate this I'll need to know your steps/mm for each axis, your acceleration and your junction_deviation, these can all be found using M503, so just paste the output of that here. Thanks

rfelten, 2015-02-18:
Thank you for the fast reply! We are using the default configuration: 
```
M503
; No config override
 ;Steps per unit:
M92 X80.00000 Y80.00000 Z1600.00000
;Acceleration mm/sec^2:
M204 S3000.00000 Z0.00000
;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
M205 X0.05000 Z-1.00000 S0.00000
;Max feedrates in mm/sec, XYZ cartesian, ABC actuator:
M203 X500.00000 Y500.00000 Z5.00000 A500.00000 B500.00000 C5.00000
;Motor currents:
M907 X1.50000 Y1.50000 Z1.50000 E1.50000 
;Home offset (mm):
M206 X0.00 Y0.00 Z0.00
;E Steps per mm:
M92 E140.0000 P57988
;E Filament diameter:
M200 D0.0000 P57988
;E retract length, feedrate:
 M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
;E retract recover length, feedrate:
M208 S0.0000 F480.0000 P57988
;E acceleration mm/sec^2:
M204 E500.0000 P57988
;PID settings:
M301 S0 P10.0000 I0.3000 D200.0000
;PID settings:
M301 S1 P10.0000 I0.3000 D200.0000
ok
```
Thanks!

wolfmanjm, 2015-02-18:
you may want to try the master branch 

https://github.com/Smoothieware/Smoothieware/blob/master/FirmwareBin/firmware.bin?raw=true

See if that alleviates the issue or makes it worse.

WilstonOreo, 2015-02-21:
We tested master branch, it makes it worse. 
Stepper loses steps more frequently and still the jitter. :(

wolfmanjm, 2015-02-23:
can you tell me the length of the gaps (in ms) on the Y axis? thanks

rfelten, 2015-02-24:
The gaps increase over time (see 1st screenshot). I measured ~0.35ms as the largest gap, see http://i.imgur.com/pXtCo0p.png .
(Created on a Smoothieboard, with mentioned GCode/commit).

Were you able to reproduce the bug/jitter?

wolfmanjm, 2015-02-24:
No I have not been able to duplicate it yet, but I am pretty sure I understand what the problem is. and if it is that there is not an easy fix for it other than the new motion control.
A slight change in the speed on the second segment would help a lot too, maybe try 300 mm/sec instead of 400? or even 200?

PetteriAimonen, 2015-03-01:
I was able to reproduce the issue using default config with mm_per_line_segment = 5. I think this may be related to the lost step issues I'm seeing, which occur on smooth curves that are driven fast.

I used ARM ITM to get info about the interrupts:
http://kapsi.fi/~jpa/stuff/pix/smoothie_pause.png

Looks like there is PendSV occurring in order to calculate the next move. The stepper has no instructions before PendSV finishes. Maybe it could be modified so that there is always one move in buffer that can be given immediately to stepper?

wolfmanjm, 2015-03-01:
these are two different issues. this particular issue has nothing to do with short segments, it is a problem with only doing acceleration every 1ms. This means that axis do not finish at exactly the same time, so the fastest axis has to wait for the slower axis to finish. in extreme cases this can be a long wait if the slower axis has only a few steps to do vs the long axis has a great number of steps. This issue will be mitigated with the step per tick fix.

The short segment issue is a known issue, for very short segments the lookahead queue can not be kept full as the steps complete before the G code has been planned and entered into the queue. It is a little better from SD, but even then very short segments complete before the next gcode is read from the sdcard. Filling the block queue before starting can only delay the inevitable in this case. Eventually the planner queue will run dry and the segments start behaving as individual moves starting and stopping at 0. By only starting to process the queue when it is full will help, as the start stop can be applied to 32 segments vs 1 segment, but every 32 segments it would need to stop and wait for the queue to fill again, this will result in jerky movement but not as bad as every segment starting and stopping.

winkelmann, 2015-03-02:
Might it help to increase the planner queue size to lets say 64?

PetteriAimonen, 2015-03-02:
I don't think the problem has much anything to do with the planner queue.

But yeah, like wolfmanjm said, the 80us gap in my capture is actually a bit different from the 300us one in the original captures: the step on the other axis is positioned differently, when it occurs much later the next move start is delayed even more.

Maybe if the stepper was made to continue at the move end speed until next move starts, then the few "predicted steps" will be taken into account in the next move.

wolfmanjm, 2015-03-02:
It does already try to compensate for the time it takes to switch over to the new block. if you look at the code you can see that. However there is a limit to how much you can compensate for and how many steps you can issue at the highest frequency before the stepper starts to complain.

PetteriAimonen, 2015-03-02:
Hmm, I must have missed that compensation. Can you point to the compensation code, just for curiosity's sake?

PetteriAimonen, 2015-03-02:
Ah, I guess by compensation you mean the steps issued at high frequency after the new block begins.

I was thinking about a different approach: just keep issuing the steps at the latest speed when the block ends. Then when the new block begins, fixup any difference by subtracting/adding to the new move length.

wolfmanjm, 2015-03-02:
that is certainly an interesting idea. A better solution, althoug somewhat more complicated, would be to calculate the next block ahead of time so there is no delay when switching over to the new block.

PetteriAimonen, 2015-03-02:
Yeah, that was what I was trying to say by "one move in buffer". But as you say, it is quite complicated.. especially as sometimes the moves are very long and sometimes short, so the G-code executed actions would still have to be done later.

Also it would only resolve the delay after last stepper stops moving. I don't know what this "step per tick" solution for the "not finishing at the same time" problem is, but I think the "continue at same speed" would resolve two things at the same time: the "block switchover delay" and "not finishing at the same time". I agree that it is not quite as clean solution as precalculating the next move, but due to how the motion control works the speed differences between blocks should be minimal anyway.

PetteriAimonen, 2015-03-02:
Here is a beta version of the "predictive stepping". Seems to fix the issues on logic analyzer, but I haven't tried it on actual hardware yet. (Also it still has a few bits of debug ITM writes)
https://github.com/PetteriAimonen/Smoothieware/commit/2f42dfdd1357842d34359e9d767403082409e40c

I'll do more tests tomorrow and see if it fixes more than just this testcase.

arthurwolf, 2015-03-02:
> PetteriAimonen@2f42dfd

What is "predictive stepping" ? Can you explain it for an idiot ( me ) ? :)

wolfmanjm, 2015-03-02:
does this fix your issue (short segments) or this issue which is a fast axis waiting for a slow axis to catch up? Should really have a different issue open for the short segment issue.

PetteriAimonen, 2015-03-03:
@arthurwolf: After StepperMotor finishes move, it takes some time before it gets the new move from the next block. This time gets longer if the axes finish at different times due to rounding errors. Previously this caused a pause in the step signals, which jerked the motors and could cause lost steps. "Predictive stepping" keeps issuing steps at the same speed until new move is received, then it compensates for any difference in step counts.

@wolfmanjm: I don't think I have a short segment problem, the symptoms aren't like block queue emptying. I must admit I don't really understand my problem very well yet, as I have been unable to reduce it to a test case. I'm kind of hoping it is this one, but I'll get more data now that I have a suitable logic analyzer.

wolfmanjm, 2015-03-03:
from your description of your problem it was a block queue empty issue. I know that very short segments will drain the queue.

As for the predictive stepping, I presume you check that there are in fact more steps for that axis, otherwise you will introduce very nasty errors. FYI it does not always decelerate to zero if there are no more moves, due to the junction deviation. So I am not sure how you can predict without looking ahead in the block queue.

PetteriAimonen, 2015-03-03:
Yeah, let's wait until I have more data on my issue before we discuss it further. I haven't reported it because I haven't been able to isolate it; I just jumped on this issue because it had an easy to reproduce test case.

I do not look ahead in the block queue. Instead it relies on the fact that it will decelerate to slow speed, even though not zero. The delay between move end and PendSV finish is short, typically 0.1 ms and probably 1-2 ms worst case if one axis is particularly late. So the motors at slow speed will not move many steps in that timeframe.

When PendSV finishes, any axes that did not receive a new move are given a fake 0-step move. They then take back any unwanted steps (if they happened to make them, probably at most 1 or 2) and stop.

arthurwolf, 2015-03-03:
@PetteriAimonen : That's a clever solution, but unfortunately, the side
effect of moving some extra steps then having to take them back, is not
acceptable.
There are better solutions here ( mainly the new motion control code that
has been worked on recently ), once that's re-integrated into Smoothie I
don't think there will be a need for this fix
And if there is still a problem, there are ways to make block loading much
faster by pre-loading them, so we'll do that.

PetteriAimonen, 2015-03-03:
@arthurwolf: Mechanical inaccuracy will in any case be much higher in acceleration/deacceleration situations than 1-2 microsteps. But yeah, it's ok if the fix is not wanted, I'm only doing these fixes to make Smoothie usable for myself, whether they end up in mainline is just bonus. Currently I cannot operate my printer at full speed due to the issues, and I guess it'll take months before the new motion control will (possibly) fix them.

PetteriAimonen, 2015-03-03:
Yeah, turns out my issue has different cause but similar symptoms (N millisecond pause in steps). The predictive steps fix would probably fix that also, but as it is probably a race condition, better to fix the root cause.

(After consideration, I still think that overshooting by 1-2 microsteps is much more acceptable than pauses in the step stream. Also that is more of a theoretical overshooting, absolutely no overshooting occurs in practice with the default config and the test case of this issue, difficult to say about all possible configs with huge junction_deviation.)

PetteriAimonen, 2015-03-23:
For what it is worth, I have my own fix here:
https://github.com/PetteriAimonen/Jerkware/commit/5f4ee095a5c031207e609d2816e2f2aeabe54e92

It seems to work better for my machine. The main fix is changing acceleration calculations so that they try to keep the axis positions in sync. Step prediction just bridges over the small gaps in block changes, not so important really.

Fabi0San, 2016-01-12:
Hello guys, 
I've been hitting this issue on a 200 step/mm delta at about 60kHz stepping rate.

I'm curious whether the new motion control @arthurwolf mentioned is ready for edge.
If not, in which branch can I find it? 

@wolfmanjm, is there a way to get tracing info out of the motion control component? 
I would be useful to know the actual step frequency hitting the steppers since it is obfuscated by delta geometry.

I might also give @PetteriAimonen solution a try, thanks for sharing Petter.

Thanks a lot.
    Fabio.

wolfmanjm, 2016-01-12:
@Fabi0San are you sure this isn't a stepstuck issue? DRV8825 are particularly susceptible to this problem when one tower is running very slowly? using fast decay rate may fix it for you. It is unlikely to be the problem referenced in this issue. Allegro drivers also have this issue, but not quite as noticeable.
Exactly what makes you think you are missing steps? or that there are gaps in the step pulses?

wolfmanjm, 2016-01-12:
The new acceleration stuff is ready to go, I implemented it my own firmware wolf3dware (where I can use the FPU to do floating point for the steps), however it is a huge amount of work to get it into smoothie without breaking everything, and at the moment I don't have the energy to attempt such a thing. I am working up to it :)

Fabi0San, 2016-01-12:
I need to look more into it be 100% sure but with my simple repro g code(2 long segments) the beta tower makes horrible noises and the effector ends 5mm lower than it should be. Reducing speed fixes it.
I'll investigate stepstuck and give fast decay a try...

Looking forward to your new acceleration stuff on smoothie :)

Thanks,
    Fabio.

wolfmanjm, 2016-06-08:
This is fixed in the upcoming stepticker/aceeleration per tick rewrite.

closed 2016-06-08 by wolfmanjm