Smoothieware/Smoothieware #619: Wrong progress reports

minad, 2015-02-18:
The number of bytes in total is wrong. Maybe some fs problem.

~~~
0 % complete, elapsed time: 1687 s, est time: 1245307818 s
SD printing byte 5652/3735928559
~~~

Also this elapsed time report is wrong. I just let it run for a few seconds.
This reports somehow the full 4GB?

Someone else seeing this?

minad, 2015-02-18:
maybe elapsed time is about 30 minutes since I aborted the print before. Then it also showed "SD printing byte 0/3735928559". I have to investigate a bit better what is going on here.

wolfmanjm, 2015-02-18:
Maybe this is a file you uploaded via p9? and the size is screwed up for some reason. I have never seen this issue with files uploaded or written using msd or file upload. 
Elapsed time also works pretty well, but it is possible an abort does not reset it properly.

minad, 2015-02-18:
Yes, but the file is playing correctly. Also ls reports correctly. For me the issue seems limited to progress. I will investigate and come back here.

arthurwolf, 2015-02-18:
does ls -s show it with the right size ?

minad, 2015-02-18:
@arthurwolf 

~~~
progress -b
SD printing byte 725/3735928559
ls -s /sd
castle.gcode 2640289
~~~

minad, 2015-02-18:
3735928559 = deadbeef

is this what malloc does?

minad, 2015-02-18:
lol i think i found the bug by simple code inspection. I use M32 and you forgot to initialize file_size there. There is also quite some code duplication and this can result in such a bug. Srsy guys?

wolfmanjm, 2015-02-18:
I didn't write or add the Mxxx stuff someone else did, use git blame before you blame me.

minad, 2015-02-18:
I don't blame anyone. Here is the fix :grinning:

minad, 2015-02-18:
But this code needs cleanup badly.

PR #621 (2015-02-23): M32 fix
Fixes #619
closed 2015-02-23 by wolfmanjm