Smoothieware/Smoothieware #635: Unexpected 2 ms pauses in middle of stepping

PetteriAimonen, 2015-03-03:
Somewhat similar to issue #617, but seems to have a different cause.

http://kapsi.fi/~jpa/stuff/pix/smoothie_steppause_overview.png
http://kapsi.fi/~jpa/stuff/pix/smoothie_steppause_begin.png
http://kapsi.fi/~jpa/stuff/pix/smoothie_steppause_end.png

Total 2 ms pause, with no deceleration before / after the gap. I think if the block queue was running empty, there would be deceleration? Now it is running at 10ksteps/s before the pause, which easily causes lost steps when 20 steps are missing from the flow.

What is different from #617 is that all axes have finished moving. Also PendSV completes quickly and the kernel is idle until next PendSV is triggered by .. something. It seems like a race condition, but the PendSV handling seems fine to me.

It seems to be quite random, but I have a gcode that triggers it every now and then; I'll keep debugging, just added this issue as a note to myself and others.

PetteriAimonen, 2015-03-03:
Note about the signals there: 0 is SWO trace data, 3, 5 and 7 are step signals (x, y, z motors).

wolfmanjm, 2015-03-03:
You need to add a trigger pin to the start and end of pendsv, this is already in the code and can be enabled with make STEPTICKER_DEBUG_PIN=P4_29 then you can trigger off that, it will show when the axis stop/start relative to the end of moves, and how long the pendsv takes before the steps start again.
If these are extremely short segments then I still think it is a feeding issue, although it would decelerate to 0 in that case unless the next block is already partially computed, which would be a bug.

PetteriAimonen, 2015-03-04:
Nah, I'll just take ETM trace and see what the code does. The PendSV begin/end is already visible from the ITM trace above.

They aren't particularly short segments, some 1-2 mm long.

PetteriAimonen, 2015-03-04:
Ah, it is closely related to the "axes finish at different times". But in this case, the slow axis is the Extruder. This causes a bit different symptoms in the trace, because the delay is not in Stepper::stepper_motor_finished_move(), but in Extruder having not called block->release().

Also underlying cause is a bit different because Extruder controls its own speed. Maybe that is why I'm getting 2 ms delays whereas the other bug has 0.4 ms max. Maybe also the reason why it seems to be somewhat random for me?

Is this also something that will be fixed by whatever is the correct fix for #617?
In that case I'll just sweep this under the carpet with my predictive stepping hack for now.

These files reproduce the issue for me, at least several times per each gcode run:
http://kapsi.fi/~jpa/stuff/other/smoothie_config_bug635.txt
http://kapsi.fi/~jpa/stuff/other/smoothie_gcode_bug635.txt

arms22, 2015-03-09:
I think the IRQ is enabled again.

    StepTicker::PendSV_IRQHandler
        StepTicker::signal_a_move_finished
            StepperMotor::signal_move_finished
                Stepper::stepper_motor_finished_move
                    Block::release
                        THEKERNEL->call_event(ON_BLOCK_END, this);
                            Conveyor::on_block_end
                                HeapRing<kind>::is_empty
                                    __enable_irq

triffid, 2015-03-09:
On 9 March 2015 at 13:29, arms22 <notifications@github.com> wrote:

> I think the IRQ is enabled again.
>
Unlike AVR8, the Cortex-M3 core leaves the global interrupt flag enabled
when servicing an interrupt.

This allows higher priority interrupts to preempty lower priority ones.

The N in NVIC stands for "nested" ;)

Toggling the global interrupt flag inside an interrupt is therefore not
problematic like AVR8.

arms22, 2015-03-09:
triffid 

Thank you for information.
I understand.

wolfmanjm, 2016-06-08:
This is fixed in the upcoming stepticker rewrite

closed 2016-06-08 by wolfmanjm

qharley, 2016-06-08:
what is the projected timeline for the stepticker rewrite?

[image: photo]
*Quentin Harley*
Founder, Morgan 3D Printers
p:+27120041883 | e:quentin@morgan3dp.com | w:www.morgan3dp.com | a:4 Burger
Ave, Lyttelton Manor, Centurion

wolfmanjm, 2016-06-08:
a few weeks. But you are not going to be happy as a lot of your stuff will no longer work or even be 
included in the edge build. It will be conditionally compiled.
But as you maintain your own branch I think that will be ok.

qharley, 2016-06-08:
How much effort will it be to get it to work with the new code?

wolfmanjm, 2016-06-08:
extuder, endstops and zprobe has to be completely rewritten. once that is done you can look to see what you need to do. your arm solution should be ok, but your zprobe stuff and scaracal may need some work. they will not be included in the default build, once you get them updated we will reconsider including them in the build. However if all your users use your builds then they should probably remain optional builds.