Smoothieware/Smoothieware #636: when use m109&m190 for wait TEMP, send m105 check TEMP  the answer different form the pringting

davidzter, 2015-03-08:
when use m109  T:150.9/200.0 @210 
when use m190  B:34.2/60.0 @255
when printing     ok T:197.3/200.0 @210 B:60.1/60.0 @10
How to do the same.

wolfmanjm, 2015-03-08:
can you please rephrase, I do not understand the question at all.

davidzter, 2015-03-08:
when use m109 s200 for wait TEMP for targer, this time i use printrun send m105 the answer is T:150.9/200.0 @210 ,when printing ,i use printrun send m105 the answer is ok T:197.3/200.0 @210 B:60.1/60.0 @10, why the same command the answer is different

wolfmanjm, 2015-03-08:
They are different commands M109 shows the temp as it is heating up M105 shows the current temp, this is not a bug so closing this issue. If you have questions this is not the place to ask them this is to report bugs. Thank you

closed 2015-03-08 by wolfmanjm