Smoothieware/Smoothieware #637: Abort should shut off heaters

tempo502, 2015-03-09:
Default behavior for an "Abort" command should be to de-energize all heaters / lasers / or other energy-emitting loads to produce a safe shutdown state. I would recommend switching all FETs to an off-state as default behavior. This includes overriding active "heat and wait" type commands. 

If users require any FETs to remain on during an abort, that should be a user-configured option rather than default behavior.

Ccecil, 2015-03-09:
User configured for sure.  I prefer that my bed stays on when I need to abort/restart a print.

GearheadRed, 2015-03-12:
I would second the user configurable option. I too prefer my bed remaining on after an abort. I would be fine with the hot end going off during an abort though.

Bouni, 2015-03-23:
And the motors should also stop if moved through a jog command or a directly issued command.
At the moment if you click for example "move X 100" you can't abort that move, which is kind of dangerous in my opinion!

arthurwolf, 2015-03-23:
Bouni : if your machine is dangerous, you shouldn't rely on Smoothieboard
to stop it anyway, you should have an emergency stop button that actually
turns off the power.

tempo502, 2015-03-23:
I have to disagree. Some machine configurations become less safe if power is completely removed. Killing primary power likely shuts off cooling fans, which may need to stay on to keep the machine in a safe state. Or a machine that suspends loads against gravity may need to keep motors immobilized to prevent damage/injury. The proper behavior for an e-stop button is technically supposed to come from an equipment-specific risk assessment, but most e-stop usage cases will be covered by immediately de-energizing output FETs and canceling all active move commands.

Bouni, 2015-03-23:
I didn't thought about emergency situations in this case, but about situations like i issue a command that let a axis move for a certain distance and by accident i issue the wrong direction which may result in lets say my spindle crashing into my work piece. In this situation it would be nice to have a command   that stops all dangerous movements.

If i cut the power to the steppers, the machine would stop, that's right! But if i release the button immediately after pushing it, the machine will continue to move, because the smoothieboard doesn't get a signal that it should stop all movements.

Maybe i'm not aware of this pin/function, so is there something like a emergency stop input?

By the way, i work with huge CNC machines in my day job and I saw some serious accidents caused by improper designed emergency circuits (like the immediate release i mentioned above). Nobody got injured but the damages were enormous and i want to prevent such situations for my machine :-)

arthurwolf, 2015-03-23:
> I have to disagree. Some machine configurations become less safe if power
> is completely removed. Killing primary power likely shuts off cooling fans,
> which may need to stay on to keep the machine in [...]

What I was trying to say is : we do not want to take the responsibility to
tell you that Smoothie can be relied on to handle an Estop signal and do
the right thing. And we most probably never will. If you chop off your ear
with your lathe-laser-3000 of death and you were relying on Smoothie to
prevent that from happening : NOT OUR FAULT ! Do -not- rely on Smoothie for
safety, and implement security precautions that do not need Smoothie to do
stuff.

If using the estop to cut power is dangerous in a given specific machine,
then a solution that is not cutting power -and- is not relying on smoothie
should be implemented ( spindle controllers have estop inputs typically,
and some solution could be found with the enable line of stepper drivers
that need to stay on ).

tempo502, 2015-03-23:
Arthur, with utmost respect to you, that's a shitty answer. Declining to implement safety functionality because of the risk it may not always work is completely ass-backwards. You are never safer by not including software safety functions. A 90% reliable function is better than nothing. Systems are made sufficiently reliable via overlapping defenses and failsafes across multiple architecture levels, not by ignoring components that are not 100% reliable. 

Most people implementing machines with Smoothie do not have the awareness or competence to design independent external safety functions. If this was solely an industrial controller, the "leave it to the user to figure out" attitude would be fine, but this platform is marketed at hobbyists. If average users are not able to use it safely by following the setup instructions provided, the product is not fit for purpose. At minimum you need to very clearly warn users that Smoothie alone is not sufficiently safe for reliable machine control without additional external safety functions. (That is, to my reading, what you're saying here. And I agree with that assessment.)

You can and should disclaimer out of end-use liability, but you still have a moral responsibility to make some effort to keep users from burning their houses down. Smoothie is WAY behind the curve on this. Every other controller I've looked at has better safety functionality. I am genuinely concerned by how little attention has been paid to this issue.

arthurwolf, 2015-03-23:
> Arthur, with utmost respect to you, that's a shitty answer. Declining to
> implement safety functionality because of the risk it may not always work
> is completely ass-backwards.
No you still don't get it : I want as many safety features as possible,
please, people should implement them, tons of them.
BUT people shouldn't be relying on them

You are never safer by not including software safety functions. A 90%
> reliable function is better than nothing. Systems are made sufficiently
> reliable via overlapping defenses and failsafes across multiple
> architecture levels, not by ignoring components that are not [...]
Then they shouldn't be implementing those machines, and should DEFINITELY
NOT be relying on Smoothie for safety.

External safety functions are not optional, if people use Smoothie safety
features it should be on top of those.

> If this was solely an industrial controller, the "leave it to the user to
> figure out" attitude would be fine, but this platform is marketed at
> hobbyists. If average users are not able to use it [...]
That's in several places in the documentation, and I add it wherever I see
that it makes sense to.

> You can and should disclaimer out of end-use liability, but you still have
> a moral responsibility to make some effort to keep users from burning their
> houses down. Smoothie is WAY behind the curve on this. Every other
> controller I've looked at has better safety functionality.
I'm not exactly sure what you are refering to here.

> I am genuinely concerned by how little attention has been paid to this
> issue.

tempo502, 2015-03-23:
@arthurwolf Ok, so here's what I'm hearing:
1) The only people who should be using Smoothie are people competent enough to design and implement hardware safety functionality: this platform is not intended for hobbyists or tinkerers, just people like you and me who have some background in electronics design or safety engineering
2) You want safety features in the product, but want/need someone else to implement them
3) You have not looked at what safety features have been implemented by other controller platforms to see what capability gaps exist

Please correct me if I'm misinterpreting any of this. I'm not trying to be rude here, I just genuinely want to understand why the response to "your product doesn't appear to be safe enough" has been so lackadaisical.

qharley, 2015-03-23:
Ai Ai.

What I hear is:
1) People that don't care about safety never will.  They probably won't
implement any safety feature loving crafted into the code.  You cannot
force it on them either.  They are the people driving on the freeway with
the windows open, safety belt tightened on the seat under them just to stop
the car screaming, and the kid riding loose between the front seats.

2) This is Open Source.  If you want it, implement it, or persuade somebody
that can that they want it, or pay somebody to do it for you.

3) Have a look at the code.... it is quite a monster at the moment.  Lots
of work happening at the moment, and getting that right is paramount in
getting it to the next level.  Get it wrong, and nobody will want it -
making safety features moot, right?

What you should hear is this. Everybody want their own features implemented
right away.  Somebody should do it, because if I want it, it must be
important.
I had to implement all the ones I wanted myself.  It turned out good enough
for others as well, and so got incorporated into smoothieware.  If I as a
non programmer can do this, so could anybody else.

Even if nobody else wants it, at least you will have it then.  Everybody
wins. ;-)

  [image: photo]
*Quentin Harley*
Founder, Reprap Morgan 3D Printers
 p:012 004 1883 | m:082 782 3902 | e:quentin@reprapmorgan.com | w:
www.reprapmorgan.com | a: 4 Burger Ave, Lyttelton Manor, Centurion, South
Africa

wolfmanjm, 2015-03-23:
@Bouni there is a solution to your specific request http://smoothieware.org/pausebutton#toc2

Bouni, 2015-03-23:
@wolfmanjm Thanks for the hint!

I think i will write a wiki article about external safety mechanisms and how to make them work with smoothieboard. Maybe not interesting for people building small 3D printers, but for tose building CNC mills/lathes etc.

tempo502, 2015-03-23:
Here's a quick list I put together of possible safety features to implement, just brainstorming. https://groups.google.com/forum/#!topic/3dp-ideas/u9NmyanuR2o 

Let me know what you guys think is high priority and I'm happy to help suggest methods or algorithms or whatever. There are a lot of basic safety checks that can be implemented with only a few lines of code or can be copied from other open-source platforms. That's pretty easy stuff for a core team dev to implement but pretty hard for a non-programmer new to the platform (like me) to implement.

arthurwolf, 2015-03-23:
> @arthurwolf <https://github.com/arthurwolf> Ok, so here's what I'm
> hearing:
> 1) The only people who should be using Smoothie are people competent
> enough to design and implement hardware safety functionality:
Most definitely.

> this platform is not intended for hobbyists or tinkerers,
Of course it is.

You are positing a contradiction where I don't think there is one :
hobbyists should be competent enough to realize the projects they venture
into. Especially in matters that have safety considerations.

If somebody is not able to identify what is dangerous about what they are
doing, and is not able to find solutions to ensure their safety, then that
person should definitely not be allowed to use Smoothie.
( Actually in most countries, people that can not ensure their own safety
or identify if their actions are dangerous, are not allowed to interact
with other people, and are protected from themselves. )

And that's whether you are a hobbyist or not.

> just people like you and me who have some background in electronics design
> or safety engineering
I don't. I have no background whatsoever in electronic design, or safety
engineering, or even programming for that matter.

When I assembled my first 3D printer, at the moment where I had to wire my
PSU to mains, having only done low-voltage stuff in my life before, I took
two hours off to read on mains power, what's dangerous about it and what I
should do to ensure I won't die doing something stupid.
Anybody who doesn't do that ( when they need to ) shouldn't be using
Smoothieboard.

2) You want safety features in the product, but want/need someone else to
> implement them
That is 100% correct.

I haven't actually added any code to Smoothie for years.
I help a shit-ton of people figure out how to best add code to Smoothie
though, and I'll do that for safety features too, most definitely ( I
currently am doing that actually, with several different people/features ).

I want people to add safety features to Smoothie, i won't do it myself, and
if somebody wants to do so, I'll help them ( possibly a lot ).
And if somebody points to a safety feature that's missing from Smoothie,
I'll actively try to get the community to implement that.

BUT, and I won't repeat that enough : people shouldn't be relying on the
Smoothie firmware for safety.
All the safety features Smoothie can provide, should be considered EXTRA.

> 3) You have not looked at what safety features have been implemented by
> other controller platforms to see what capability gaps exist
I actually have.
I'm asking you to provide specific examples so I can more precisely
understand what's bugging you here.

> Please correct me if I'm misinterpreting any of this. I'm not trying to be
> rude here, I just genuinely want to understand why the response to "your
> product doesn't appear to be safe enough" has been so lackadaisical.

It was not.

tempo502, 2015-03-23:
@arthurwolf, thanks for clarifying your position. 

I'm of the opinion that people designing hardware/software have an obligation (whether ethical or legal) to protect users from common and obvious mistakes or system faults, to the greatest extent reasonably possible. Users are human. People make mistakes. Some of those mistakes can be easily detected and mitigated. And if a system designer can eliminate a potentially dangerous failure mode via relatively simple code, s/he should do so. Some generic examples:
- A heater should not be allowed to turn on if the temp sensor associated with it is disconnected
- The heaters should not "fail on" if the MCU freezes
- Default temp alarms and limits for should be in place, and those defaults should be suitable for the most common usage cases

If users choose to override the defaults to be less safe, that's on them. But you can cover some of the genuine oversights and mistakes that are INEVITABLY going to happen to hundreds of people using the product.

This is super basic stuff. There is open-source code freely available for use (albeit with some gaps). It's genuinely irresponsible to release a 3DP controller (or even just a "hotend" module) into the wild without even a cursory attempt at heater fault detection.

arthurwolf, 2015-03-23:
Hey.

> @arthurwolf <https://github.com/arthurwolf>, thanks for clarifying your
> position.
>
> I'm of the opinion that people designing hardware/software have an
> obligation (whether ethical or legal) to protect users from common and
> obvious mistakes or system faults, to the greatest extent reasonably
> possible.
Are you implying we don't full-fill that obligation for some reason ?
Or that we don't recognize it exists ?
I'm extremely confused here.

I guess reasonable is a very fuzzy term too : implement the feature-set of
what people consider reasonable, and you get a new bigger feature-set that
is now considered reasonable. Talking from many-times repeated experience
here.

We try very hard to be up to date with the state of the art for this kind
of features, and we push strongly towards advancing it. I think that's
pretty reasonable.

Again : I never said safety features shouldn't be implemented, I'm saying
it'd be dumb as fuck to entrust your eyes or fingers or life or house or
family, to something that was coded/designed in part by non-professionals
on their spare time, and is tested by a random hackers around the world.

You should never let your reprap run un-attended, you should never leave
your laser cutter open while it's powered on, and you should most
definitely not entrust Smoothie with anything safety related on a CNC mill
( those things kill people ).

Yes, Smoothie should try to implement as many safety features as possible,
NO these should never in any circumstances be your only line of defence
against a failure, if there is an alternative ( ie. if your spindle
controller has an estop input, USE IT ).

If that's not how things work, then it's way too dangerous to contribute
code to Smoothie, and nobody will do it.

> Users are human. People make mistakes. Some of those mistakes can be
> easily detected and mitigated. And if a system designer can eliminate a
> potentially dangerous failure mode via relatively simple [...]
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/TemperatureControl.cpp#L117

>
>    - The heaters should not "fail on" if the MCU freezes
https://github.com/Smoothieware/Smoothieware/blob/edge/src/libs/MRI_Hooks.cpp

>
>    - Default temp alarms and limits for should be in place, and those
>    defaults should be suitable for the most common usage cases
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/TemperatureControl.cpp#L117

>
>
> If users choose to override the defaults to be less safe, that's on them.
> But you can cover some of the genuine oversights and mistakes that are
> INEVITABLY going to happen to hundreds of people using the product.
Well it's not like we did not fork from existing firmwares that had years
of user experience behind them, or did not ourselves get years of user
experience, or did not release thousands of boards into the wild ...

> This is super basic stuff. There is open-source code freely available for
> use (albeit with some gaps). It's genuinely irresponsible to release a 3DP
> controller (or even just a "hotend" module) into the wild without even a
> cursory attempt at heater fault detection.

... ?

tempo502, 2015-03-23:
We're arguing past each other here. 

I'm saying Smoothie does not have sufficient safety functionality. It does less to protect users than other popular platforms. Even MBI's terrible Mightyboard code from ~2012 does more. 

You're saying 1) Smoothie has some safety checks and 2) it's up to users to make it safer. 

And, to be clear, speaking as an end-user who does safety automation for a living and gives a shit if my house burns down, I'm saying that's inadequate. I'm not running my Smoothie-based bot again until I get more hardware safety features installed. In hindsight, I should have done that from the start, but I didn't expect the software side to be so under-developed.

But I do appreciate you linking to those examples -- I have not read every single obscure code module yet, and didn't know MRI_hooks had any safety functionality. (Not that I would have been able to tell what that code did even if I HAD already read it.) But do you really think one line of code in TemperatureControl.cpp checking if the hot end is TOO COLD is really a good example of safety code? You must, because you reference it twice. But here are some further issues:
- Smoothie doesn't throw any warnings when sensors are disconnected, live or on power-up. 
- It will let me start prints with both sensors disconnected and reading 999C. I tried it just now. For some insane reason, 999C will actually make the "heat and wait" gcode command register as complete so the print will start. That is un-fucking-believable.
- It doesn't check that the heater or temp sensor is physically attached to the hot end, eg by monitoring for temp rise upon initial heating. 
- It doesn't do anything at all if the temp rises to unacceptable levels, such as if users set PID gains too high and get extreme oscillations. (I'm not going to try this one, because I don't want my house to burn down.)

I could go on.

arthurwolf, 2015-03-23:
> We're arguing past each other here.
>
> I'm saying Smoothie does not have sufficient safety functionality.
Well you have your own definition of what sufficient safety is here.
And I absolutely promise you : if we implemented -everything- you want to
be implemented, there'd be some other guy in a few weeks dropping me an
email saying we suck at safety features and we need much more.

This is an Open-Source project, you get what people have implemented.
And there are other people that are working on implementing more.
And you actually have the opportunity to implement stuff yourself.

It does less to protect users than other popular platforms. Even MBI's
> terrible Mightyboard code from ~2012 does more.
If Smoothieboard is not good enough, we'll gladly refund your board, and
you can use the safer makerbot stuff.

> You're saying 1) Smoothie has some safety checks and 2) it's up to users
> to make it safer.
Ok you definitely do not read what I write ...
This is actually getting very annoying.

> And, to be clear, speaking as an end-user who does safety automation for a
> living and gives a shit if my house burns down, I'm saying that's
> inadequate.
You showed you don't even know what Smoothie does ! How can you say it's
inadequate ???

> I'm not running my Smoothie-based bot again until I get more hardware
> safety features installed.
YES, PLEASE, YOU SHOULD DO THAT WITH *ANY* OPEN-SOURCE BOARD, and that's
true for anybody else reading this !
You should have a fire extinguisher, and you should never let your machine
run un-attented, and you should have a fire alarm, and all that obvious,
common sense stuff !

AGAIN, this is coded by amateurs on their FREE TIME !

> In hindsight, I should have done that from the start,
THANK YOU !

> but I didn't expect the software side to be so under-developed.
Saying it's under-developped means nothing : you don't know what's in
there, and you are comparing it to makerbot's stuff which I don't know a
single other person who would ever thing to use that as a point of
reference ... The idea is just bizarre to me. Why choose that abritrary
comparison point, why is that "the right amount of safety", why not more,
why not less ???

> But I do appreciate you linking to those examples -- I have not read every
> single obscure code module yet, and didn't know MRI_hooks had any safety
> functionality. (Not that I would have been able to tell what that code did
> even if I HAD already read it.)
Well like most of Smoothie it does have comments ...

> But do you really think one line of code in TemperatureControl.cpp
> checking if the hot end is TOO COLD is really a good example of safety
> code? You must, because you reference it twice. But here are some further
> issues:
>
>    - Smoothie doesn't throw any warnings when sensors are disconnected,
>    live or on power-up.
>
> That's what that line does ... it essentially cuts everything when a
sensor is disconnected.
This is probably clearer :
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/TemperatureControl.cpp#L410

>
>    - It will let me start prints with both sensors disconnected and
>    reading 999C. I tried it just now. For some insane reason, 999C will
>    actually make the "heat and wait" gcode command register as complete so the
>    print will start. That is un-fucking-believable.
>
> Well that's going to cause a lot of pointless motion, and grind your
filament, but it's not going to burn anything down ...

>
>    - It doesn't check that the heater or temp sensor is physically
>    attached to the hot end, eg by monitoring for temp rise upon initial
>    heating.
That's currently being implemented as part of the contest.

>
>    - It doesn't do anything at all if the temp rises to unacceptable
>    levels, such as if users set PID gains too high and get extreme
>    oscillations. (I'm not going to try this one, because I don't want my house
>    to burn down.)
>
> Same.

>
>
> I could go on.

Please do.

tempo502, 2015-03-23:
To be clear, my stance here is that a hardware controller should not go into the wild (say, out of beta test status) unless the creators can stand in front of a judge and say that they [could not have foreseen / did everything they could to prevent] any catastrophic failure modes that a competent person (ie electrical engineer) would find in a risk assessment. Control systems for hazardous energy sources have significantly greater minimum expectation of safety than, say, an iPhone app. Failure of high-current electronics can kill people. Saying "I just made the product, someone else can make it safe" is an unacceptable answer. That won't hold up in court and it shouldn't set well in your stomach.

The proper time to do safety engineering is at the early system design level, not as a few code snippets added after basic functionality is released. This stuff really needs to get into the design so it can be done right with proper hardware integration. That said, we have a functional product that is growing in popularity, being sold to many people who do not have adequate technical capacity to understand the risks. (The easier you make the product to use, the lower the average user competence level will drop.)  

The challenge is, how do we get the product to the "minimum safe state" it needs to be in to cope with more popular adoption? I'd like to focus on that.

arthurwolf, 2015-03-23:
On Tue, Mar 24, 2015 at 12:23 AM, Ryan Carlyle <notifications@github.com>
wrote:

> To be clear, my stance here is that a hardware controller should not go
> into the wild (say, out of beta test status) unless the creators can stand
> in front of a judge and say that they [could not [...]
Please return the Smoothieboard to us, and we will refund you as soon as we
have it. We'll even pay for shipping.
You do not understand what it is, or how it came about, and I don't want
you anywhere near it, for your safety, and for mine too.

I am serious, I don't want you using it.

-- 
Courage et bonne humeur.

tempo502, 2015-03-24:
Ok @arthurwolf , where do I ship my boards for returns? Back to Uberclock? If you aren't willing to listen to criticism and address technical concerns, I'm not interested in using the platform. I will, however, wait to see what ongoing changes are made before actively discouraging other people from using it. (Jim seems to be doing some good work.)

I use Mightyboards as an example because it's the platform I'm most familiar with. I have several Mightyboard-based printers. (It outperforms Smoothie in print quality, and the volunteer developers of Sailfish are very responsive with concerns and bug reports.) The Mightyboard platform also has an estimated 500,000-1,000,000 machines in the wild, and the only documented electrical safety issues to date (reported in English, anyway) were attributable to user error during non-routine maintenance and/or negligent assembly by Chinese clone manufacturers. It, too, has some design flaws (which I have communicated), but those are multi-point failure modes that only occur after at least two consecutive faults / user errors occur. It's a pretty good comparison point for this discussion. Not perfect, but worth emulating. I'm confused as to why you would be so dismissive of it.

Why am I so concerned about the printer bypassing sensor faults and proceeding with the print cold? Because it's indicative of an embarrassing lack of code testing. This general class of issue should be caught before any non-beta build is released. Having volunteer devs is the exact opposite of an excuse: the lack of centralized control makes it all the more important to perform thorough functional checks of each build. If this were a one-off bug, it wouldn't be significant, but this is not the only concerning behavior I've observed in my Smoothie bot. Multiple issues in a couple weeks makes a trend. 

I think you guys have done a ton of great work on what you see as core functionality for the platform. But the state of the heater control code -- the highest risk part of the system for most users -- is simply deficient. I strongly recommend you reconsider the project priorities. Safe function should be a condition of build release, not an after-thought left up to users to implement several years into the project.

tempo502, 2015-03-24:
I'm done discussing this issue. Thanks to all who took the time to reply.

Bouni, 2015-03-24:
@tempo502 I understand your concerns about the safety, but blaming @arthurwolf or any other of the founders of Smoothieboard for not having implemented more safety features from the beginning is just not fair. They are all together (as far as i know) Hobbyists. 
In my opinion its ok to say that something is not good enough, but then you should submit a pull request that fixes this problem instead of blaming the people who invest many many hours of their free time to build this board.

And, Smoothieboard is still a Hobbyist product and at no point comparable to an industrial grade CNC controller. Just think about what a proper safety setup for a industrial machine costs! That's way beyond the price of a smoothieboard. 

Another fact is that even Industrial grade CNC controllers, as long as they are not safety integrated controllers, need an external safety setup. The controller just get a signal if the safety chain is interrupted so that it can stop its current operations and go to a safe state, but the actual action of safely prevent the drives from moving is done by a safety relay because its NOT allowed that the controller does that in software. 

In my opinion we should not fight each other but instead bring the Smoothieboard (which is a great product :-) ) further!

arthurwolf, 2015-03-24:
> Ok @arthurwolf <https://github.com/arthurwolf> , where do I ship my
> boards for returns? Back to Uberclock?
Yes.

> If you aren't willing to listen to criticism and address technical
> concerns, I'm not interested in using the platform.
You are completely misrepresenting everything I've said in this discution.
Criticism and concerns about technical issues have always been, are
currently, and will always be very welcome.
I think that's been pretty well established by years of community discution.

That's not the issue at all.

The issue is you saying that there is a legal or moral obligation to
implement any of those features.

This is a community driven open-source project, it is what the community
makes it, no more, no less.

If somebody tomorrow adds a plasma cutter module to Smoothie, and then the
day after that somebody else burns their house down using it, it is in
absolutely no way any responsibility of the first guy.

-Everybody- I've talked to in the Smoothie and Reprap communities, these
past years, understands that !

You are the first person ever that I talk to that doesn't ! ( in thousands
! )

So if you don't understand that, you are essentially dangerous to yourself
and this community, and you should please send us your board back.

> I will, however, wait to see what ongoing changes are made before actively
> discouraging other people from using it. (Jim seems to be doing some good
> work.)
I'm pretty sure the Marlin, Repetier or Teacup people would say exactly the
same thing as me about legal and moral responsibility.
I don't think you understand open-source and open-hardware.

If you want to tell people not to use anything open-source, go ahead, be
one of those people, that's fine.

Singling-out Smoothie is pretty dumb though.

I use Mightyboards as an example because it's the platform I'm most
> familiar with. I have several Mightyboard-based printers. (It outperforms
> Smoothie in print quality, and the volunteer developers of Sailfish are
> very responsive with concerns and bug reports.) The [...]
Yes, you definitely are confused here.

I am not dismissive of any given safety feature.

But setting the Sailfish feature-set as some set point bellow which we "do
not have sufficient safety functionality", is arbitrary and useless.

Again : if we had that given feature-set, somebody ( not you, you'd be
happy ) would complain it's not enough and we are criminally responsible
for damage our users cause. So it doesn't mean anything as a comparison
point.

So no, there is no single standard under which it's not enough : for some
it's already enough, for others it'll never be enough. Your preference is
irrelevant.
The only thing that's true here is : We should aim at having as much safety
as possible. Everybody agrees on that.

But NO, nobody in the Smoothie project is responsible for any damage you
cause using the board or firmware. This is an open-source community
project, blame has no place anywhere in it.

And if you don't understand that, again, please give us your board back and
stay very far away from the project, you don't understand it, and if you
get into any kind of trouble you'll end up hurting the community for your
own faults.

> Why am I so concerned about the printer bypassing sensor faults and
> proceeding with the print cold? Because it's indicative of an embarrassing
> lack of code testing.
Please stay away from this project.
I'm pretty sure any other Open-Source project will tell you the same thing
about their project after reading this thread.

> This general class of issue should be caught before any non-beta build is
> released. Having volunteer devs is the exact opposite of an excuse: the
> lack of centralized control makes it all the more [...]
You have no idea what the project priorities are.
You have no idea what the project is ...

> Safe function should be a condition of build release, not an after-thought
> left up to users to implement several years into the project.

PLEASE stay away.

MetalTree, 2015-03-24:
Hi All,
Just thought I’d jump in and support Arthur, I watch this thread as a smoothie board user and I’m not involved in the development of the code, but as a mechanical engineer who has built machines for 25 years. WHAT IS HAPPENING NOW IS AMAZING!!! It wasn’t possible even 5 years ago, don’t start the litigation bullshit with open source stuff because no one wins, we’re on the most amazing ride developing new 3D printers and CNC machines and I don’t really give a fuck about your hot end issues as in reality your not going to burn the house down. I want you super smart guys to keep going and build amazing firmware so I can use it to build bigger and better machines!!!
So when are you boys going to give me a 5 axis CNC (haha….)
Keep up the good work guys!!!!
Cheers
Jason

Bouni, 2015-03-24:
Hi everyone,

i tried to be helpful instead of spill critics all over the place :-)

what do you think about this: http://smoothieware.org/emergencystop

arthurwolf, 2015-03-24:
Thanks a lot bouni, great work as always :)

qharley, 2015-03-24:
Sometimes you are lucky when a customer like that decides to leave.

  [image: photo]
*Quentin Harley*
Founder, Reprap Morgan 3D Printers
 p:012 004 1883 | m:082 782 3902 | e:quentin@reprapmorgan.com | w:
www.reprapmorgan.com | a: 4 Burger Ave, Lyttelton Manor, Centurion, South
Africa

tempo502, 2015-03-24:
I guess the part that I should have explained in advance is that I CANNOT contribute code to Smoothie. Section 12 of the GPLv3 and my pre-existing employee IP agreement are incompatible. Any IP I produce is encumbered, and thus the GPL specifically excludes me from modifying or conveying GPL-licensed works. (Yay, Texas employment law.)

I'm also not much of a programmer, so no real loss there. But I still wanted (past tense) to try to contribute to improving the code. I was under the impression that bug reports and algorithm suggestions would be appreciated. These are concrete improvement suggestions to address specific issues, including some fairly detailed code change suggestions to fix long-standing algorithm flaws:
https://groups.google.com/d/msg/smoothieware-support/1W3ww9LIDlw/jszrLC2r7HkJ (seriously, read this)
https://github.com/Smoothieware/Smoothieware/issues/637 (this issue)
https://github.com/Smoothieware/Smoothieware/issues/646
https://github.com/Smoothieware/Smoothieware/issues/640
https://github.com/Smoothieware/Smoothieware/issues/639
https://github.com/Smoothieware/Smoothieware/issues/638
https://groups.google.com/forum/#!topic/3dp-ideas/u9NmyanuR2o
Nobody ever said you had to implement everything, or even a quarter of it. But the "if you give a mouse a cookie" excuse for not doing anything is baffling. The concept that you shouldn't implement "peer standard" safety features because it's some kind of slippery slope to receiving more safety feature requests is just... I don't have words for that. 

I really don't understand why simply matching what other open-source platforms do to keep users' homes from burning down is an unreasonable request. Why is that considered "we'll let somebody else code it if they want to" territory, and not core functionality? There's a lot of visible movement on improving motion control (which already works quite well) but none on improving heater control (which is demonstrably broken). That's why I think some development priorities are off-kilter.

What's most surprising to me is that pointing out issues with Smoothie generates so much hostility. I have noticed this on multiple occasions from multiple dev team members through multiple channels. It's really remarkable to me that you guys demonstrate such defensive behavior when subject matter experts point out issues. (Not just talking about me here, it's a pattern.) How well a project team takes criticism and tries to correct community misconceptions is a really important factor for long-term success. 

It's my friendly, well-intentioned recommendation that you guys try to be more inclusive of people who want Smoothie to be better, but aren't able to directly work on the code. That group includes me and probably around 99% of the potential userbase for one reason or another.

arthurwolf, 2015-03-24:
> I guess the part that I should have explained in advance is that I CANNOT
> contribute code to Smoothie. Section 12 of the GPLv3 and my pre-existing
> employee IP agreement are incompatible. Any IP I [...]
Bug reports and improvement suggestions are extremely appreciated, you are
completely misrepresenting what is going on here, and that's extremely
upsetting.

I have corrected you on this twice now and you keep saying the same wrong
stuff, which is enough to safely assume you are engaged in a strawman
<https://yourlogicalfallacyis.com/strawman> argument at this point.

You essentially claim the following : you pointed out flaws in Smoothie and
that got some lead smoothie devs upset.

When what *actually* happened, is that you put forward an argument that
smoothie devs are morally or legally responsible for the consequences of
not implementing enough safety features in firmware.

*THAT* is what got me upset, not the pointing out of technical stuff. THAT
is what brings me to asking you to send back your board.

It is the second time I explain to you that you are wrong about what I
object to, and yet you keep coming back to this dumb narrative of :

« Oh I was just pointing out some technical flaws in Smoothie and the devs
got all upset for no good reason, they must not be able to handle criticism
very well, oh oh oh »

NOT what is going on here.

> These are concrete improvement suggestions to address specific issues,
> including some fairly detailed code change suggestions to fix long-standing
> algorithm flaws: [...]
This is again a strawman argument, that's absolutely not what I was saying.

> I really don't understand why simply matching what other open-source
> platforms do to keep users' homes from burning down is an unreasonable
> request.
WHO EVER said that was an unreasonable request ????

I said two times already that we have, currently are, and will in the
future be aiming at providing the most and best safety features possible.

Is that not clear enough ?

I DID say you are unreasonable, but absolutely not in asking for more
safety features.

Where you are unreasonable is in saying that we have a moral responsibility
to add those.

WE DON'T !

That is the only, singular problem I have with what you said.
Stop with the straw-men, stop with the prude attitude, and just concentrate
on the actual issue.

Why is that considered "we'll let somebody else code it if they want to"
> territory, and not core functionality?
Nothing is core functionality !!!
Everything is "if somebody codes it it's awesome" territory !

And NOTHING is "I can trust this bit of code to keep my house safe"
territory.
EVER.

If you do not understand that, please send your board back and we will pay
you back !!!!

There's a lot of visible movement on improving motion control (which
> already works quite well) but none on improving heater control (which is
> demonstrably broken).
That's actually incorrect, we have two contest entries that are about
adding safety features to heater control, and two others that are for more
general safety stuff. So heater safety is one of the things that should
move the most when the coding phase of the contest begins ( some have
actually already started coding/pushing ).
That plus discussions on github, and regular fixes.

The big current re-factoring is motion control because Jim wanted to tackle
that. He does this for fun ! What kind of asshole would go tell him "hey,
instead of doing that you should be doing heater safety stuff".
We should be thankful he's doing the motion control stuff, not asking him
to stop doing useless shit because safety stuff is more important ...

This is all voluntary work, by people that want to help and/or have fun,
there is no boss, there is no plan, there is no obligations, and there is
certainly NO FUCKING RESPONSIBILITY for people contributing code to an
open-source project.
When somebody gives you nice code, for free, you tell him thank you, not
"hey you forgot to add safety features".

> That's why I think some development priorities are off-kilter.
There are no development priorities, there is only whatever whomever wants
to add at a given time.

What's most surprising to me is that pointing out issues with Smoothie
> generates so much hostility.
DOES NOT
STRAWMAN, see above.

> I have noticed this on multiple occasions from multiple dev team members
> through multiple channels. It's really remarkable to me that you guys
> demonstrate such defensive behavior when subject matter experts point out
> issues. (Not just talking about me here, it's a pattern.)
I know of only one other instance, and that has prior-events that you are
most probably not aware of, and that factor into how we reacted.
I can find you hundreds of people that came to contribute in one way or
another, and had a lovely experience.
Making sure they -do- have a lovely experience is actually 50% of what I
consider my job nowadays.

And that sometimes ( rarely ) involves having to face trolls, which is
insanely unpleasant, and can maybe be misunderstood by bystanders as us not
being receptive, but that's not the case.

When somebody does what you are now doing, and misrepresents rejection of a
given position, as us not being receptive to cricism, it can look like we
are indeed not receptive to criticism to somebody outside, but what's
actually happening is you used a strawman fallacy, and caused damage to the
community.

> How well a project team takes criticism and tries to correct community
> misconceptions is a really important factor for long-term success.
It's my friendly, well-intentioned recommendation that you guys try to be
> more inclusive of people who want Smoothie to be better, but aren't able to
> directly work on the code. That group includes me and probably around 99%
> of the potential userbase for one reason or another.
We are inclusive.

tempo502, 2015-03-24:
Arthur, do you really think I'm damaging the community when I assert that people who make open-source products should feel some sense of responsibility for easily-preventable harm to end-users? 

I have some opinions on what damages the community. I think Marlin's use of software PWM with the MCU watchdog disabled is damaging the community. I think Smoothie's inability to tell whether heater/sensor hardware is connected properly is damaging the community. (But I'm glad this is being worked on -- thank you for supporting that effort.) I think more or less everyone's inability to tell when dualstruder temp sensors are installed backwards is damaging the community. I think Smoothie's poor PID implementation and tendency to overshoot temperatures is damaging the community. I think the plethora of controller boards using bootloaders that do not proactively assert-low heater FETs at startup are damaging the community. I think everyone who uses the old Kapton tape J-head thermistor mounting style is damaging the community. And so on.

I do NOT think that pushing for more "ownership" of safety from developers is damaging the community. I don't care if that position disagrees with the way open-source projects are traditionally run. This is not a smartphone app. Hardware control is inherently higher risk than typical open-source software projects, and that needs to be reflected in the attitudes and behaviors of the people working on the project. Standard GPL disclaimers may let you win a wrongful death case in civil court in most jurisdictions, but they won't let you sleep at night after someone's child dies.

Consider an analogy. If I released an "open source chainsaw" design, I would certainly warn and expect users to wear gloves / glasses and respect safe chainsaw handling practices. But the design would also include finger-guards and a safety grip cutoff from day one, and I wouldn't release versions that lacked those features. If I didn't have the competency required to design such features into the chainsaw, I wouldn't design a chainsaw in the first place. That's me. 

My point here is, what the community used to do is not good enough anymore. 3DP is going mainstream. Open-source designs have become easy enough to use that it is no longer reasonable to expect 100% competence from end-users. This is simultaneously cause for celebration and cause for alarm. The hobbyist community has to start respecting and prioritizing safety-in-design much more than it currently does, period. I don't care if thousands of people disagree with me on that -- if they do, they're wrong.

But we really don't have to agree on all that. What's important is that we do clearly agree that additional safety features are worth including now. I don't particularly care if you like me, but I'm not a troll, and I'm not wasting my time arguing about this for kicks or to try to make you look bad. I am an extremely competent person with domain-specific expertise who has operated a Smoothieboard, encountered very concerning machine behavior, looked at the code, and found gaps that I think are worth fixing. I've pointed at open-source code with better methods, put a ton of work into algorithm proposals, and logged lots of apparent bugs. I may be mistaken about some issues, and that's fine, I appreciate you guys correcting my mistakes. But there are real code shortcomings that I can't fix myself. What I can do is provide assistance to someone else interested in fixing them. 

So, here's my question: Am I too dangerous to touch a Smoothieboard, or do you want my help? Please pick one.

arthurwolf, 2015-03-24:
So essentially you now changed your position. Now it is :

> Arthur, do you really think I'm damaging the community when I assert that
> people who make open-source products should feel some sense of
> responsibility for easily-preventable harm to end-users?
But before it was :

I'm of the opinion that people designing hardware/software have an
> obligation (whether ethical or legal) to protect users from common and
> obvious mistakes or system faults, to the greatest extent reasonably
> possible.

Do open-source developers have a moral obligation, or should they just feel
a sense of responsibility ?

Because I ( sorta ) agree with one, and not the other. And it's the one I
don't agree with that I have been talking to you about.

As far as I can tell, none of the Smoothie devs ( so far ) has shown that
they do not care about safety. We do care about safety. It may surprise
you, but there are very few sociopaths in this project.
So yes to your new position, that's obvious, common sense, true pretty much
everywhere, people care about safety ( if only their own ).

But about your old position, the one I've been battling with, that
contributors to this open-source project have an *ethical obligation* to
protect users from common and obvious mistakes or system faults, to the
greatest extent reasonably possible.

To that I say : NO FUCKING WAY.

This is crystal clear to everybody I talk with but you, we say this to
community members, users, and everybody agrees : THERE IS NO guarantee of
safety. Always have a smoke detector, a fire extinguisher, and NEVER leave
your printer unattended !
THERE IS NO WAY we can guarantee any kind of safety, and there is NO
FUCKING WAY people that are nice enough to contribute code for free for the
general good are to be held to an *ethical obligation* to do ANYTHING.

There is no guarantee of safety for Smoothie ( or any other open-source cnc
controller I know of ), and there is no ethical obligation for the
developers to implement any feature set.
And if it wasn't like that the project pretty much wouldn't exist !

If users want something they can let run unattended, they should go buy
some ( much more expensive ) closed-source commercial product that TELLS
THEM exactly how much safety it provides, and that they can sue if it
doesn't provide that level of safety.

WE on the other hand DO NOT guarantee -any- level of safety.

And it'll be like that even after we've implemented 10 times the amount of
safety you want.

So, here's my question: Am I too dangerous to touch a Smoothieboard, or do
> you want my help? Please pick one.

I'm fine with you using a Smoothieboard as long as you show you understand
what I wrote above.
WE on the other hand DO NOT guarantee -any- level of safety.

tempo502, 2015-03-24:
To my mind, a moral responsibility is the same thing as an ethical obligation. I've used both terms in this thread. But that's fine if you see a difference, we can work with that. 

I see Smoothie as an engineered product, and thus you would generally want to apply concepts from any typical engineering codes of ethics. Engineers do, unquestionably, have an ethical obligation to make a safe product. This is universally agreed across all fields of engineering. I see your distinction now though. If you don't consider the development of Smoothie to be engineering, that's fine, you're under no generally-agreed ethical obligation to attempt to ensure safety. (Your personal moral judgment on the issue is up to you.) We can agree on those terms and stop arguing the issue.

But who's making strawman arguments? I've never claimed Smoothie needs to guarantee safety. My position has always been that reasonable efforts should be made to protect users from predictable mistakes and common machine failures that are easily mitigated via software. This does not mean "guaranteeing" safety. "All reasonable efforts" is a legally defined term that does not require implementing every request, nor all that much work at all when you get right down to it. 

Meeting or exceeding the standards set by peers would very likely be interpreted by a court as "reasonable efforts." Thus why I use the features of the most-used open-source controller as the metric for judging Smoothie. It's a very fair standard to apply. I believe a court would see it similarly.

My point is, and has always been, that I don't believe Smoothie's current state represents "all reasonable efforts" to ensure user safety. In particular, being aware of an issue but not fixing it in a timely manner would open the development team to legal action. Safety-related issues truly require a different sort of response from other feature requests / bug reports. The fact that this is open source and based on pre-existing code is only marginally relevant -- your name is prominently attached to the product, so you're a lawsuit target if someone gets hurt. 

Again, we can see things differently on this issue, that's fine. But the opinions of the dev team on this sort of issue do strongly affect how people will view the product.

nophead, 2015-03-24:
You can never rely on firmware for safety unless it is written to aerospace
standards and runs on triple redundant hardware. You must know that for
basic safety a single point failure should not cause a dangerous fault
condition. That means if you want a safe system you need some thermal fuses
or heaters that max out at safe temperatures.

tempo502, 2015-03-25:
@nophead, no one here is saying firmware should be the only safety layer or that it needs to be 100% reliable. Trust me, I'm well aware of safety engineering principles, my day job is working with systems that keep offshore oil rigs from blowing up. You make a system sufficiently safe by implementing DEFENSE IN DEPTH until the residual risk drops to an acceptable level. How many layers of safeguards that requires is a function of the reliability of each safeguard and the desired risk profile.

What is that desired risk profile in this case? I don't know, that's something y'all have to ask yourselves. It can be pretty high for industrial equipment but needs to be pretty low for consumer/commercial applications. Based on anecdotal failure rate reports, what I've seen of typical machine builds, and the "safety pyramid" concept, I'd ballpark the "burn a user's house down" risk for Smoothie to be around 1/100,000 per user per year. (Before you call bullshit, yes, that's a SWAG, but I assess system risks for a living, so please grant me a little credibility on the subject.) 

Admittedly, a lot of this risk comes from common non-Smoothie-specific hardware architecture decisions like low-side switching of high-current loads, or non-redundant temp sensing. There's obviously some hardware risk that software can't fix. But there are a lot of issues that software can prevent from escalating at the hardware level.

Thermal fuses are a last-resort safeguard that are supposed to cover the very small risk after all "soft" design measures have failed. A recommendation to use thermal fuses is not an excuse to avoid fixing code issues. We should have BOTH good code AND good hardware backups. (We're all in agreement on that.) But hot ends are a very specific weak point for thermal fuses: there are remarkably few reasonable options for hot-end-mounted thermal fuses. Hot end manufacturers don't provide fuse mounting points, and no commonly-available fuses will hold indefinitely at 300C but reliably and rapidly trip below the auto-ignition temperature of common filaments. And no thermal fuse will catch a heater cartridge coming dislodged from the hot block and falling onto the print. There are BIG gaps in the ability of end-users to implement hardware safeguards in this case. That's where software safeguards really shine: it's quite reasonable for the controller to detect a heater dislodged from the hot block. The most appropriate safeguard for that is code.

nophead, 2015-03-25:
I am not saying its a bad idea putting it in the code but what I am saying is that even with every trick in the code you won't have a safe system, i.e. one that would pass any safety certification because single fault such as a shorted MOSFET can set your filament on fire.

On the other hand I run several machine with my own firmware that has no safety checks at all and I leave them running over night and in an empty house. The reason I take that risk is my PCB beds max out at 130C due to the resistance rising. My mains powered resistor beds have a 150C thermal cut out. My hot ends use low power resistors that max out around 300C. I have experienced stuck on heaters and all that happened was I built very strong brown ABS objects that smell bad.

There is a small but finite chance my machines made from MDF will catch fire. Most likely from an electrical fault in the wiring or PCBs. My fridge freezer nearly caught fire the other day because water got into the PCB and made it burn. It won't kill me because I have several smoke alarms and a fire extinguisher. My house might burn down when I am out but it is detached so it won't kill anybody.

The really dangerous thing is 40W heater cartridges and no amount of firmware is going to make them safe for attended use without some hardware redundancy. People need to realise that even with all the firmware tricks the risk is still high. Also the risk of from any domestic electronics is better, but not a lot, and they can and do burn people's houses down.

tempo502, 2015-03-25:
If you truly don't mind if your house burns down (no pets, no family, no cherished possessions, lots of insurance) then sure, it makes sense to run your printers unattended I suppose.

If you're using the kind of power resistor I'm familiar with, it's probably ~100% de-rated at this temperature, and not particularly reliable for long-term stable resistance. Using a power resistor outside its rated temp range is vastly more dangerous than a 40W heater cartridge within its rated temp range. Internal resistor breakdown due to thermal cycling, leading to partial coil shorts, is fairly likely over long-term use. Then your 20W resistor becomes an 80W resistor or 200W resistor or whatever. In response to the abruptly high process gain and inevitable sensing dead time, the temp controller will oscillate between insanely hot and only kinda hot. That's where "oops I hit 300C, shut down heaters" code saves your house.

Or even if the power resistor lasts forever, maybe an unattended print fails and filament blobs around your hot block, adding significant insulation so the "max failure temp" becomes a lot higher, in the process maybe dislodging thermistor wiring, shorting heater leads together, whatever. There's lots of failure scenarios here where you really, really want the firmware to have a few lines of code to turn off the power.

Did you know the auto-ignition temperature of PLA is only about 340C?

nophead, 2015-03-25:
> If you truly don't mind if your house burns down (no pets, no family, no
> cherished possessions, lots of insurance) then sure, it makes sense to run
> your printers unattended I suppose.
I would mind, but it wouldn't be a disaster for me. Probably more chance of
dying in a road accident but I still use the roads.

> If you're using the kind of power resistor I'm familiar with, it's
> probably ~100% de-rated at this temperature, and not particularly reliable
> for long-term stable resistance. Using a power resistor [...]
Actually vitreous enamel resistors are rated for a surface temperature of
350C but when they are well bonded to a heat block with cement the surface
doesn't go much above 250C. I have found them to be totally reliable under
these conditions. I have never had one fail even when used 24/7 for years.
The coils are separated by ceramic material, so I don't think they can ever
get a shorted turn and produce more power when run from a voltage too low
to flash over. Since they are run outside their power in free air rating,
but well within the surface temperature rating it is possible I have simply
made a high power resistor that will last for ever by adding a heatsink to
a low power resistor. Certainly it is the same construction as a cartridge
heater which is just nichrome wire in ceramic inside metal and must have
good thermal connection to the heat block.

Alternatively the inside might get too hot despite the outside being
cooled, in which case the nichrome will eventually fail open circuit but if
that is the case it must be a very slow process.

In contrast I have found AL clad resistors do fail after a couple of years
when run outside their power rating at bed temperatures. They go open
circuit and at the same time leaky to the casing. They have a lower surface
temperature rating, so must use lower temperature materials than vitreous
enamel.

> Or even if the power resistor lasts forever, maybe an unattended print
> fails and filament blobs around your hot block, adding significant
> insulation so the "max failure temp" becomes a lot higher, in the process
> maybe dislodging thermistor wiring, shorting heater leads together,
> whatever.
Another feature of my simple firmware is if my host stops talking it can
halt with the extruder running as I use on-off commands rather than E
codes. That has happened and totally encased my hot end in a ball of ABS.
Nothing worse happened. I think a ball of ABS takes away heat rather than
insulating it as it is a much better conductor than air.

> There's lots of failure scenarios here where you really, really want the
> firmware to have a few lines of code to turn off the power.
Yes it has been on my to do list for years, but never got round to it. Some
day I will add all the safety features mentioned but I still wouldn't
consider using a 40W cartridge heater unattended without hardware
redundancy such as the ability to switch PSU off when the hot end is over
temperature from two separate processors, or at least one processor plus a
hardware watchdog.

> Did you know the auto-ignition temperature of PLA is only about 340C?
So I have read, but I don't think my heaters will get that hot when full
on. Also I don't print PLA unattended as I don't use it as a production
material. ABS has an auto ignition temperature of 466C, which my hot ends
will definitely not get to but would with a 40W heater, which can melt
aluminium and probably ignite all filaments.

arthurwolf, 2015-06-21:
Closing this. Please feel free to open a new issue if you want to discuss safety features.

closed 2015-06-21 by arthurwolf