Smoothieware/Smoothieware #638: Tweak temperature safety cutoff defaults

tempo502, 2015-03-10:
The default "max temp" safety cutoff of 1000C is too high to catch some thermistor short conditions. Long runs of small-gauge hookup wire or oxidized thermistor leads may have sufficient resistance to register <1000C temp during a thermistor short. (Yes, the example configs use much lower max temps, which is good, but it would be preferable to also use a safer value for the card-coded default in TemperatureControl.cpp.)

Using 0C as a low-temp cutoff may cause spurious faults with machines used outdoors / in garages in cold climates. I think -20C or -40C is a more reasonable default "min temp" cutoff, although I'm not familiar enough with the ADC to know whether these are actually measurable temperatures or not.

wolfmanjm, 2015-03-11:
The max temp setting actually just stops someone setting the temp too high, it does not shutoff if the temp spikes. the max temp control is if the reading from the ADC is >= 4096. There is currently no high temp shutoff setting.
The min temp was recently added by someone and does shutoff if the temp read is under that value.

tempo502, 2015-03-11:
Hmm, ok. What's the current state of safety code to catch common heater faults? Like thermistor falling out of the hot end, disconnected thermistor, shorted thermistor, etc. 

To my mind, this is "minimum acceptable functionality" level stuff... keeping users from accidentally burning their house down. I don't know if that's in line with y'alls mindset but I figured I'd raise the issue.

wolfmanjm, 2015-03-11:
well currently very little. if there is a short or open circuit the ADC reads either 0 or 4096, in both those cases we shut down. The minimum temp test will shutdown sooner if the temp goes too low. I guess we need to at least add an equivalent high temp cutoff as well. Unfortunately max_temp is already (mis-)used for not allowing accidental setting of a high temp.

qharley, 2015-03-11:
Something to consider would be detecting that the nozzle is "not" heating
even on full on condition. This might indicate broken heater, or worse,
dislodged thermistor

I've seen enough melted hot ends...

wolfmanjm, 2015-03-11:
I agree Smoothie has not enough safety cutoffs. At a minimum there should be a maximum cutoff temp. I also like the idea of a cutoff if the heater has been on for a certain time and not detected a temp rise.

arthurwolf, 2015-03-11:
Two different persons offered to do this as part of the Smoothie Contest.
I'm pretty sure at least one published issues related to this, probably
even a merge request. Not sure what the current status there is.

tempo502, 2015-03-11:
If you really want to get ahead of the pack, add support for redundant temperature sensing. If two thermistors assigned to the same hot end differ by more than, say, 50C, kill the associated heater. It's kind of absurd that people trust a single sensor to run unattended machinery when a variety of sensor failures will burn your house down.

nophead, 2015-03-11:
More like catch up with the pack. Marlin already has high limit, low limit,
redundant sensor and on for too long without rise.

tempo502, 2015-03-11:
I didn't know Marlin had redundant sensing support. Of course, Marlin also disables the hardware watchdog and uses software PWM, which is an insanely dangerous combination. If the AVR freezes, it can output 100% heater power forever. (Smoothie doesn't do that, does it?) I will not use any system with that kind of critical design flaw in my home, period.

nophead, 2015-03-11:
Well if your MOSFET shorts (most common type of failure) then a watchdog wont save you unless you can shutdown your PSU.  To be fail safe you need thermal fuses or heaters with insufficient power to ignite plastic.

tempo502, 2015-03-11:
There's lots of failure modes that the MCU can detect, and lots it can't. 

I'd say if there IS a detected fail-on condition, the bot should retract filament as far as possible and move the extruder to the home position. That minimizes exposure to flammables as much as you can. Might be the difference between just some nasty smoke damage versus igniting the filament spool and burning your house down. (Pound for pound, plastic has more combustion energy than gasoline.)

tempo502, 2015-03-11:
Thermal fuse is probably the ideal solution. Unfortunately, there's not a lot of thermal fuse options that are even remotely suitable. I'm not aware of any that work indefinitely up to 280C but will quickly trip before filament hits its autoignition temp. 3DSystems has a pretty good mechanism that encloses the entire hot end in a little metal shroud and uses that ambient temp to trigger a ~120C thermal fuse, but I wouldn't be surprised if that's patented. I'm of the opinion that we need a mix of software safeguards and hardware failsafes that overlap sufficiently to get up to an acceptable system reliability level. For example, use a high-side failsafe relay on a charge pump type circuit to fix all the safety problems with FETs only switching the ground side. 

This discussion probably merits more depth than a Github issue...

tempo502, 2015-03-11:
Hit the wrong button, sorry.

arthurwolf, 2015-03-11:
I use 120C thermal fuses on my heated beds and they work very well.

But for a hotend, having double thermistor input per hotend would be very
nice.

triffid, 2015-03-12:
Teacup Firmware has had an algorithm that detects "heater not heating despite power input" for years, but it's languished since no-one seems to use it.

If there's this much interest, I imagine it could be ported to smoothie fairly easily.

Essentially it has a window of expected temperatures based on history of power delivered to the load, and if the temperature falls outside this window for too long it flags an error and shuts down.

tempo502, 2015-03-12:
That's a good feature, although it may run into RAM issues again. And if we're going to go to the trouble to log an empirical power/temp curve, there's an entirely better way to control the heater instead of PID. At that point, you should just switch to an autotuned heat loss model based controller. It's fairly straightforward. I've posted about this in various settings in the past, eg see https://groups.google.com/forum/?nomobile=true#!topic/3dp-ideas/934X-Q4kJEs

nophead, 2015-03-12:
I have been thinking along the same lines but the problem I see is the
warmup and response to change would be very slow. For example if you
calculate the target temperature requires 50% power and just apply that it
will heat slowly until your integral term ramps up. I.e. you clearly want
100% power until you are close to the target temp and 0 if far above it,
not the predicted power. If that is applied it would approach the target
asymptotically. The power you calculate is only the right power to use when
it has reached the target. It is never the correct power when it off from
the target.

Perhaps a combination of an auto tuned model with PID using the model to
make the PID variables linear?

tempo502, 2015-03-12:
You shouldn't be using PID when you make major setpoint changes anyway. There's no point in applying it during initial heating. I winds up, D acts against the desired change, and P is always saturated until you get close to the setpoint. 

Makerbot/Sailfish bypass PID and use on/off control when error > 10C. That works well. The specific bypass range you want to use depends on a parameter like (power / heat capacity)*tau for the system. (High rate of temp change and long time constant = overshoot. You want to avoid the initial overshoot taking you out of PID range.)

So, if you're using a heat loss model, you would probably just implement a simple model bypass to apply max PWM until within X degrees. Or, if you really want to do it right, you can actually calculate how many PWM-seconds (heater duty cycle * seconds of power applied) are required to reach the setpoint. Then push the exact amount of power into the system that is required to reach the setpoint, and then switch over to the heat loss model controller.

Let's be blunt -- heating up a block of aluminum is an insanely simple physical system. At any normal extruder temp, heat loss is almost exactly first order with respect to temperature delta to ambient. (I've measured this.) We could EASILY implement exact analytic solutions (with auto-calibrated parameters like heat capacity) and just include a small integral correction to fix the small residual error due to non-ideal behaviors.

nophead, 2015-03-12:
Yes I found it to be nearly linear a long time ago: http://hydraraptor.blogspot.co.uk/2007/08/thermals.html.

I also found I get only +/-2C with bang bang with my own firmware which samples quickly (hundreds of times per second). Marlin doesn't sample fast enough, so everybody moved to PID but I don't get much, if any, improvement with Marlin's PID and its reaction to set point changes is worse. It only does PID in a window but as soon as it crosses that threshold it starts to wind up the integral so it then has big overshoot when it gets to the target.

Also noise actually has a beneficial effect with bang-bang as it just turns it into PWM and creates a P term inversely proportional to the noise amplitude.  

This led me to considering the fastest possible way to achieve target without any overshoot. That is to apply full power until you are within the overshoot distance from the target, then apply zero power and let it overshoot to the target and then apply the power needed for that temperature with some PI feedback. I don't think any control system can beat that performance can it? To reach target faster it would need to apply power for longer and would then overshoot.

I made an add hock re-flow oven control program using more or less that method for cooking PCBs: http://hydraraptor.blogspot.co.uk/2009/12/cooking-with-hydraraptor.html. I couldn't see how PID would work any better.

So yes I agree a self calibrating model that knows the power temperature curve and the heat capacity should work best. It should be able to work out how far it will carry on rising after power has been removed so knows when to switch to PWM.

tempo502, 2015-03-12:
Right, pure feedback control is inherently non-optimal because it only responds to error. Feed-forward control is better for well-behaved systems. 

Being blunt again -- 3D printers use PID because it's insanely robust to bad tuning, not because it's anything resembling an optimal approach. Most people's printers are tuned wrong. (Inappropriate autotune algorithm / poor hand-tuning / failure to tune at all.)

From a professional/industrial control standpoint, PID is used when you're too lazy to make a model, or the process is too complex/random to make a model. Auto-tuned model controllers with hybrid feed-forward/feed-back loops have largely replaced pure PID for performance-critical control applications. 

There are even algorithms that can do the model tuning live-online with a series of tiny setpoint changes, but that's overkill here. A one-time autotune each time you change hardware configuration is fine.

tempo502, 2015-03-12:
On the sample rate issue... Sampling parameters don't matter half as much as sensor responsiveness. There's really no point in measuring temp with a faster sample period than, say, 1/10th the time constant of the process. Measuring ethos kind of very half second is perfectly adequate for heater control. The lag is hugely important, though... Bang-bang control is unacceptable for processes with significant sensing lag. It's too easy to get oscillation. It's only really suitable when you have excellent heater/load/sensor thermal coupling. Typical 3DP hot ends and HBPs have too much sensing lag for bang-bang to be a good idea. 

Using noise to simulate PWM on a bang-bang controller is an ugly hack. You're assuming the noise distribution is symmetrical around the real temp. That's not always true. 

Why are you worried about rapid heat-up? It's a very low-value criteria for this kind of heater control. For example, if I'm waiting 5-10 minutes for HBP surface temp to stabilize, it's pretty meaningless whether the hot end is at temp in 1 minute or 3 minutes. Much more important criteria are disturbance rejection, long-term stability, tolerance of sensing lag, and lack of overshoot. This is also where bad PID tuning bites people. They tune for rapid initial heatup/settling when it's far more important that the temp stays stable mid-print. The two criteria use fairly different tuning parameters. 

Consider printing PVA. It will pyrolitically decompose into tar at temperatures only slightly above the target printing temp. Overshoot at initial heatup is unacceptable. Mid-print overheating due to significantly dropping extrusion rate / melt power is unacceptable. 

Or, consider a filament where visual color/shininess varies significantly with temp. A slow +/-2C oscillation will put faint zebra stripes in the print. My autotuned Smoothie bot does this. My ZN-tuned Sailfish bot does not.

arthurwolf, 2015-06-21:
Closing this. Please feel free to re-open if you want to discuss it further or even better have some code.

closed 2015-06-21 by arthurwolf