Smoothieware/Smoothieware #639: PID Derivative calculation requires input filtering

tempo502, 2015-03-10:
Summary:
The PID loop is extremely sensitive to noise in the derivative term, which is fairly common due to ADC sampling error / measurement jitter. Some simple input filtering would greatly improve the stability and consistency of PWM output.

Details:
The default 20hz sample rate is around a hundred times faster than the time constant of typical thermistors, so the derivative term is being recalculated WAY more often than is necessary or physically meaningful. The resolution of the temperature sensing is sufficiently low relative to the sample frequency that d_factor*dT/dt appears to be either zero or an extremely large number for most PID loop ticks. In particular, renormalizing the D gain to suit the 20hz tick frequency greatly amplifies the D response on ticks where the temp measurably changes. This puts a lot of undesired junk values in the PID PWM output. And it actually clamps / diminishes the impact of the D term on the heater control since it's being applied in very brief impulses that exceed the max/min PWM range. 

Any sort of low-pass filter on the D term would significantly improve the stability of the temperature control. An easy option (used by Makerbot firmware & Sailfish) is to calculate a moving average (eg store last n dT/dt values in a ring buffer and recalculate a new average each PID tick) and use that average to calculate the D parameter output. For example, see 
https://github.com/jetty840/Sailfish-MightyBoardFirmware/blob/418058e57e63eec62d90572d49ac56b6b366692a/firmware/src/MightyBoard/shared/PID.cc#L72
This implementation uses a 2.5 second moving average (PID tick every half second, five point moving average). Ideally, you want to use a window somewhat shorter than the time constant of the temperature sensor, but long enough to adequately smooth out most measurement noise. A half second or one second window would probably work well for Smoothie.

Here's an example log of bad heater PWM behavior (note intermittent 0 and 255 values):

B:62.2 /70.0 @255
B:62.1 /70.0 @83
B:61.9 /70.0 @87
B:61.8 /70.0 @91
B:61.7 /70.0 @94
B:61.6 /70.0 @97
B:61.6 /70.0 @100
[...]
B:61.9 /70.0 @155
B:62.0 /70.0 @157
B:62.1 /70.0 @0
B:62.2 /70.0 @160
[...]
B:63.2 /70.0 @174
B:63.3 /70.0 @0
B:63.4 /70.0 @176
[...]
B:64.8 /70.0 @183
B:64.9 /70.0 @184
B:65.1 /70.0 @0
B:65.2 /70.0 @184
[...]
B:66.0 /70.0 @185
B:66.2 /70.0 @0
B:66.3 /70.0 @184

wolfmanjm, 2015-03-10:
I'm not sure if this is exactly the same thing, but we already filter the temp readings in a ringbuffer. we sample 32 readings and take the one median value which effectively filters out most spikes.
The ADC on the LPC1769 is extremely noisy, this is a well known fact. add to that people like to run the thermistor wires along motor cables there is a lot of noise.

You can change the frequency of readings in the config file...

```
temerature_control.hotend.readings_per_second   20    # 20 per second
```

Setting this much lower is known to reduce noise.

I think what we already do is equivalent to what you suggest here, we just do it higher up the chain.

tempo502, 2015-03-10:
Good to know, but, I don't think sensor filtering alone is adequate with slow-changing temperatures and rapid PID refresh frequency. Say the physical system is changing temp at a rate of 0.1C/sec, measuring temperature with resolution of 0.1C and doing PID at 20hz. Even with zero sensor noise and a perfectly linear slope, the measured temperature will only change once per 20 PID cycles. When that happens, the derivative parameter will blow up, because for that single PID tick where temperature changed by 0.1C in 1/20th of a second, it will act like the temp change slope is 2C/sec. This occurs because the derivative gain is (rightly) renormalized according to the sample rate, see https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/TemperatureControl.cpp#L493. 

Net effect, this behavior completely saturates the PWM output for 1/20th of a second, and then for the next 19 PID cycles, you're back to zero measured temp gain and zero derivative response until the actual temp rises enough to register the next measurement increase at the sensor resolution. That's the zero-noise case: the real case with noisy readings is actually worse. 

You can see this saturate/zero/saturate/zero behavior is happening if you watch the FET LED while heating. It flickers between bright/dim/off like crazy. It should be a constant dim light with whatever intensity corresponds to the PWM duty cycle. 

Yes, you can dramatically lower the PID refresh frequency as a workaround, but then you don't get good responsiveness in the P or I terms. That's a bad solution.

wolfmanjm, 2015-03-11:
Same question here, this will take a significant amount of extra RAM (something we are very short of). So what will we see as an end result? I already get my temps stable to +/- 0.5°C worst case +/- 1°C
Will we see better stabilization? increased heatup time? other than being the "correct" way to do it is there an ultimate goal or improved behavior?

tempo502, 2015-03-11:
Well, the D term seems kind of useless for slow-acting loads right now, because (from the minimal testing I've done) it's either doing nothing or saturating the PWM output, with nothing in between. I don't know if you have any way to test that -- maybe a script to dump all the live PID values to a log file? 

I have to imagine it's putting more low-frequency electrical noise and heating element wear into the system than necessary since it's regularly switching to 255 and 0 for no good reason. You could probably comment out all the D code entirely and get more consistent and reliable performance. (D is usually the least important part of the PID controller, lots of processes use PI control only.)

Aside from the spurious output noise and annoying heater FET LED flickering -- which may or may not matter enough to you guys to be worth fixing -- the heater will respond to disturbances faster if the D term is functioning properly. Say, a nozzle blower kicks on and introduces additional extruder cooling, or the extruder's melt power requirement (extrusion rate) changes significantly when changing print speed. I currently see temp upsets of ~3-5C when this happens, using auto-tuned PID values.

There's probably a way to get creative with the filtering to avoid sucking tons of RAM with a ring buffer, maybe something like this crummy pseudocode to only update the D term once per second:

initialize {
  int d_window = readings_per_second
  int d_counter = 0;
  float d_sum = 0;
}
each PID tick {
  d_sum += (temperature - this->lastInput);
  d_counter++;
  if (d_counter >= d_window){
    d_counter = 0;
    d = d_sum;
  }
  do all the other PID stuff normally, except don't scale d_factor = d / PIDdt 
}

There's probably better ways to do it, but that's just a first-pass thought at avoiding a 20 float array to do the same thing.

nophead, 2015-03-11:
For a cheap smoothing filter for applications like this I use a one term
IIR: ave += (input - ave) / k and make k = 2^n to replace the divide by a
shift. It responds similarly slowly to averaging k samples but only needs
one store.

PetteriAimonen, 2015-03-11:
See here for previous efforts to fix this problem by filtering: https://github.com/Smoothieware/Smoothieware/pull/564

tempo502, 2015-03-11:
Filtering on the temp sensing side is only part of the fix. The problem is that the D term of PID is extremely sensitive to noise while the P and I terms are not very sensitive to noise and really benefit from faster response to real temp changes. If you do all the filtering at the temp sensing level, you lose some responsiveness on the P and I terms, which increases overshoot/oscillation.  So you really want more filtering on the D term than the other two terms. That means doing it in the PID code rather than the temp sensing code.

triffid, 2015-03-12:
@nophead The issue with that algorithm is when you're using integer arithmetic and (input - ave) is less than k. In this case, the delta becomes zero due to integer divide, and the average gets "stuck" and ceases to move. With large k, that can be quite problematic!

nophead, 2015-03-12:
In practice I use the ADC value scaled up by shifting and +1/2 rounding. I.e. +k/2.

tempo502, 2015-03-13:
I'm trying to figure out what exactly the code is doing right now for oversampling... it looks like every time temperature is checked by the PID loop, a new ADC value is loaded into a 32-entry ring buffer and the median value from the buffer is converted to temperature for use by the PID loop. Is that right?

Why is median being used here? It throws away all the data about the extrema, which means the PID loop can't see trend changes until up to half the ring buffer has been cycled through. If I'm understanding the code correctly, that effectively means at 20hz PID refresh there is up to 0.8 seconds of sensing lag, and that delay gets worse if you drop the PID frequency.

A mean filter would register trend changes faster. There are probably lots of other options though.

bgamari, 2015-03-13:
@tempo502 the median filter is used specifically because it is robust against outliers. The ADC on the LPC is quite susceptible to noise, especially on boards with poor ground impedance. When I started using my board I found that the temperature loop couldn't even stay with 5 degrees Celcius of the setpoint. The median filter cleaned this up substantially.

nophead, 2015-03-13:
A hot end can easily change more than a degree in 0.8 of second. This is the reason why my high speed bang-bang tends to be on a par or better than RepRap implementations of PID and why Marlin bang-bang doesn't work.

tempo502, 2015-03-13:
@bgamari According to an external TC I have mounted on my hot end, the median filter is actually dropping quite a bit of real short-term temp variation caused by extrusion rate changes. It's dropping signal, not just noise. If we're going to use such a long buffer, I think it makes more sense to use a mean filter so that, say, 10 consecutive rising/falling measurements will actually be registered in the filter output. 

It also wouldn't be hard to do a hybrid filter, such as drop the n highest/lowest values for noise rejection and then take the mean of the 32-n middle values. That might be a decent compromise.

@nophead Agree on the rate of change, but not the conclusions. For a general-purpose firmware platform, you can't control the end-user's sensing lag. I'm not just talking about sampling delay, I mean thermal conduction into the sensing element. This is a non-trivial length of time -- the classic Semitec 104GT-2 thermistor has a ~0.5-5 second sensing response time constant depending on the quality of thermal coupling. This is a hugely important variable in control theory.

It is quite literally a mathematical impossibility to control a significantly time-lagged process via ANY speed of bang-bang control without output oscillation. (I don't want to get into the frequency domain math, since I haven't worked with it for years, but trust me on the conclusions.) The more time lag between heater action and temp sensing, the more oscillation you get. In fact, this exact bang-bang oscillation behavior is used by the autotune algorithm Marlin, Smoothie, et al utilize. The autotune algorithm puts the hot end in bang-bang control mode, turning the heat on when the temp drops below the setpoint, and turning the heat off when the temp goes above the setpoint. This forces an oscillatory response because sensing lag causes over/under-shoot, and the magnitude and period of the resulting sine-like curve is used to calculate the PID parameters. The autotune method just outright fails if you don't get oscillation from bang-bang control. And that's with Smoothie's high speed operation. The sensing lag and resultant oscillation primarily comes from the physical process itself, not the sampling rate.

So, if you happen to be controlling a process with negligible sensing lag, that's great, and high-speed bang-bang is viable. But there's no way to ensure all Smoothie users have such well-behaved hardware. So it's not a viable option for implementation here.

bgamari, 2015-03-13:
@tempo502 a hybrid filter would be a great way to get the best of both worlds. My point was that there are large outliers from the ADC that need to be accounted.

tempo502, 2015-03-13:
Some quick math to prove the issue being caused by median filtering and the high PID refresh rate on the D term:
My most recent auto-tuned D gain was 95. 
Let's say the typical temp change is either -0.1, 0, 0.1, or 0.2 per PID tick.
d_factor = d / PIDdt = 95 * 20 = 1900
the PID D term is calculated from d_factor * temp change = 1900 * 0.1C = 190
PWM range is 0-255
A typical desired PWM output might be ~50% power, or 127
Here's the resulting PWM output for various temp changes per tick:
-0.2: 127+380=255
-0.1: 127+190=255
0.0: 127-0=127
0.1: 127-190=0
0.2: 128-380=0
See where this is going? For almost any non-zero temp change, the derivative term saturates the PWM output. This happens because the PID loop assumes each single-tick change should be multiplied times the PID refresh frequency. Thankfully, the median filter outputs enough consecutive zero-change ticks for the PI control to work tolerably well despite the broken D term.

It's funny, because median filtering is a good decision, and normalizing the gains to suit the sample rate is a good decision, but if you do both, you screw up the PID controller.

tempo502, 2015-03-13:
@bgamari what kind of noise are we talking about? How "out there" are the outliers? Any idea if it's a bell curve, power law, or just a random spread? It'd be nice to see a chart of unfiltered datapoints to try to get a feel for the distribution.

I'm used to the 14bit thermocouple amp / ADC combo on Mightyboards, which don't seem to be even remotely as prone to sampling noise. I want to suggest some specific PID algorithm improvements, but you can't really optimize the PID algorithm without considering the input noise and sensor behavior.

wolfmanjm, 2015-03-13:
FYI the plan I have discussed with the other developers a few weeks ago, is to change the way we get the readings from the ADC. As the ADC tends to get a large amount of noise which is totally out there (like huge spikes) we will constantly DMA 32 readings to a buffer, we will sort the array, eliminate the top 8 and the bottom 8 and average the middle 16. This both gives us elimination of the worst spikes, and oversamples the readings. The resulting temp will be given to the PID whenever it asks for it. This should help with the temperature accuracy. 
The PID algorithm is a whole different issue, but now you know where the temp comes from and how it is averaged and filtered maybe you can come up with a new PID algorithm that we can implement.

tempo502, 2015-03-13:
Sounds great. How fast will it sample? IE, how much time does 8 reads represent?

bgamari, 2015-03-13:
@tempo502 They are quite extreme outliers, (nearly full-scale, in my experience; substantial departures from the usual Gaussian noise of the ADC) and should be easily removed with weaker filtering than the median. There are a few links describing the issue in #80. I implemented median as I knew it would be easy to understand its effect and simple to write, not because it was the "best" choice.

tempo502, 2015-03-14:
@bgamari Ok, thanks for the detail. Median filtering is a very reasonable approach in general, and "up to 0.8sec" delay isn't all that bad in the scheme of things (versus up to 0.5 sec in other platforms) but we do need to sort out the PID code so it doesn't over-react so much  :-)

tempo502, 2015-03-17:
Draft of new PID algorithm (pseudocode) and auto-tune algorithm (procedure)... very similar, just better  :-)
https://docs.google.com/document/d/16U3f0zCnfF7ZrIULSXhErqUoYj8u6mmENN2EbmX4loA/edit?usp=sharing
Comments wanted, but please go easy on my inability to write code, I'm just an engineer!

tempo502, 2015-04-03:
To make it clear why I think it matters that the PID routines are broken... Here's what one of my Sailfish bots does with an insulated 40W E3Dv6 hot block and "low overshoot" Ziegler-Nichols PID tuning:
[image]
It is printable in 1 minute, has 2C max overshoot, and is fully settled in two minutes.

Using the same PID refresh rate, same PID tuning parameters, and same hot block hardware, Smoothie does this:
[image]

It takes the same minute to reach target temp, overshoots by 35C, and never settles. 

I think this is a very good 1:1 comparison. The only big differences are in how the temp measurement is processed and the PID algorithm is applied. 

Raising the Smoothie PID refresh rate back to the default 20hz and autotuning gives it about 7C overshoot and a permanent 1C oscillation. That's workable, but not particularly good. I tried doing manual ZN tuning to get around the mediocre / overly-aggressive autotune output, and that allows it to settle much better, but the lack of a functioning D term paradoxically causes a lot more overshoot when you make the P tuning less aggressive. So you have to pick between minimizing overshoot or avoiding oscillation around the setpoint.

Basically, the temp filtering and way the D term is being scaled with respect to time combine to significantly de-stabilize the system. If you use high PID refresh rates, the D term is non-functional due to noise amplification and the system is much more oscillatory / prone to overshoot. If you use low PID rates, the temperature median filter adds a huge sensing lag that makes the system more oscillatory / prone to overshoot. Screwed either way. I'm a little surprised it works at all.

triffid, 2015-04-04:
Smoothie's PID works great for me, I see a graph very similar to your
Sailfish one, using P10 I0.02 D100 X128 with a 45w nozzle heater.

Perhaps sailfish and smoothie have slightly different PID implementations,
and you simply haven't tuned the PID for smoothie properly?

tempo502, 2015-04-04:
When I tune it "properly" according to an established industry practice for PID loops (ZN manual tuning) it overshoots like crazy but is extremely stable thereafter. This is probably a combination of needing more integral anti-windup action and not having a particularly functional D term. (Maybe I should play with I_max.) Then when I use autotune, it overshoots by an acceptable amount (7-8C) and then oscillates +/-1 degree instead of settling.

I'm not familiar with the X command -- is X128 reducing your max heater PWM to half? Is that what you have to do to get it to behave in a reasonable manner? Kind of defeats the point of using a high-power heater cartridge, doesn't it?

tempo502, 2015-04-04:
Yeah, if you have to reduce the max heater power to make your PID controller stable, then your PID implementation isn't very good. You should be able to completely handle different heater powers via the P,I,D gain parameters alone. Doubling the heater output and halving the PID gains should give roughly the same control response. (Barring some sensing lag effects that should be small here.) Larger heater = larger process gain = smaller feedback loop gain.

nophead, 2015-04-04:
Is +/-1C consider good with PID?

The funny thing is I can get that with bang-bang if I use an aluminium
heater block, a 20W heater and the transient response is good as well. One
line of code in an interrupt handler!

triffid, 2015-04-04:
X is I_max, not pid_max

tempo502, 2015-04-04:
You should be able to maintain +/- 1x the true sensing resolution with PID when everything is "sitting still". (As in, nominal sensor resolution + noise band). You'll get the exact same accuracy for bang-bang when you have a process suited to it. Specifically, both PID and bang-bang work very well when there is negligible sensing lag. That requires the sensor to accurately respond to process temp variation faster than the real process temp changes. So, bang-bang works great with high heat capacity blocks and small heaters, because the process temp changes slowly relative to the sensing response speed.

If you want to see how a hotend responds to bang-bang control, just run the autotune sequence. The relay autotune technique used by Marlin, Smoothie, etc just runs the hot end on bang-bang control for a while. That's literally all it's doing. When the temp rises above the setpoint, the heater turns off, and when the temp falls below the setpoint, the heater turns on. Systems with significant sensing lag will oscillate because the forcing is out of phase with the sensing, and the autotune will succeed at calculating system parameters from the wave response. Systems with very low sensing lag will be stable under bang-bang control and the autotune will fail. 

Bang-bang is great when you can use it. But you can't always use it. PID is vastly more robust to sensing lag than bang-bang. The integral reacts slowly in order to phase-lag the response past the unstable 180 degree pole, and the derivative predicts the future to phase-advance the response ahead of the 180 degree pole, so it's capable of being stable across a wide range of time-delayed processes. It's possible to tune a PID controller to work effectively with very long sensing lags. (Say >10 seconds for combined dead time plus two sensor time constants.) Bang-bang mathematically cannot be stable when there is too much phase lag between heater action and sensor response. So it comes down to hardware design, specifically ratio of process gain [(heater power - heat loss) / heat capacity] -- to sensing lag.

tempo502, 2015-04-04:
Thanks @triffid, I'll try that, should help my manual tuning work better.

triffid, 2015-04-04:
@nophead I get something like 100°C overshoot with bang-bang, I need a
really strong D term to keep it under control during warmup.

I could nerf the power of my heater, but I like quick warmups, I like it
maintaining temp no matter how fast I stuff plastic into it, and I like it
barely flinching when the fans turn on.

For this I need high (possibly excessive) power, and a tight PID routine :)

tempo502, 2015-04-04:
I think a relevant detail here for discussing overshoot is that the Sailfish PID code bypasses PID entirely when the temp is more than 10C away from the setpoint. It's full on or full off until you get close to the target, which means it doesn't wind up the integral during initial heatup. Now, I don't think 10C is a magic number, and you can achieve the same behavior other ways (like the anti-windup code that doesn't seem to be in use, or reducing i_max to a low number) but the take-away is that letting the integral term rack up during initial heating is going to cause a lot of overshoot.

nophead, 2015-04-04:
If @triffid gets 100C overshoot then if PID is only active 10C below the target he will presumably get at least 90C overshoot. The power needs to be zero when he is at something like target - 100C during warmup to land at target with no overshoot.

tempo502, 2015-04-04:
Yep, you have to pick your bypass band to suit the physical process. The 10C window Makerbot picked is way too narrow for high power heaters. (I ran an ~85 watt heater on my Sailfish CoreXY bot for a while, there was no tuning that could keep it from overshooting on initial heating. I had to set it to 180C, let it overshoot to 210C, then manually switch to the real setpoint so PID would take over.)

It's not hard to algorithmically tune all this stuff. Just measure dead time and heat-up rate when heating starts and you can calculate when to exit bypass mode and start running PID. 

You don't even really need a "bypass" mode, you just need to turn off the integral error accumulator until you get fairly close to the setpoint. Smoothie's anti-windup code should do this, but I haven't figured out how to make it work.

PetteriAimonen, 2015-04-04:
@tempo502 You can enable the windup patch with config "temperature_control.hotend.windup           true".

Also I suggest to compare behaviour with https://github.com/PetteriAimonen/Jerkware and "temperature_control.hotend.oversample_freq  500", as that already improves it a lot. The D term behaviour is not perfect, but much better than current Smoothieware edge.

In my experience the Smoothie's current algorithm works particularly poorly with high heater power (= more lag compared to heating time) combined with high temperatures (= poor measurement resolution).

triffid, 2015-04-05:
@nophead If I'm reading this correctly, @tempo502 isn't suggesting that we
turn off PID until we're within 10°C, we just don't accumulate I,
essentially turning it into a PD loop during warmup.

I think that would work fine.

tempo502, 2015-04-05:
A special bypass mode is kind of superfluous, since the proportional controller completely saturates the output with large errors anyway. Depending on how you have the gains set, PD is always going to apply max (or zero) power when you're far from the setpoint. Say you're using P15 I0 D50 and gaining temp at 3C/second during initial heatup. (This is roughly what my 40w hotends do.) The D term will (if it's working right) subtract 150 pwm units due to the rise rate. The P term will add [15*error_degrees] PWM units. Solving (255+150)/15 = 27C error to saturate the output. The heater is held full-on until you get to [setpoint-27C]. Then it starts tapering down as you get closer.

The typical anti-windup scheme is to define the "proportional band" where the PWM output from the P term alone is between min and max output. Then only accumulate error when inside that band. All normal running operation will be entirely within the proportional band, unless your tuning is horrifically wrong (eg P150), so it's safe to assume that any time outside the proportional band is due to a recent setpoint change. And you don't want to apply the integral term when you make large setpoint changes, because it very rapidly winds up.

There are multiple ways to implement anti-windup schemes. You can bypass PID entirely until you get close to the setpoint, or you can freeze the I term when outside the proportional band, or you can cap the error term fed into the accumulated error sum, such as [acc_error += min(error,2)] to keep the big setpoint-change errors from blowing up the integral term quite so rapidly. Lots of ways to do it.

The optional anti-windup code in Smoothie is ok, but it runs into issues when you have really high P and D gains like the autotune spits out. Say you have P75 Ixx D150. When you're oscillating around the setpoint due to initial heatup overshoot, P*error+D*change_rate is going to be <0 or >255 a large fraction of the time. That keeps the integral term from updating, meaning it's "stuck" on whatever value it happened to be at when the output saturated. That's not great behavior.

nophead, 2015-04-05:
> initial heatup overshoot, P*error+D*change_rate is going to be 255 a

Would it be better to clear the accumulator as well as stop updating it
outside the proportional band?

triffid, 2015-04-05:
@nophead

no, the LPC1768 ADC is very noisy, if we cleared it outside the
proportional band then it would be constantly being zeroed due to noise
excursions, leading to incessant oscillation

nophead, 2015-04-05:
All in all it seems Smoothie is broken firmware on top of broken hardware. Time to bin mine and move on I think.

arthurwolf, 2015-04-05:
> All in all it seems Smoothie is broken firmware on top of broken hardware.

Well that's not very nice ...

> Time to bin mine and move on I think.

nophead, 2015-04-05:
Sorry, didn't mean to rude, but it is very disappointing to find the temperature control works less well than stuff I knocked up of breadboard nearly 8 years ago. Are there any hardware fixes for the full scale ADC errors?

arthurwolf, 2015-04-05:
> Sorry, didn't mean to rude, but it is very disappointing to find the
> temperature control works less well than stuff I knocked up of breadboard
> nearly 8 years ago. Are there any hardware fixes for the full scale ADC
> errors?
In my experience, the noise in the Smoothieboard ADC means that the input
will from time to time read 0 or max_value. We have a filter for this in
the code which works quite well.

Nothing is really -severely- broken here, and the temp control in Smoothie
works very well for many people.

Most users can get their hotend temp swings to bellow 1 degree delta ... as
far as I can tell the main problem we still have to fix, is the initial
overshoot.

Clearly we can make the temp control and PID better ( in particular the
autotune ), but you'll find many people ready to tell you that in their
experience it's better than what exists on 8-bit controllers.
I'm not sure if that's true or if it's just that they were better able to
tune the parameters on their Smoothie ( or luckier ).
What I'm trying to say is things are not -obviously- broken here :)

There is no reason to say the hardware or firmware is broken. It's not
perfect but it works very well for many people ...

PetteriAimonen, 2015-04-05:
I agree that the PID issues are 100% fixable in software, so even though the LPC17xx ADC is a bit noisy, it is not broken beyond hope. The current firmware doesn't suffer from the noise, just from filterless PID (and/or lack of oversample, depending on how you want to think about it).

The bin or not to bin question for me is whether Smoothieware can ever beat machinekit in any way, i.e. maybe everyone should just focus development on it when they are already in the same price range. Smoothie's software development is clearly suffering from the lack of CPU and RAM, which is silly when 1 GHz ARM boards cost less than $50.

tempo502, 2015-04-05:
Everything we've talked about is fixable via software if you tune the code to the application and expected hardware a bit. If an AVR can give good performance, a Smoothieboard can give awesome performance. BUT, that may require some code optimization that de-generalizes the platform. It's the usual Swiss army knife problem -- it gets harder to use and ultimately more limited as you try to cover every possible usage case. 

For example, we can get the current ADC chip filtered appropriately and set up some specific PID and autotune behavior that works really, really well for 3DP, but that might make it less appropriate for future boards with better ADCs or less-common uses. (Like, I don't know, running a reflow oven or something.) It's all about right-sizing the code for the desired application and accepting non-optimal performance elsewhere. 

As for MachineKit... The benefit (and downside) to MK is that it's insanely modular. That makes it awesome for highly competent tinkerers but downright frightening for casual users. I think that will get better over time but there seems to be an insane learning curve right now. It's got flexibility and growth potential out the wazoo, but it's not even remotely stable or mature yet. 

What concerns me is that the MK guys are working from a very different set of paradigms and a VERY different codebase, so relatively little of what's historically been done in 3DP will cross-pollinate easily. There's currently way less 3DP-specific MK development than any other platform. The really smart guys doing the brunt of the dev work are CNC milling and robot control experts, not 3D printer users. That leads to a lot of architecture decisions that aren't optimal for 3DP. For example, Rob's blending trajectory planner can only handle 3-axis control which means Bas's velocity extrusion technique needs to take a lot of extrusion volume control away from the slicer in order to run the E axis. That's not necessarily bad, but it's very different from the established toolchain concept decisions that evolved in the community over the years. The MK guys are coming from a different place and not (in my opinion) paying a lot of attention to established conventions that are working well.

arthurwolf, 2015-06-21:
Closing this. Feel free to re-open it if you want to further discuss it, or even better, have some code.

closed 2015-06-21 by arthurwolf