Smoothieware/Smoothieware #640: PID integral error term should be allowed to be negative

tempo502, 2015-03-10:
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/TemperatureControl.cpp#L455

The integral term's accumulated error value should be clamped between +/- i_max, not 0 and +i_max. Integral error will USUALLY be positive for simple heater control, but that's not always necessarily true.

(I haven't looked at spindle control, but this may apply there too.)

wolfmanjm, 2015-03-10:
the pid output o must be between 0 and 255 (or i_max) so I think clamping it to 0 is probably correct. If it gored negative then we could get a negative o which would be bad.

wolfmanjm, 2015-03-10:
@triffid ? what do you think?

tempo502, 2015-03-10:
You want all three terms to be able to be positive or negative, then later clamp the final PWM output to a valid value. Example scenario: someone sets a high I gain for a system with long measurement lag (eg HBP), and temp overshoots for an extended period on initial heatup. Obviously the sustained overshoot error will drive the P term negative and start decreasing the accumulated I term. This overshoot will drive down the accumulated I term until it's also negative, if the overshoot condition lasts long enough. So PWM output rightfully goes to zero until the measured process temp eventually drops below the setpoint. THAT gives you a positive error for a positive P term, and you still have a negative accumulated error giving a negative I term. In this case, it's possible to have a valid negative accumulated error that results in a positive PWM output. 

If you clamp the I term to be >=0, you skew the feedback behavior to be biased to higher average heater output. Not the end of the world, but not optimal.

wolfmanjm, 2015-03-10:
Our Pid is based on Bretts pid library. He clamps the iterm to the min output which in our case is 0.

https://github.com/br3ttb/Arduino-PID-Library/blob/master/PID_v1/PID_v1.cpp#L59

are you saying his library is incorrect?

tempo502, 2015-03-10:
Yes, Brett's implementation is pretty mediocre. I was really surprised by the poor algorithm quality when I first saw it, eh, a year ago when I was reviewing Sailfish code. (Sailfish inherited it unchanged from Makerbot, who reworked Brett's code and added more errors.) 

But then, I'm a chemical engineer with training in process control and industrial PID tuning, so I tend to have high standards. I don't think most people would notice the difference.

wolfmanjm, 2015-03-11:
ok so what end effect will we see if this is corrected? ie better temp regulation? faster heat up time?
(I am not a pid expert, and hence why I copied Bretts code verbatim).

tempo502, 2015-03-11:
You and everybody else copied Brett's code, apparently  :-)

From an academic standpoint, it violates the assumptions used to derive the Ziegler-Nichols tuning parameter calculations for quarter-wave decay response. These parameters are used by the auto-tune code. (see https://github.com/Smoothieware/Smoothieware/blob/43215feb346ac1f8b4027650dd774bbccec5c96a/src/modules/tools/temperaturecontrol/PID_Autotuner.cpp#L247)
Which is a fancy way of saying the settling behavior after a setpoint change is less symmetrical and probably takes somewhat longer to stabilize. Subsequent overshoots after the first may be larger than they need to be. The settling curve will likely bias high for a while, rather than being symmetrical around the setpoint.

Because Brett's code seems to be most often used for heater control, which is 1) generally pretty tolerant of overshoot/oscillation and 2) has highly asymmetrical output response to begin with (heating faster than cooling), it's really not a very high-impact algorithm error. Not high priority thing to change if you don't feel like working on the heater control code.

Practical result of the proposed change? Depends on user-configured gain parameters and hardware configuration. Might not be noticeable at all. Might produce, say, 10% faster settling time after large setpoint changes.

triffid, 2015-03-11:
As you say, our highly asymmetrical output is something that PID isn't really designed for, and breaks some basic assumptions of the algorithm.

I haven't done any tests, but I think I understand PID fairly well and I feel that capping I_min to zero would actually improve response to large setpoint changes, since P and D will be able to pick up the output earlier after an overshoot.

Indeed, if the output were more symmetric it would be preferable to cap the I accumulator to -I_max.

I'd be interested to see some empirical results if you care to make some.

tempo502, 2015-03-12:
A common modification to PID to cope with asymmetrical processes is to use asymmetrical gains. You simply use separate P,I,D gains for overshoot / rising slope versus undershoot / falling slope. This easily accounts for the rate difference between active heating and passive cooking. Downside is, it makes initial tuning more difficult since common oscillatory tuning methods (chiefly ZN and the autotune algorithm Smoothie/everybody uses) don't work anymore. 

Related SI denote, the PID autotune method Smoothie uses is completely inappropriate for asymmetrical processes. It gives poor results when the heating rate is much larger than the cooling rate. This is one reason why so many Marlin/Repetier users recommend dropping max PWM output to get better PID stability -- gimping the heater power makes the process more symmetrical so the autotune method used is able to perform tolerably well. (Overall performance is actually diminished versus a more appropriate PID tuning method, but the autotuned output is so bad at full heater power that gimping the heater gives better net results.) An automated implementation of ZN would be a lot better.

tempo502, 2015-03-12:
Passive cooking... Heh. Hopefully none of that happening :-) Meant to say passive cooling.

arthurwolf, 2015-06-21:
This is unfortunately way over my head. 
I'm closing this as this kind of "description of a solution" is not really useful unless somebody is capable of turning it into actual code.
Anybody with some code or a simpler explanation is welcome to re-open this.

closed 2015-06-21 by arthurwolf

tempo502, 2015-06-22:
I did actually write an algorithm to fix this, along with issue #639 (and a bunch of other shortcomings of the Brett Beauregard PID code everybody in the 3DP space is using). Jim said he would implement the algorithm but I'm guessing everybody forgot about it by now. See: https://groups.google.com/d/msg/smoothieware-support/1W3ww9LIDlw/jszrLC2r7HkJ