Smoothieware/Smoothieware #642: G4 dwell bug in edge vs master

mcgyvr, 2015-03-19:
Bug in current Edge build for how dwells are handled.. 

Example code that works correct in the current Master firmware. 
M8 G4 P1000 ; turn on switch/Mosfet tied to M8 command and then dwell for 1 second
M9 G4 P1000 ; turn off switch/Mosfet tied to M9 command and then dwell for 1 second.

In current edge firmware  the dwell happens "before" the M8/M9 switch is turned on/off..
So in edge when it comes across that it will dwell for 1 second..THEN turn mosfet on.. Then dwell for 1 second THEN turn mosfet off.. This is incorrect.

wolfmanjm, 2015-03-19:
Try this firmware...

http://blog.wolfman.com/files/firmware.bin

If it works let me know and I'll merge it

mcgyvr, 2015-03-20:
Jim, 
That fixed it.. Perfect.. 
Close and comment....

closed 2015-03-20 by mcgyvr