Smoothieware/Smoothieware #646: Heat and wait gcode bypassed by disconnected thermistor

tempo502, 2015-03-23:
I just tried to print a file with my thermistors disconnected. This file usually takes a significant period of time to preheat before starting. To my surprise, with the thermistors disconnected and the LCD reading 999C, the machine immediately sprang into action and started trying to print.

arthurwolf, 2015-03-23:
Could you reproduce this and tell us what M105 returns ?
Thanks.

tempo502, 2015-03-23:
I'm running prints from the LCD / external SD card. Give me time to find a USB cable, the roomba ate my last one...

wolfmanjm, 2015-03-23:
Actually I know what the problem is, the point is what we want to happen there.
The disconnected thermistor makes the read temp INF, but the target temp is set to UNDEFINED which is -1, hence that M109 test passes. Setting UNDEFINED temp to INF would make M109 hang indefinitely.
Not sure that is what we want.

tempo502, 2015-03-24:
My personal preference is that it refuse to play files and/or abort active commands while the sensor is disconnected. That would be in line with other popular platforms.

tempo502, 2015-03-24:
If the "abort active commands" route is chosen, it would be reasonable to wait a certain number of updates/seconds first in case the issue is an intermittent wire break.

commit 1c6c41dc (2015-03-24): Merge pull request #647 from wolfmanjm/upstreamedge

Fix #646
commit 487976f7 (2015-03-24): Fix #646
If M109 or heat and wait and thermistor reading is unreliable then the system is put into a HALT state.
Compliant hosts will comply with this and abort the current print.
closed 2015-03-24 by wolfmanjm