Smoothieware/Smoothieware #649: [ZGrid] Several bugs/Todos

wolfmanjm, 2015-03-31:
1. needs to support bots that have 0,0 as center of bed, so -X an -Y need to be accomodated
2. while(!zprobe->getProbeStatus()); <---- This is a big NONO, must call ON_IDLE this will kill USB.
3. Support circular beds
4. Initial move after homing needs to be fast rate not slow rate (deltas have a long way to move).
5. the initial positioning for probe adding/removal needs to be optional.

sorki, 2015-03-31:
> while(!zprobe->getProbeStatus()); <---- This is a big NONO, must call ON_IDLE this will kill USB.

This doesn't even work for me for some reason and the probe attach/detach routine should be implemented elsewhere - preferably configurable like  "Pause when out of filament" is implemented.

Is there a G-code that would wait for user pressing a specified button?

> Support circular beds

This should be relatively easy to implement with masking/skipping parts of the grid that are outside of the delta radius.

I'll add one more issue - this feature would be extremely useful for PCB milling but it has to support small dimensions (e.g. 3x3cm). According to testing of one user on the IRC this doesn't work as expected and results in weird behaviour. For this to be convenient for users it should be possible to specify dimensions to be probed with G-code params.

Leveling in this case also seems to depend on line segments per second which is not enabled on cartesians by default.

qharley, 2015-03-31:
The list is growing fast! ;-)

qharley, 2015-04-07:
I am working on some of these issues, but I do not have a Delta.  Round bed and middle zero will be a best guess fix if I attempt it.

wolfmanjm, 2015-04-07:
I think there are several people who can test on a delta. I certainly can.

qharley, 2015-04-08:
I will take you up on that offer.

barsamgr, 2015-07-17:
Hello
I noticed to bugs in the Zgrid code. I have corrected them in my own fork, but the correction is not in a separate brach.
1. (Not definitely a bug, but my opinion) Z should move to zprobe->getProbeHeight() and not to std::get<Z_AXIS>(this->probe_offsets) + zprobe->getProbeHeight(). It is supposed that Probe Height is defined as a safe distance to clear the probe movement.
2. This is surely a bug.  In line 507 the code is  float z = 5.0f - zprobe->probeDistance((this->cal[X_AXIS] + this->cal_offset_x)-std::get<X_AXIS>(this->probe_offsets). 5.0f should change to std::get<Z_AXIS>(this->probe_offsets) + zprobe->getProbeHeight().

After I did the changes, the grid is working just fine.

arthurwolf, 2015-07-17:
Hey.

Would you mind making one pull request for each change, so that each can be discussed and evaluated separately ?

Cheers.

wolfmanjm, 2015-07-17:
I think both are valid.
@qharley needs to look at it to see if it affects his platform, and make the fix appropriately.

qharley, 2015-07-17:
I will make the changes. Thanks

qharley, 2015-07-17:
The above breaks my machine...

The 5.0f is not the probe height.  It is the extra bit added beyond the
probe so that it starts a short distance away from the bed.  The reason why
I did not just add this to the probe length before hand is that the probe
code calculates the M206 Z homing offset using the probe length, and will
calculate an incorrect offset.

Probe Height is the actual probe length.  Should I add another configurable
value for the safe distance for probe movement, in stead of using 5.0F?

Tell me about how your machine behaved when it did not work for you.
Perhaps I need to make the documentation clearer.

qharley, 2015-07-17:
Looked at it again and it needs some work...  I just want to repair it in a
way that will not break machines in the field.

barsamgr, 2015-07-17:
Ok. I will try to explain it as simple as possible.
We have 2 things defined in the configuration.
a. Probe Height. Distance from 0 where it should start probing. In other words, safe distance for the probe to deploy in case you have an auto probe with servo ( Like me). In my configuration I have a value of 5 because is the minimum distance I need from the bed to deploy the probe.
b. Probe offset. Which defines where the probe is located in space when triggered, according to the nozzle.  For instance mine is X-26.0, Y-8.0, Z-2.9.

So before the probing process is started, the head must move to the "safe" probe height position ie. 5.
In the code you have "std::get(this->probe_offsets) + zprobe->getProbeHeight()" which gives a value of 2.1 since (-2.9 + 5 in my case). This is wrong. The probe has to go to Z position 5 so "zprobe->getProbeHeight()".
Now the probing process calculates the travel distance of each point which is not the same for each machine. You are assuming  that the nozzle is at Z position 5 that's because yours is there. The correct positions of the nozzle when the probing starts is the probe height (5) plus the Z Probe offset (-2.9).
I suppose that you have it like this because your true probe is your nozzle, but it is not always the case.

If you do as I say and you define your Z offset to be 0 in your configuration, it will have the same effect as it is your code now.

I hope it is clear now.

qharley, 2015-07-17:
Thanks.  That makes sense.

Your probe offset cannot be negative for Z: that would mean your nozzle is
below the probed surface when the probe hits.
My probe attach to the nozzle, and have an offset of 0,0,14.5 for
instance.  The probe hits with the nozzle 14.5mm above the bed.

That said, I am going to dissect the code and clean it up a bit.  It
changed a lot over the year, and could be a lot more economical.

  [image: photo]
*Quentin Harley*
Founder, Reprap Morgan 3D
 p:+27120041883 | e:quentin@reprapmorgan.com | w:www.reprapmorgan.com | a:4
Burger Ave, Lyttelton Manor, Centurion

barsamgr, 2015-07-17:
That is where you are wrong...
It is called an offset. Look at the 3D space we have Z+ and Z-. 
Z+ is above the nozzle and Z- is below the nozzle as it is for the axis left is -X right is +X etc.

qharley, 2015-07-17:
Perhaps I should have called it nozzle offset then.
The way I wrote the code is in line with the movement space of the nozzle.

barsamgr, 2015-07-17:
By the way... It is not my theory.
Every firmware works like this.
I have played a lot with both Marlin and Repetier. So I am not playing it smart, I just want to help.

qharley, 2015-07-17:
I really do not want to break working machines in the field though

barsamgr, 2015-07-17:
For instance Marlin is using the probe for homing too. Something that is unfortunately missing from Smothie. I had to install a separate endstop for homing. But I will try to change it a some point and make it work like Marlin does.

qharley, 2015-07-17:
Just had a look at the Marlin instructions.  I see what you mean.

For sake of uniformity I could change it... I could add something to make
sure it does not break current systems.

qharley, 2015-07-17:
In my defence... My original morgan leveling routine predates mainline
marlin probe code... ;-)  A pity I did not know how to use git back then...

barsamgr, 2015-07-17:
I understand what you mean. I am a developer my self too and I started developing with basic on a spectrum 48K, but some things are already there and working. We don;t have to invent the wheel once again.  Marlin is a very mature and nice firmware. The only problem is that it is limited to 8bit controllers.  This was the reason I changed to smoothie.  I am missing a lot of things from Marlin, but at least I have the speed I wanted now.

sorki, 2015-07-17:
Backwards compatibility shouldn't block fixes or improvements. If there's a release cycle backward incompatible changes can be tracked and documented easily.

qharley, 2015-07-17:
Of course not.  But improvements should not break systems working properly

  [image: photo]
*Quentin Harley*
Founder, Reprap Morgan 3D
 p:+27120041883 | e:quentin@reprapmorgan.com | w:www.reprapmorgan.com | a:4
Burger Ave, Lyttelton Manor, Centurion

wolfmanjm, 2015-07-17:
Actually I am not in favor of copying Marlin on this. We did a lot of our zprobe before Marlin did and I do not see any reason to play catch up with Marlin whenever they change something. Plus I do not agree with many of the decisions made in Marlin.

Lets fix this but not copy the Marlin does things just because Marlin does it that way. and the Zprobe stuff in Marlin is NOT mature or stable. When I wrote the original Zprobe stuff for SMoothie Marlins Z probe did not even work. So just becuase Marlin does something a certain way is not a reason to change Smoothie.

wolfmanjm, 2015-07-17:
Now to clarify...  zprobe->getProbeHeight()  is the height above the bed the Probe starts to probe, so it is the Z to which the initial move travels to.

Probe Offsets are the offsets the probe is from where the nozzle is. (This is not needed in delta calibration but is used in the three point level). This definition needs to be consistent between the leveling strategies.

wolfmanjm, 2015-07-17:
Backwards compatibility should be maintained if at all possible. It is NOT ok to just break things for existing users with existing configs.

The rule for Smoothie is the rule of least surprises.. If a user upgrades to the latest edge they should not be surprised by something that was working now not working.

qharley, 2015-07-17:
How is probe offset defined in three point?

qharley, 2015-07-17:
Do you set Z as negative or positive in Three point is what I mean...

wolfmanjm, 2015-07-17:
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/zprobe/ThreePointStrategy.cpp#L263

wolfmanjm, 2015-07-17:
If I recall the probe offsets are the offsets from the nozzle, so to be consistent +x would be the probe is to the right. +Z would be the probe is below the nozzle by +Z mm
But if you have it reversed it is ok we will just have to live with the inconsistency.

qharley, 2015-07-18:
This is the way I have it.  Remember that I modelled ZGrid from three point
initially.  Just wanted to see if I kept that in tact.

We could perhaps have a small change to allow the topsy turvy way:
leveling-strategy.ZGrid-leveling.marlin-offset           false  // Set to
true to have marlin style probe offsets (Default false)

barsamgr, 2015-07-18:
Out of curiosity... since when something below is called +Z?
I am asking because in the CNC world when ever you go below it is called -Z.

wolfmanjm, 2015-07-18:
depends on how you look at it, it could be the height of the nozzle above the probe. touchless probes are above the nozzle, so really it is not really intuitive either way.

wolfmanjm, 2015-07-18:
I am no tin favor of having marlin hacks in the config :)

barsamgr, 2015-07-18:
I am not a marlin hack, I am just a practical guy that I like to do my job as easy as possible and the proper way. I have invested a lot of money to get the smoothieboards because I liked the idea of it. For sure I expect the proper support for the firmware. 
For the laser machine that I installed one of the boards, I am more than happy. But when it comes to 3D printing... sorry but it is far from reliable.
If you start doing things your way, just because you don't like people talking about Marlin, that's fine. But have in mind that other boards are coming in the market that are 32bit and are supported from both Marlin and Repetier (ie. RADDS).
I have 4 3D printers (2 delta and 2 cartesians), a laser engraver, a CNC router and a CNC lathe.
All DIY. I have learned a lot about CNC machines the last 10 years. So I am not just a newbie trying to be smart. I just like thing to be done the proper way. I can keep doing things in my own fork and don't bother any more, if that's what you want for the "Marlin hacks".

arthurwolf, 2015-07-18:
Ardash I don't think wolfmanjm was talking about you when he said "marlin
hacks", but he was refering to actualy "hacks" in the config, not any
person.

arthurwolf, 2015-07-18:
@barsamgr Just to be clear : when wolfmanjm says "I am no tin favor of having marlin hacks in the config :)" he is saying that he is not in favor of modifying the way Smoothie is configured in a less than optimal way ( which is what hack means here ) to accomodate compatibility with what Marlin is doing.
I really don't think there was any attack intended your way.

We do try to be compatible with marlin when it makes sense. But not at the cost of making Smoothie less usable to our many users that have not used Marlin before and therfore should not suffer for it.

barsamgr, 2015-07-18:
No one said anything about modifying smoothie in a non proper way. But having probe offset being in the correct X and Y direction but Z at the opposite direction, is far from proper and optimal. When we define a point in space that point has very specific coordinates. This is something that no one can argue with.
So either that point (trigger point of probe) is defined as it should be, or call it something else.
My whole argument was about the proper definition of the parameter called Probe Offset. Nothing more, nothing less. I never asked to change something the Marlin way which is not optimal or proper.
Now, trying to convince me that it is done the proper way just because you don't want to upset all the users that are using it the non proper way, it's not professional.
I am a developer my self too and I work as a freelancer for very large organizations. If the customer ask me to fix something, while it was done the wrong way, and try to argue with him telling that it was done this way since the begging, and that the users have learned it this way, I would got fired the same moment.
So accepting mistakes is not something terrible, trying to cover them on the other hand... it is.

wolfmanjm, 2015-07-18:
@barsamgr I was referring to @qharley's suggestion of adding a ```leveling-strategy.ZGrid-leveling.marlin-offset``` to the config, and in no way was referring to you as a Marlin hack. That is not an insult I would throw at anyone.
As for the probe offset, think of it as the height of the nozzle above the probe if you prefer. However it is unacceptable to make changes to the config that are not backward compatible and would break someones machine if they upgraded to the latest firmware. It is unfortunate that the definition appears to be the reverse of what you want. No one else has complained or mentioned it when the patch was up for review. So it is a shame it slipped through, and to reiterate this was done BEFORE Marlin did it in a way that actually worked, so comparing to Marlin is irrelevant IMO in this case.

qharley, 2015-07-18:
I love helping everyone out, but in the end my I am still trying to run a business, and will not intentionally sabotage by own operations.  I freely contribute any code that I may write to Smoothieware in the spirit of the open nature of the project, and for the good of all using it.  In the end my customers are more important to me than keeping absolutely everyone happy.  I hope this is acceptable.

Peace,
Q

barsamgr, 2015-07-18:
@wolfmanjm Of course I can think of the nozzle being above the probe of n Distance, but I don't like the inconsistency. X and Y are defined according to the nozzle, but Z is according to the probe.
And no one complained about this...
Anyway. If it's just me, I have it the way I want it anyway at my fork.
And by the way, since we are talking about

wolfmanjm, 2015-07-28:
The WIki and the comments need to be updated to reflect the new config options. The comments are also inconsistent with some of the config settings.

BartyBorris, 2015-08-10:
I am very new to smoothie and c++ so please be patient...

I was getting odd behaviour when I change the x and y grid resolution in the config file from the default 5 x 5.  I believe this is due to the int(this->cal[X_AXIS] / this->bed_div_x) typecast below resulting in "random" incorrect identification of an odd/even row:

void ZGridStrategy::next_cal(void){
    if (int(this->cal[X_AXIS] / this->bed_div_x) % 2 != 0){  // Odd row
    ...

The above truncation is not reliable, I think, due to the potential for a slight "random" error in the division (for example if this->cal[X_AXIS] / this->bed_div_x results in 3.9999999... it would be interpreted as odd, when it is even).  I fixed this by rounding before the typecast to int:
    ....
    if (int(round(this->cal[X_AXIS] / this->bed_div_x)) % 2 != 0){  // Odd row
    ....

With this change all seems to work regardless of the grid resolution I specify.

Thanks for your work qharley.  I hope this helps, and again, I am very new to all this (only a week in learning c++), so I apologize if I missed something obvious here.

wolfmanjm, 2015-08-10:
actually int(...) is not a cast and is not really a supported function, it is an arduino ism I believe.

a cast is (int) ( expression)

This should be removed... the correct line would be...
```
if (((int)roundf(this->cal[X_AXIS] / this->bed_div_x)) & 1 != 0){ // Odd row
```

using a MOD is a waste of CPU cycles.

BartyBorris, 2015-08-10:
Thankyou

qharley, 2015-08-11:
Thank you BartyBorris,

Jim is the real programmer... I am a self taught coder.  I think that line
slipped through from the original code I wrote for my fork of Marin (on
arduino)
Will fix and test, and do another pull request later today.

BartyBorris, 2015-08-18:
In addition to the change describe by in my post and wolfmanjm's improved version, I noticed something else that needs to be fixed in the same function (for the even rows):

if (this->cal[Y_AXIS] > (this->bed_y)){

needs to be changed to something like

if (this->cal[Y_AXIS] > (this->bed_y + tol)){

where I defined tol to be 0.1 mm (I know magic constants are bad, but this was a quick fix)  to deal with the buildup of errors from this->cal[Y_AXIS] += bed_div_y;

Before this change I noticed the following which caused problems:

DEBUG: Y (230.000015) > cond (230.000000)

Clearly the intent was for this condition to be false, but the build up of errors from the successive additions of bed_div_y caused this to not be exactly equal (bed_div_y is not necessarily an integer multiple of bed_y... it's a float) resulting in this being true and hence skipping of a location to probe causing multiple issues.

qharley, 2015-08-18:
I will take a look.  Seems simple enough, especially if you make the magic
constant half bed_div_y

[image: photo]
*Quentin Harley*
Founder, Reprap Morgan 3D
p:+27120041883 | e:quentin@reprapmorgan.com | w:www.reprapmorgan.com | a:4
Burger Ave, Lyttelton Manor, Centurion

wolfmanjm, 2015-08-18:
This is not a buildup of errors this is normal IEEE floating point inaccuracies. Floats are NEVER exact..

BartyBorris, 2015-08-19:
Yes, with the exception of special cases. The addition just makes things worse, but is not the source. I am also noticing that the grid does not save correctly at times.  I will comment when I have time to work on reproducing the error consistently. I am testing on a Morgan by the way...

qharley, 2015-08-19:
What is the size of the grid you are using on the Morgan, and the bed size?

[image: photo]
*Quentin Harley*
Founder, Reprap Morgan 3D
p:+27120041883 | e:quentin@reprapmorgan.com | w:www.reprapmorgan.com | a:4
Burger Ave, Lyttelton Manor, Centurion

BartyBorris, 2015-08-19:
400 x 230mm bed (it's a modified Morgan design).  Problems seem to occur with "fine" grids, say 20 x 10 and finer.  I will try to find some time later this week to test and provide details. I am not convinced it's entirely related to grid resolution.

sorki, 2015-08-25:
I was able to trigger some of these issues when trying to use ZGrid for PCB probing (probing 50x50mm pcb) - one row probed twice, then head crash at another point (I've used spring instead of tool directly not to damage anything).

BartyBorris, 2015-08-25:
After making a few minor changes (a few more than described here...) to deal with finite machine precision, I have not had any issues on my end (aside from the grid not saving correctly with M374 in some cases).  I just reviewed Quentin's updates from a few days ago (see latest ZGridStrategy.cpp), and it appears consistent with the changes I made.  Obvious question... are you sure you are running the latest?

BartyBorris, 2015-08-25:
Regarding not saving the Zgrid properly with M374.  The following grid resolution "appears" to be saving correctly consistently:

leveling-strategy.ZGrid-leveling.rows           15            # X divisions (Default 5)
leveling-strategy.ZGrid-leveling.cols           11            # Y divisions (Default 5)

However, the following consistently does not save correctly:

leveling-strategy.ZGrid-leveling.rows           16            # X divisions (Default 5)
leveling-strategy.ZGrid-leveling.cols           11            # Y divisions (Default 5)

By consistent, I mean three tries in a row.  I can't even open the file.  It is recognized as a binary.  If I try to open it as a text file, no dice.  Hope this helps.

qharley, 2015-08-25:
Will take a look.

sorki, 2015-08-28:
@BartyBorris my observations were prior float fixes. I'll retest with the latest edge.

tehfailsafe, 2015-10-18:
Coming from a kossel build running marlin on ramps, my second printer is a larger scaled delta using the azteeg x5 mini and smoothieware. I have the same FSR setup mounted under the glass with an FSR controller board installed. 

According to the docs, I can use the zgrid leveling strategy on a delta, but I'm not sure of the config settings to make it work without a 200x200 bed.

The following config makes the hotend go way off the bed and I am a bit nervous to try random sizes in there.  How do I define the config for a delta with a 120mm circular radius?

```
leveling-strategy.ZGrid-leveling.bed_x           200
leveling-strategy.ZGrid-leveling.bed_y           200
```

bilsch, 2015-12-03:
@qharley have you had time to work on this yet? I tried running the automated ZGrid and have found it to not work properly post calibration.

My sequence of activities:

1. Home all axes
1. Center the nozzle on the bed
  * paper test as usual
1. Home all axes again
1. G32
  * system homes, then goes to center of X/Y and lifts Z by 5mm
1. Attach zprobe
  * trigger probe manually to start automated probe
1. G31 to view [grid results](https://github.com/bilsch/3d_printer_configs/blob/ZGrid/Smoothieware/ZGrid_Notes#L7-L16)
1. M500 to save results 
  * creates [ZGrid](https://github.com/bilsch/3d_printer_configs/blob/ZGrid/Smoothieware/mm2.Zgrid)

So far, attempts to print post ZGrid leveling have resulted in very bad results. After homing X/Y/Z the Z axis lowers by some amount ( I did not see anything on the LCD after multiple attempts ) burying the nozzle in to the plate - thankfully not breaking boro! Seems the extruder mount is preventing breakage.

When asking in the IRC channel it was suggested I open an issue. Since this is actually related to this existing issue I thought it more appropriate to brain dump here. Sorry if it is verbose, maybe something will jump out.

My [config](https://github.com/bilsch/3d_printer_configs/blob/ZGrid/Smoothieware/mm2.config#L176-L190) may also be helpful.

tehfailsafe, 2015-12-03:
>    2. Center the nozzle on the bed
>    3. Home all axes again
>    4. G32
>    5. Attach zprobe
>    6. G31 to view grid results
>    7. M500 to save results

This happened to me as well, but it did break the glass after storing the
z-grid...

Any idea what would cause that?

[22 more replies]

closed 2016-05-30 by wolfmanjm