Smoothieware/Smoothieware #655: wrong config option in example config

Bouni, 2015-04-13:
Hi,

there is an error in the [config example](https://github.com/Smoothieware/Smoothieware/blob/edge/ConfigSamples/Smoothieboard/config#L198).

The example says `switch.spindle.enable` but the [code expects](https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/spindle/Spindle.cpp#L27) `spindle_enable`.

Should i submit a pull request for this minor change?

arthurwolf, 2015-04-13:
It's not an error.
Using a switch module, or using the spindle module, are two different ways
of controlling a spindle, both valid.

But it looks like the spindle module doesn't have an example in the config
files, so that should probably be fixed.

Bouni, 2015-04-13:
Ah, ok, then i got that wrong. But it would make sense to provide a mill/router config example anyway!

closed 2015-04-13 by Bouni