Smoothieware/Smoothieware #656: feature request GCode multi-directory

flety, 2015-04-14:
Add a directory to store GCode.
Possible tree support.

qharley, 2015-04-14:
Having a separate optional folder for config cannot hurt either.  It is too
easy to wipe the config files when cleaning the SD from old gcodes

  [image: photo]
*Quentin Harley*
Founder, Reprap Morgan 3D Printers
 p:012 004 1883 | m:082 782 3902 | e:quentin@reprapmorgan.com | w:
www.reprapmorgan.com | a:4 Burger Ave, Lyttelton Manor, Centurion, South
Africa

wolfmanjm, 2015-04-14:
mkdir /sd/gcode
cd /sd/gcode

We already do that...

closed 2015-04-14 by wolfmanjm

arthurwolf, 2015-04-14:
@qharley if you want your config file to be safe, I recommend you store your g-codes into a /gcodes/ folder ( like @flety asked for, which we already have ).

flety, 2015-04-14:
We can't read the file in the gcode directory via the web interface. (I know that we will one day switch to Mousse but...)