Smoothieware/Smoothieware #660: Mac dot files "._" on display

Veloce7, 2015-04-25:
When copying gcode files from a Mac to a micro SD card to print from a smoothie compatible display, two files show up on the display. The unnecessary one begins with "._". Can these files be hidden from displaying?

Thanks,
Mark

wolfmanjm, 2015-04-25:
they already are, try the latest firmware.

closed 2015-04-25 by wolfmanjm

Veloce7, 2015-04-25:
I’m pretty sure I’m running the latest FW from two months ago. I’ll double check

Thanks,
Mark

triffid, 2015-04-25:
On 25 April 2015 at 15:26, Veloce7 <notifications@github.com> wrote:

> from two months ago.
>

In a project that moves as fast as smoothie, that's barely recent, let
alone latest :P

WZ9V, 2015-04-25:
Yes but it is the latest that has been put into the firmware directory.

Veloce7, 2015-04-25:
Is there another place I should be looking to find more recent firmware?

triffid, 2015-04-25:
http://triffid-hunter.no-ip.info/Smoothie-edge.html

Veloce7, 2015-04-26:
Thanks Michael