Smoothieware/Smoothieware #668: Extruder feedrate ignored

lars-danielis, 2015-05-29:
Feedrate of extruder is ignored in case that no other motor is moving

wolfmanjm, 2015-05-29:
that is incorrect. It works exactly as expected. If there is some misunderstanding please explain.

closed 2015-05-29 by wolfmanjm

lars-danielis, 2015-05-29:
As discussed here: http://smoothieware.org/forum/t-1206226/extruder-default-and-max-speeds
for me it is not working exactly as expected with the latest edge binaries.
There is no difference in the extruder speed between G1 E5 F10 or G1 E5 F100.
The extruder will move in this two cases at the same much too fast speed. I dont know what speed it is but it seems to be somewhat around 300.
With a binary of the master branch it is working as expected.

wolfmanjm, 2015-05-29:
Maybe you settings are being overriden by something in the config-overide? check M503
Having tested it here it seems to work fine. Looking at the code it also looks fine.
Also have you tried this by using a direct gcode command? or only through the extrude command on a host?

lars-danielis, 2015-05-29:
config-override is empty.
Direct gcode command? Not clear what you mean.
I do it with the Terminal in Octoprint. So for my understanding that is a direct gcode command.
It is printing all fine at 40. Only in the case that no axis is moving the extruder is to fast.
In the following case the extruder is at the expected feedrate
G28 X
G91
G1 X100 E50 F40
But in this other case the extruder is approx 5 to 10 times to fast
G91
G1 E50 F40

wolfmanjm, 2015-05-29:
Those are two completely different cases.
The first is a print command and feedrate is not applied to the extruder it takes the speed from the print head move. This is exactly how it needs to operate.

The second case is an extrude and will extrude at 40mm/min. and seems to work as expected for me.

A direct command would be to type G1 E50 F40 in the octoprint command window instead of clicking the extrude button.

wolfmanjm, 2015-05-29:
Maybe you need to reduce your extruders acceleration?

wolfmanjm, 2015-05-29:
FYI the problem in the thread you mention is a different issue, in that case he is using an old invalid config command to set the default feedrate for an extruder. However default feedrate is only used directly after a reset if no Fxxx has ever been specified.

arthurwolf, 2015-05-29:
@lars-danielis 

G1 X100 E50 F40 moves the X axis at 40mm/minute, and the E axis "follows" whatever speed the X axis is moved at while extruding 50mm

G1 E50 F40 moves the E axis at 40mm/minute

Both are completely different and will result in different speeds for the E axis.

lars-danielis, 2015-05-29:
OK so then it is done by a direct gcode in the command window. I am not clicking a button. 
And also understood that I was not clear enough in the thread - sorry. My english is not the best. I am german.
Clear so far, there are two cases. The first case is wihout any issue.
In the second I was never able to change the speed. But I did exactly the same with a master branch. And there I was able to change the speed for the case 2.

I have also understood that the resulting speeds are in the two cases different. But in case two there is never a change. I tried values of 1, 10, 100 and it was always to fast.

wolfmanjm, 2015-05-29:
Ok so if I understand it you can do  ```G1 E50 F1``` and ```G1 E50 F100``` and the speed is exactly the same?

lars-danielis, 2015-05-29:
YES

lars-danielis, 2015-05-29:
and if I reflash smoothie with a binary from the master branch the speed is different

wolfmanjm, 2015-05-29:
one should take 50 seconds the other 0.5 seconds... how long does it actually take? Are you sure you have the correct steps/mm? can you copy your config and results from M503 to pastebin or gist?

wolfmanjm, 2015-05-29:
Also make sure yo udo not have M220 or M221 in effect

lars-danielis, 2015-05-29:
As the printer is printing fine I feel the settings should be ok. I will post the config und result, as soon as I am at home.
The time is around 1 second.
I have nopheads Mendel90 with a Gregs based geared extruder, AZSMZ with 8825 drivers at 1/32, resulting at 1244 steps/mm.

wolfmanjm, 2015-05-29:
I'll ignore the fact for now you are not using a supported version of smoothie hardware.
a greg wades extruder at 1/32 would be around 700-800 steps/mm so yours seems a little high.

lars-danielis, 2015-05-29:
nopheads standard fomula
(200 * 16 * 39.0)/(11.0 * 6.75 * 3.142) = 535 (200 steps, 1/16, 39/11 gears, and hobbed bold with resulting 6,75mm)
and with 1/32 the double value of around 1070

lars-danielis, 2015-05-29:
here the result of M503: http://pastebin.com/C795g90X
and here the config.txt: http://pastebin.com/32zaVGA9

triffid, 2015-05-29:
On 29 May 2015 at 19:09, lars-danielis <notifications@github.com> wrote:

> nopheads standard fomula
> (200 * 16 * 39.0)/(11.0 * 6.75 * 3.142) = 535 (200 steps, 1/16, 39/11
> gears, and hobbed bold with resulting 6,75mm)
>
oh I thought it was mine -
http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide#Calculate

where does nophead cover it?

lars-danielis, 2015-05-29:
Please excuse me, I did not know that the formula was already given in the raprap wiki and invented by you.
Until now I only knew it from nopheads buildinstruction for mendel90

nophead, 2015-05-29:
I don't think anyone can claim to have invented it. It is just applied maths and anyone equipped with high school maths given the problem should come up with the same formula. I was certainly using it before that calibration guide was written.

wolfmanjm, 2015-05-29:
The only issue I see is your E acceleration is way too high, it should be in the order of 500 not 5000.
I also still think your esteps are way too high, my greg wades is 300steps/mm but I presume you extruded 100 mm without a hot end and measured it correctly.
So i can only conclude there is something funky going on with your system as no one else seems to be having this issue. I also have looked at the code and can see no reason why G1 E50 F1 would not take 50 minutes (50mm at 1mm/min). But I would try setting your E acceleration to 500 and see if it helps, if it does then there is a bug somewhere.

nophead, 2015-05-29:
Acceleration of 5000 is fine because an extruder has very little inertia and it works fine in Marlin after I fixed an overflow in the multiply: https://github.com/nophead/Marlin/commit/e4595fa24a2051e95df8a4ed044d2c94cca18e63.

Also I can confirm esteps are about right for a Mendel90 hobbed bolt and x32 microstepping.

wolfmanjm, 2015-05-29:
thank you @nophead it helps to eliminate potential user errors.

DanielisLars, 2015-05-29:
I was using Marlin before with this values. But even with this, for me new, smoothie system I am printing successfully and very happy that I have switched to it. 
I want to thank you all a lot for your support. It is amazing what you are producing.

Is there something I can do or test to bring some light in the dark? Is there a way to get additional output to debug?

wolfmanjm, 2015-05-29:
Ok I see the bug... it is caused by the very high acceleration you are using., I'll fix it ASAP.

wolfmanjm, 2015-05-29:
fixed by https://github.com/Smoothieware/Smoothieware/commit/dda375163abecffde421725841b0570a0e02481e

DanielisLars, 2015-05-30:
Thank you, it is working now, hoping the result of the retracts will be better also.
Printing in this moment.