Smoothieware/Smoothieware #678: Please add contribution guidelines

sorki, 2015-06-30:
To have readable and actually useful commit history this project needs something like this.

Ideas:
* use meaningful commit message describing what has changed, commit messages like Updated extruder.cpp are not useful at all)
* prefix commit messages with subsystem they are relevant to (example: zprobe: fix ignored feedrate parameter of G30)
* keep wiki documentation synced with your changes - this should be automated as it's hard to keep wiki in sync with many features and different versions of firmware

wolfmanjm, 2015-06-30:
We have stuff on the wiki like this. Not sure what else you want, but the wiki is publicly editable. 
http://smoothieware.org/coding-standards
http://smoothieware.org/github

closed 2015-06-30 by wolfmanjm

arthurwolf, 2015-06-30:
Maybe you want this in a file on the github repo like other projects have ?
I think the info should be on the wiki, but we can link to it from a file in the github repo ...

sorki, 2015-07-07:
Looking at today's commit history I don't see any improvement - there are redundant commits that should have been merged into one and commits without meaningful messages:

```
ca3814f Merge pull request #682 from st0ne/edge
51871fb patched arc from grbl project - sometimes arcs did not work correctly / changed comment
5fa0c17 patched arc from grbl project - sometimes arcs did not work correctly
51eef6d Update ZGridStrategy.h
e17ee4a Update ZGridStrategy.cpp
0a37d9f Merge pull request #681 from RepRapMorgan/ZGridupdates_commit
36256be ZGrid: Text based option selection Changed option selection to text based configuration: nohome homexy homexyz
fcdc2e1 Merge remote-tracking branch 'upstream/edge' into ZGridupdates_commit
1cf6dbd Merge pull request #680 from RepRapMorgan/SCARAupdates_commit
4bbbd84 fixed G28 X0 Y0 again (thought I was right initially... ;-)
c449d57 zgrid: add homing options
962b3c7 ZGrid Homing. Added homing options to ZGrid: NOHOME HOMEXY HOMEXYZ (Default)
```

Should be only two commits instead of twelve...

wolfmanjm, 2015-07-07:
well it is as it is, it will not change, so I guess you can stop commenting now. Thank you.