Smoothieware/Smoothieware #683: Does Smoothieware support RTD temperature sensors?

BrianGilbert, 2015-07-07:


euan3826, 2015-07-07:
I'm looking for some support on this as well. Would like to be able to use a PT100 RTD sensor to allow for high-temp printing. As far as I understand, it may be possible by editing the formula in the thermistor.CPP file. Not sure if any signal conditioning hardware is needed though.

BrianGilbert, 2015-08-29:
appears that at least one person git an RTD to work http://smoothieware.org/forum/t-1086570/using-ptc-pt100-pt1000

euan3826, 2015-08-29:
Brian, total coincidence that you've just posted this half an hour ago - I've spent the last 2 hours digging through the thermistor code to try and figure out how to use a temp sensor that has a linear output. Haven't looked at this since we both posted last back in July! Going to have a look at that forum post, cheers!

wolfmanjm, 2015-08-30:
one problem is it is not simply a firmware issue it is a H/W issue.
The thermistor inputs have an R/C circuit on them which you need to remove as they are designed to measure resistance not voltage. The PT100 amplifier outputs a voltage which you first have to ensure never exceeds 3.3v (otherwise you will kill the ADC pins). Then you need to measure voltage not resistance. You could connect it to a bare ADC pin that is spare as well. 

Additionally driving this from a 3.3V line will inject noise into the ADC circuit and make the other thermistors readings innacurate, the ADC voltages are supposed to be derived from the AVCC line not from 3.3v to avoid this, however the AVCC is not broken out on smoothieboard.

NOTE that the E3D PT100 was NOT designed to be used by anything other than an Ultimaker control board, so this is a misuse of this device anyway so YMMV and you use at your own risk.
From the E3D site.. "The Ultimaker 2 uses PT100 sensors by default, and this sensor allows you to connect your E3D Hotend and Ultimaker electronics with no alteration to the electronics or firmware of the Ultimaker 2. A PT100 sensor does require special electronic amplifiers, and will not work on machines other than the Ultimaker 2 without some additional electronics hardware and firmware hacking...." so be warned.

BrianGilbert, 2015-08-30:
@wolfmanjm The way I read their PT100 Kit page that includes amplifier is that the kit is designed to be used on almost any printer..

"This is a kit that provides everything you need to get up and running with a PT100 sensor on most machines"

I've emailed e2d for some support from them.. I will post their reply here.

euan3826, 2015-08-30:
@wolfmanjm Appreciate you taking the time to point out the issues! So aside from the code side of things, we'd need to send the amplifier signal output to a spare analogue input pin. Just to check I'm understanding this properly - the signal input from this will introduce noise into the onboard smoothie ADC, potentially upsetting the other analogue input signals? And this is due to the fact we AVCC isn't accessible on the Smoothieboard? I think this might be worth trying anyway as this noise introduced may or may not be that significant.

Then in terms of setting up the code, some additions and changes needed to the thermistor.cpp file, setting a new configurable pin for the input signal, and then either using the derived temp/voltage equation (preferred) or maybe even the old school thermistor table in the style of marlin.

May be completely out there, but for the sake of getting this working, I wonder if you could use an external microcontroller (maybe a spare RAMPS board running Marlin to interprate E3D amplifier output) to control a digital potentiometer that you then connect to the existing thermistor inputs on Smoothie?

I think there is going to be an increasing number of people wanting higher-temp temperature sensors now as more high performance materials are coming out and hardware/hotend providers are starting to make the hardware more accessible.

triffid, 2015-08-30:
As long as the amplifier is happy driving a 4k7 pullup to 3.3v and a 10uF
capacitor, it should be ok plugged straight into a normal thermistor input.

I can't comment on whether or not it'll work on 3.3v, or put out more than
3.3v (if run from 5v) which may permanently damage your smoothieboard.

euan3826, 2015-08-30:
@triffid Was just reading this thread you were discussing with jetguy: https://groups.google.com/forum/#!topic/smoothieware-support/FVf4FQAL9Ic 

I guess leaving the pullup resistor in may not have that much of an effect, but also could even be compensated for in the voltage/temperature conversion?

BrianGilbert, 2015-08-31:
I just phoned e3d (from Australia) and spoke with Josh, he said it should be possible to make it work.. he's having a look at the situation.. hopefully I'll hear back soon or he'll post something here.

euan3826, 2015-09-01:
Been testing things out this morning. I'm completely new to C++, so been using arduino to experiment. 
Managed to take the reading from the E3D amplifier output and use a formula (temp = (sensorValue - 153.22) / 0.522) to approximate the temperature. The problem is that the step between 1 value and the next on the ADC equates to a difference of nearly 2deg, so looking into taking maybe 20 readings and averaging them to approximate the temperature (based on frequency of readings for each step in the sample). I'm sure this can be more or less translated to C++, but my worry is how often can the smoothieboard read the sensorvalue from ADC and store n readings to the buffer.

arthurwolf, 2015-09-01:
Arduino and Smoothie are both C++
And Smoothie already does the averaging thing.

Cheers.

BrianGilbert, 2015-09-02:
@euan3826 Are you using the PT100 with the e3d amplifier board or just by itself?

euan3826, 2015-09-02:
@briangilbert using the E3d board. The pullup resistor (or R/C circuit) on he thermistor inputs seem to be causing problems, so it'll need to use a spare analogue input pin

euan3826, 2015-09-02:
Going to try and run that through smoothie later today and see if the inbuilt smoothing works well enough. I see that smoothie can do oversampling now too, so this could help get a more accurate signal from the ADC on smoothie.

wolfmanjm, 2015-09-03:
FYI for high temp stuff it is highly recommended to use the SPI based thermocouples, they are already supported. and require no modification of the smoothieboard
One for instance is here...  http://shop.uberclock.com/products/spi-thermocouple which BTW is cheaper than the e3d PT100 and amplifier board which is currently not supported.

BrianGilbert, 2015-09-03:
@wolfmanjm that would be great, but we already have the PT100 boards, so another purchase isn't ideal.

BrianGilbert, 2015-09-07:
@euan3826 any progress?

euan3826, 2015-09-09:
Can't seem to get smoothieboard to use a non thermistor pin as an input. Connecting the signal cable to pin 1.22, but smoothie is still reading pin 0.23. Is there something else i need to change?

wolfmanjm, 2015-09-09:
@euan3826 1.22 is NOT an ADC pin you can't use that.
the valid ADC pins are 0.23, 0.24, 0.25, 0.26, 1.30, 1.31 No other pin can be used for ADC

BrianGilbert, 2015-09-20:
@euan3826 are you still working on this at all?

BrianGilbert, 2015-10-03:
E3D have added table with voltage reading for various temperatures http://wiki.e3d-online.com/wiki/E3D_PT100_Amplifier_Documentation (I don't think this was there last time I checked)

BrianGilbert, 2015-10-03:
@wolfmanjm said in IRC

```
first you have to make sure it runs at 3.3v and not 5v
then you need to connect it to a spare unused ADC port that is NOT one of the thermistor ports
then you need to derive the formula for that table as we do not allow tables, we calculate it.
then you nened to create a module like the thermistor.cpp or max31855.cpp and return the temp based on the ADC reading

ADC is 3.3v MAXIMUM
5v WILL KILL an ADC port

BrianGilbert, 2015-10-05:
From e3d:

```
This is not something we've tested or designed for, but if the amp is powered at 3.3v instead of 5v then the output signal should simply be scaled to vary from 0-3.3v instead of 0-5v. However I cannot assure this behaviour, and you should check with a multimeter etc  before hooking up to your smoothieboard etc.

I am afraid I am not able to supply you with code/formulae for converting the adc values to temperatures, but if it's of any help this should be fairly easy code as the output from the amplifier is essentially completely linear.

Regards,
Sanjay

wolfmanjm, 2015-10-05:
actually it is NOT linear at all. It is more linear than a thermistor though.

euan3826, 2015-10-06:
Hi Brian, 

 

I took lots of measurements of the sensor, compared them with the sensor 20 table in Marlin and indeed it’s the same function, just values adjusted for 3.3V max instead of 5V.

 

Stuck this into Excel and managed to get a line of best fit equation that we should be able to use. It’s resolution isn’t that great, but the oversampling should be able to compensate for that a little bit. Going to have a look at writing the .cpp file later this week, but my Smoothie Firmware won’t let me in to edit see any of the onboard files (I understand this probably isn’t the correct/best way of doing things anyway)

 

Jim, could you please advise on the best way of adding a .cpp file to the firmware?

 

Cheers, 

Euan

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

triffid, 2015-10-07:
http://smoothieware.org/compiling-smoothie

heathmanc, 2015-10-08:
Why bother worrying about 5v max, according to that table, you should never come close to hitting 5v temps when printing.

wolfmanjm, 2015-10-08:
@heathmanc because anything over 3.3v will fry the ADC port and void your warranty :)

heathmanc, 2015-10-08:
I got that, but as an interim solution that normal people wouldn't encounter....3 volts will allow you to take it to 500C...

heathmanc, 2015-10-08:
I am currently using an Auber PID controller with the E3D PT100 kit, it makes things a pain in the butt, but actually works really well.

BrianGilbert, 2015-10-09:
@euan3826 any chance you could make table and equation visible somewhere please?

BrianGilbert, 2015-10-13:
I caved in and installed the regular semitec thermistors that were supplied with the hotends

ernstc, 2015-12-26:
What about using 5V input and a diode zener 3V1 for cutting the voltage upper then 3.1V ?

mpilar, 2016-03-09:
I have been looking into this as the last thing I would need to switch to a smoothieboard from my RAMPS setup (as I already have dual extruders with the e3d PT100 kit.

Some of the research I've done has pointed me to http://www.ti.com/lit/an/slyt442/slyt442.pdf which has a link to this zip file: http://www.ti.com/general/docs/lit/getliterature.tsp?baseLiteratureNumber=SLYT442&fileType=zip that includes the formulas for the INA826    Instrumentation Amplifier. Given the same amplification circuit the output will be Vref at max, so supplying the amplification circuit with 3.33 eliminates the possibility of frying the ADC.

Having done the research I am satisfied it can be implemented and I'll order a smoothie now and try to wrangle together the code to make the INA826 board work using an ADC pin.

dintid, 2016-04-03:
Anyone figured this out yet? What I read the experts sounds as if it is a simple matter, but I have been unable to find any example of a working setup. Meaning wiring and code. 

Regarding voltage: have to reach extreme temperatures before e3d outputs 3.3v. But seems as if se Can just feed the amplifier 3.3v and adjust the formula accordingly - from what I've read. 

Some people wonder why it matters: some of us allready own multiple pt100 sensors, amplifiers and hotends or just alu-blocks suitable for the pt100 sensors. 
I personally have 3 E3D "normal" hotends, 2 volcano and 1 Olsson block for the pt100 sensors and 3 E3D amplifier boards, so I do not just stop using them and go to some thermistor or thermocoupler. 

Aside from what I own and use, the pt100 design is just so much nicer to use: easier to mount, doesn't break more accurate and can stand higher temperatures.

lyonsbp, 2016-04-13:
I also would like to know if that has been an update. @euan3826 what did you figure out?

crncrussell, 2016-06-19:
Would this board work with Smoothie to allow it to see the RTD's as thermocouples as an interim solution?  There is both a single and dual channel version of the board.
http://www.playingwithfusion.com/productview.php?catid=1001&pdid=29

soup3y, 2016-07-10:
Is this issue still being fixed with the PT100 or has something already been released?  If not would the idea from @crncrussell work?  Would really like to upgrade to the smoothieboard as my ramps just fried and this board seems to have a great community behind it.  Hope this gets fixed soon.

dintid, 2016-07-11:
I would like to know how and if solution by @crncrussell works as well.
I guess 2 of these should work: https://www.adafruit.com/products/1727 but I'm not that good at figuring out cabling and such myself.

As an interim solution I bought size-compatible thermocouples to fit in my pt100 blocks:
http://www.aliexpress.com/item/5pcs-lot-3D-printer-K-type-thermocouple-3X20-thermocouple/32283945375.html

crncrussell, 2016-07-11:
FWIW- no one answered me here on those boards, so I never tried them. Still holding off on getting a Smoothie until this is resolved one way or another.

soup3y, 2016-07-11:
@crncrussell I will be testing a solution this weekend hopefully if classes don't get in the way of me setting up the new board.  Arthur Wolf will be sending me instructions on wiring and config files to test when I have completed setup.  Will let you know if it works out.

crncrussell, 2016-08-03:
@soup3y did you ever have any luck getting this working?

wolfmanjm, 2016-09-29:
can someone here test pull request #1034 to see if it works with the E3D sensor?

arhi, 2016-09-29:
This config should work 

```
temperature_control.hotend.enable              true
temperature_control.hotend.sensor              pt100
#temperature_control.hotend.ampmod1_pin         nc
#temperature_control.hotend.ampmod2_pin         nc
temperature_control.hotend.slope               0.3173828125
temperature_control.hotend.yintercept         -299.6
```

but if the slope and yintercept are not ok for e3d board just change them to

```
temperature_control.hotend.slope               0.1
temperature_control.hotend.yintercept        0.0
```
so that you get the ADC value (divided by 10) as the temp, heat up the nozzle to the bottom of the temp range you want to use (e.g. 150C) and then heat it up to top range you want to use (e.g. 300C) and the use the two values to calculate slope and yintersect 
slope = (realtemp2-realtemp1) / (adc2*10-adc1*10)  # *10 because you are reading adc*0.1 if you set slope to 0.1
yintersect = realtemp1 - (adc1*slope) 

the PT100 is not linear (there's actually a proper calculation in the code but commented out as it not as easy to use) but in the "short" range we are using it (100-300C for e.g.) we are talking about <2C error .. and if you tighten the range for e.g. 160-280C the error is below 0.5C

pyjamasam, 2016-10-14:
@arhi I used the config you listed (slope as 0.317....  and yintercept as -299.6) and gave things a try with the e3d board and strangely I am getting a temperature thats * 100.

I am going to give your raw reading a try though I have to sort out how to heat up the hot end without a working temp probe :-)

arhi, 2016-10-14:
@pyjamasam thanks for testing, if I had e3d board handy I could find the parameters easily :D but let's try to do it remotely :D .. do you maybe have a PT100 sensor that's not lodged in the extruder? a, free one, that you can put into ice water and into boiling water? if you do, getting a raw reading from smoothieware. set slope = 1 and yintercept = 0 in config and you will get raw value .. get those for room temp, water with floating ice in it (0C) and boiling water (100C) and that's enough to get rough values for that amplifier

pyjamasam, 2016-10-14:
My multimeter (with its temp probe) shows 20C and the board is showing 5665 raw adc value.

Let me try and get the temp to swing around a bit and get more readings.
37c 6104adc
47c 6363adc
75c 6627adc
93c 6676adc
155c 7996adc
174c 7928adc
197c 8905adc

Sadly my meter doesn't show decimals, but some judicious use of a heat gun and holding the two probes at various distances got me those numbers.

Hopefully we can sort out a range with them.

arhi, 2016-10-14:
hi, weird results, can you measure the voltage output of the e3d amp at
those (or similar) temps ... as I'm looking at this and it's not making
much sense:

http://wiki.e3d-online.com/wiki/E3D_PT100_Amplifier_Documentation

arhi, 2016-10-14:
I found the source of my confusion here .. difference between lib/Adc
and lib/ADC/adc ....

the
https://github.com/Smoothieware/Smoothieware/blob/edge/src/libs/ADC/adc.cpp
ADC:read() returns the  return((_data_of_pin(pin) >> 4) & 0xFFF);
so no way in hell can it return number that's less then 0 or greater
then 4095 right?!

but the THEKERNEL->adc->read(&thermistor_pin); I'm using is not using
that adc but the other one
https://github.com/Smoothieware/Smoothieware/blob/edge/src/libs/Adc.cpp

no need to do any measurements I should have enough data now to get
proper parameters for config, just need to get back home :)

pyjamasam, 2016-10-14:
Ok after a little crazy wiring I have an e3d semitec thermistor taped to the pt100.

Extruder temp is the raw adc value to the pt100, bed temp is the thermistor, and I have my multimeter hooked up.  hahaha.

Here we go.
1.166v
RECV: ok T:5693.0 /0.0 @0 B:21.8 /0.0 @0

1.442v
RECV: ok T:7074.0 /0.0 @0 B:100.9 /0.0 @0

1.794v
RECV: ok T:8838.0 /0.0 @0 B:215.7 /0.0 @0

arhi, 2016-10-14:
perfect values, confirms that THEKERNEL->adc->read(&thermistor_pin); is
returning 2 additional bits of resolution :)
sending new coefficients in a bit

arhi, 2016-10-14:
here we go, the proper values:
slope: 0.0523681641
yintercept: -300

according to my calculation range 180-260C has error below 1.6C .. here are the errors between 150 and 280C:

150C	error: 2.8C
160C	error: 2.4C
170C	error: 2C
180C	error: 1.6C
190C	error: 1.2C
200C	error: 0.8C
210C	error: 0.4C
220C	error: 0C
230C	error: -0.4C
240C	error: 1.8C
250C	error: 1.4C
260C	error: 1C
270C	error: 3.2C
280C	error: 2.8C

arhi, 2016-10-14:
now this is slope/yintercept centerered around 220C if you wish to center it around some other value you can modify slope and yintercept for shorter precise range and around other temperature, depends what you mostly use ... but since 4-5C difference between sensor and tip is "nothing" (it goes up to 15C on some hotends, dunno about e3d don't have one to measure)  the 2C error is irrelevant really.

pyjamasam, 2016-10-14:
For reference here are all the readings from about 228c down to 100c
T:8871.0 B:228.3
T:8918.0 B:228.5
T:8899.0 B:228.9
T:8901.0 B:228.8
T:8908.0 B:232.3
T:8943.0 B:237.0
T:8991.0 B:243.2
T:9039.0 B:245.0
T:9005.0 B:240.7
T:8990.0 B:238.7
T:8988.0 B:237.1
T:8977.0 B:235.3
T:8963.0 B:233.6
T:8917.0 B:232.8
T:8907.0 B:232.0
T:8896.0 B:231.3
T:8933.0 B:230.8
T:8890.0 B:230.4
T:8883.0 B:230.3
T:8889.0 B:230.3
T:8890.0 B:230.4
T:8890.0 B:230.3
T:8899.0 B:230.2
T:8936.0 B:230.5
T:8936.0 B:230.5
T:8888.0 B:230.5
T:8890.0 B:229.9
T:8893.0 B:230.1
T:8940.0 B:230.6
T:8939.0 B:230.8
T:8929.0 B:231.1
T:8839.0 B:227.0
T:8718.0 B:216.9
T:8634.0 B:206.0
T:8499.0 B:195.5
T:8381.0 B:185.6
T:8210.0 B:177.0
T:8131.0 B:168.1
T:8016.0 B:160.5
T:7897.0 B:154.1
T:7797.0 B:147.9
T:7658.0 B:142.0
T:7553.0 B:136.7
T:7461.0 B:131.3
T:7383.0 B:125.9
T:7345.0 B:121.0
T:7272.0 B:116.3
T:7172.0 B:111.9
T:7138.0 B:107.6
T:7079.0 B:103.3
T:7021.0 B:99.2

I'll give your coefficients a go now and do the same test.

pyjamasam, 2016-10-14:
Hmm.  Not so good.

T:173.0 B:242.3
T:166.3 B:230.4
T:159.7 B:218.7
T:150.5 B:207.5
T:143.7 B:197.2
T:138.5 B:187.6
T:132.1 B:179.0
T:125.8 B:171.7
T:118.0 B:164.5
T:112.9 B:157.5
T:107.9 B:150.7
T:102.8 B:144.4
T:100.5 B:138.6
T:94.0 B:132.5
T:90.0 B:127.4
T:85.9 B:121.9
T:83.3 B:117.0
T:79.4 B:113.0
T:75.6 B:108.8
T:72.5 B:104.4
T:69.6 B:100.8
T:64.8 B:97.5
T:61.8 B:94.2
T:59.4 B:91.1
T:56.8 B:88.6
T:54.3 B:86.3
T:51.9 B:83.8

pyjamasam, 2016-10-14:
173 should be 242...  I have the two probes next to each other being heated by the same source.

Voltage doesn't seem anywhere near the listed numbers on e3d's site.  

I am seeing 1.823v for 8934 adc and a temp on the semitic sensor of 250C
According to the e3d docs that 1.823v should be between 170 and 180. (which is about what we are seeing when we use your coefficients).

And just incase I am doing something completely stupid...  I have the e3d board hooked up to ground, 5v, and my analog input.

I am never going to get into the temp where it start to return a voltage over 3.3v so that doesn't concern me.

Hmm..  I just looked again at the voltage I am supplying the board with.  It was <5v (coming from the USB power NOT my PSU). 

adc readings drop a LOT 

USB power - T:5634.0 B:31.0
PSU power - T:6017.0 B:30.9

USB is 4.69v
PSU is 5.01v

That might explain the low reading.

I'll give your coefficients a try again and see what I get.

arhi, 2016-10-14:
> Voltage doesn't seem anywhere near the listed numbers on e3d's site. 
my coefficients are calculated based on what e3d wrote on their site.

>  I have the two probes next to each other being heated by the same source.
that really does not make sure they are at same temp :( especially if they are in the air and you are heating them with heatgun .. 10C diff there is peanuts :( .. but these are too huge differences to be just that :(

What you can do if you have a pot ~100-500R is attach a pot to that amp instead of the sensor and try few different values of resistance and check the voltage out of the amp. you can get table for res-vs-temp for pt100 easily on google and you can test if what e3d claims for that amp is true (temp vs voltage as you can easily test and measure both)

> Hmm.. I just looked again at the voltage I am supplying the board with.
>  It was <5v (coming from the USB power NOT my PSU). 

I dislike the schematic of that amp big time .. it is true that power of the opamp here is not important as it's not getting nowhere near 5V *but* that R2 (2k2 to 5V) makes a huge difference so if 5V going to R2 is not 5.0 but 4.6 :( .. what I don't get is how the hell they didn't put any reference on that board .. the point of PT100 is to have "precision" compared to crappy NTC's and then they make amp that has no reference :( .. it's reeeeeeeeeally crappy solution :(

I'm attm I'm creating a spreadsheet where you can easily calculate coefficients for testing :D

lemme know how it goes with proper psu :D

arhi, 2016-10-14:
here it is, hopefully useful:

https://docs.google.com/a/crsn.rs/spreadsheets/d/1euKSu-5grHaLCSt9ofdAED0ucqRxvrGuc4NIl6BoTQw/edit?usp=sharing

copy the graph to your drive so you can edit, edit green fields so you select from and to temp you are interested on having as precise data as possible, so for e.g. 200-260C slope is 0.05255613951 and yintersect is -302.173913 and error at 230C is zero... there's the error graph too there .. 

changing voltage column to write exactly what voltage is out for any temp will also recalculate everything... all formulas are simple and visible :D

pyjamasam, 2016-10-14:
Ok feeding a 220R resistor into the inputs gives me a voltage output of 2.369v on my multimeter and 11608.0 for the adc reading

According to a PT100 resistance chart I found I should have about around 310-320C for that resistance value, and looking at the e3d voltage chart that's about right.

150R gets me 1.670v on the multimeter and 8184.0 adc.  So again about right (should be 140-150C)

So the board is doing what we expect...  

This is powered via the 5v supply from the PSU.  So it's about as close as I can get to 5v.

But then I plug in the sensor I don't seam to get anywhere near the readings I expect for temperatures I expect.

They are low across the range.

An example 
1.7359
165.7 168.1 -2.4

A voltage of 1.7359 should land me somewhere between 150 and 160 according to the e3d data, yet the thermoresistor is showing 168.1

I am going to try another amp board I have around.  This is all very strange.

pyjamasam, 2016-10-14:
Thanks for the spreadsheet.  Helped a lot.

I converged on a slope of 0.06993006993 and -438.6713287.

I am not very satisfied though with this though.  I ended up with a fairly tight error band around the "normal" temps for printing, but when cold (i.e. room temperature) I am getting a negative number.  I realize that thats a byproduct of the simplistic scaling we are doing.

A bigger problem that I am seeing is that two PT100 cartridges from e3d aren't giving me anywhere near the same numbers.  So it makes my life more difficult when I want to swap hotends (between my lite6 and v6 for example depending on the material I want to print).

I am going to continue digging a bit and possible contact e3d support about this, so I wouldn't say that this things are going to stop...

BUT that being said, the code DOES work for the e3d board.  You do get readings.

[29 more replies]

closed 2017-03-29 by wolfmanjm