Smoothieware/Smoothieware #685: Laser module Minimum and Maximum power settings

thinkl33t, 2015-07-18:
Currently, the `laser_module_max_power` sets the default power that is used when a `G1`/`G2`/`G3` command is issued with no `S` modifier.  This value is then auto-adjusted in the laser module to make it proportional to the head speed, before it is then output to the laser PWM pin.

Reading the documentation, I got the mistaken impression this config variable would set the limit value and scale the `S` values using it. 

I plan on replacing the config variables:
* `laser_module_max_power`
* `laser_module_tickle_power`

with:
* `laser_module_max_power` - New config variable, used to set the maximum PWM frequency that will be sent to the laser.
* `laser_module_min_power` - Replace old `laser_module_tickle_power`, this would be the minimum power sent to the laser, and would normally be set to a value *just under* the value at which your laser fires.  On `G0` moves, the laser would be set to this power.
* `laser_module_default_power` - The power level to use if a value hasn't been set by an `S` modifier.

Any S values sent, along with the default power, would be scaled automatically to be on a scale between min and max power.  for example, if the config is set to:
 
```
laser_module_max_power = 0.8 
laser_module_min_power = 0.2
laser_module_default_power = 0.6
```

* If the user sends a `G0` command, the laser PWM output would be set to `0.2`.
* If the user sends a `G1` command, the laser PWM would be set to `((0.8-0.2)*0.6)+0.2 = 0.56`
* If the user sends a `G1 S0.1` command, the laser PWM would be `((0.8-0.2)*0.1)+0.2 = 0.26`
* If the user sends a `G1 S1.0` command, the laser PWM would be `((0.8-0.2)*1.0)+0.2 = 0.8`

This preserves existing functionality, and provides a consistent range for scaling the laser power, and allows laser users who may want to preserve tube life to set a maximum limit to laser power.  

Also, as a laser tube ages, the power needed to make it turn on gets higher, so the minimum limit allows the user to adjust up, and continue to use the same gcode files with no changes.

wolfmanjm, 2015-07-18:
you need to maintain backward compatibility with old configs. So rather than replace you would add new settings, and the old settings would remain the same.

arthurwolf, 2015-07-19:
+wolfmanjm in this specific case, users actually expect it to behave the way he is describing, and I've gotten several reports of people thinking this is a bug, and having to do dirty host-side hacks.
For the few users for which this will break backwards compatibility, it will do so in a way that's not dangerous ( ie it'll be lower power than they expect ).
Even the documentation makes it sound like it works the way he wants to change it to, so I think this is one exceptional case where it's fine to break things a bit.

If we really want to keep compatibility, we can do : 
max_power and default_power are synonyms. 
hide max_power from configuration examples and documentation ( so it only exists in old configs )
add maximum_power and minimum_power, and document those in config examples and wiki
it costs us a few lines of code

FredMurphy, 2016-03-02:
This appears to have been completed in commit 6e986d7. Suggest closing.
https://github.com/Smoothieware/Smoothieware/commit/6e986d7ef3b7566552b20faa86fd9282192ddead

closed 2016-07-02 by wolfmanjm