Smoothieware/Smoothieware #687: ThreePointStrategy needs z move before aiming first probe point

risip, 2015-07-22:
Currently ThreePointStrategy is directly moving to first xy probe point, without moving the extruder up, according to the parameter 'zprobe.zprobe_height'. Therefor it can happen that the nozzle crashes into the print bed after homing. Example gives after homing to z-min outside the print bed with tilted print bed.

To solve this issue following code should  be inserted before line 263:

	// move up to specified probe start position
	zprobe->coordinated_move(NAN, NAN, zprobe->getProbeHeight(), zprobe->getSlowFeedrate()); // move to probe start position

right before:

	// move to the first probe point
	std::tie(x, y) = probe_points[0];
	// offset by the probe XY offset
	x -= std::get<X_AXIS>(this->probe_offsets);
	y -= std::get<Y_AXIS>(this->probe_offsets);
	zprobe->coordinated_move(x, y, NAN, zprobe->getFastFeedrate());

wolfmanjm, 2015-07-22:
I understand the issue, but I do not think I agree with the solution.

wolfmanjm, 2015-07-22:
Basically you are supposed to manually move the head initially to a height high enough above the bed to avoid hitting anything before you do the probe. This is really the only reliable way as before the leveling nothing is certain.

closed 2015-07-22 by wolfmanjm

risip, 2015-07-22:
Hello Jim,

we want to implement the autobed leveling feature to help the users to get more automated prints, without manual interaction, e.g. for manual print bed leveling.

Since the parameter 'zprobe.zprobe_height' is already available and which is used during z probe, why it should not taken into account also for the very first probe?

Could you please explain the disadvantage for my proposal.

Best regards

Ismail Schl?ter
Developement Engineer

BigRep GmbH
Gneisenaustra?e 66
10961 Berlin
Germany
Fon +49 30 208 482 60
schlueter@bigrep.com

BIG IDEAS! BIG PRINTS!
www.bigrep.com<http://www.bigrep.com/>
BigRep GmbH
Management Board: Ren? Gurka (CEO), Lukas Oehmigen (CTO), Stephan Beyer (CFO)
Court of Registration: Amtsgericht Charlottenburg, HRB 155360B
Registered Office: Berlin