Smoothieware/Smoothieware #688: ThreePointStrategy is calculating wrong z distance at the probe points

risip, 2015-07-22:
Currently during the calculation of the z distances at the probe points the probe_offset for z according to parameter 'leveling-strategy.three-point-leveling.probe_offsets' is not taken into account.
Therefor wrong values are calculated for the distance to the bed. This does not impact the calculating of the plane itself, but the bed distance itself must set manualy later at the first probe point.

To solve this issue current code at line 289:
	
	 z= zprobe->getProbeHeight() - z; // relative distance between the probe points, lower is negative z

must be replaced with code:

		z= zprobe->getProbeHeight() - std::get<Z_AXIS>(this->probe_offsets) - z; // relative distance between the probe points, lower is negative z

wolfmanjm, 2015-07-22:
This is actually how it was designed to work. It is not supposed to set the level it defines a plane relative to the first probe point. After saving the plane you then zero the head to the bed level at the first probe point. This is explained in the wiki.

IMO this is the most generalized way to handle it, as not everyone has a permanent probe that will always be at the same distance from the tool head.

wolfmanjm, 2015-07-22:
It even says so in the comment... "relative distance between the probe points"

closed 2015-07-22 by wolfmanjm

risip, 2015-07-22:
Hello Jim,

I understand your point of view. But you are an expert in 3d printing, most of the users are not.

But is there a disadvantage of my proposal, since this modification will not impact the current behavior at all, but make it more user friendly.

Best regards

Ismail Schl?ter
Developement Engineer

BigRep GmbH
Gneisenaustra?e 66
10961 Berlin
Germany
Fon +49 30 208 482 60
schlueter@bigrep.com

BIG IDEAS! BIG PRINTS!
www.bigrep.com<http://www.bigrep.com/>
BigRep GmbH
Management Board: Ren? Gurka (CEO), Lukas Oehmigen (CTO), Stephan Beyer (CFO)
Court of Registration: Amtsgericht Charlottenburg, HRB 155360B
Registered Office: Berlin

Closed #688<https://github.com/Smoothieware/Smoothieware/issues/688>.

-

wolfmanjm, 2015-07-22:
I think your suggestions will work in your branch, but is not general enough IMO for smoothie.
The basic idea with smoothie bed leveling was to attach a temporary probe. then do the G32 level, then save the results then remove the probe. Because the probe is not permanent the zprobe z offset is not very accurate. I want it to be generally useful for people using it that way. This is different from say Marlin where you cannot save the results and have to probe every print.
Setting the bed height is a fundamental skill everyone needs to have, and this way you do the bed level once and you do the setting the bed height once, save both and do not do it again unless you change the bed. Your zprobe offset in the z direction would need to be extremely accurate for your idea to be useful and produce usable results. Most of us cannot measure the zprobe offset that accurately.

risip, 2015-07-22:
I understand your point of view better now. Thanks for clarifying.

Yes you are right with your assumptions. We are using a permanent mounted laser sensor with 10um resolution to assure correct measurements.

wolfmanjm, 2015-07-22:
but you also need to know the exact offset the probe triggers from the nozzle to be accurate. if you are out by 0.1mm then you will have bed adhesion issues, and most people can't measure to 0,1mm. this is not the probe accuracy but determining where the probe triggers in relation to the nozzle tip.