Smoothieware/Smoothieware #711: Thermocouples via SPI crash smoothe

kiabonov, 2015-09-01:
When thermocouples are communicating via SPI and smoothie is connected to PC via USB, smoothie crashes. Thermocouple board is self made, based on MAX31855. First thermocouple CS is connected to pin 1.21 and the second one to the pin 1.24. Pin 1.24 is disconected from LED and is only used in thermocouple SPI. More details and timing diagram can be found in smootheware forum thread : http://smoothieware.org/forum/t-1338158/thermocouple-via-spi

arthurwolf, 2015-09-01:
Did you disable the leds in the config file ?

kiabonov, 2015-09-02:
Yes i did, with command leds_disable    true.

wolfmanjm, 2015-09-02:
please paste your entire running config and the results from M503 to pastebin and post the link here.

wolfmanjm, 2015-09-02:
Which SPI are you using? I don't think you can share the sdcard SPI with this. I would suggest you use the other SPI. see if that works better.

kiabonov, 2015-09-02:
Config : http://pastebin.com/dUn5fBP0
M503 results: http://pastebin.com/ir3wjUft
Thermocouples and SD card are using SPI channel 1, LCD is using channel 0.

wolfmanjm, 2015-09-02:
well that is probably the problem you can't use SPI1 (the internal sdcard) this way. It will cause corruption on the sdcard reads and hence the crash.

Use the same SPI channel as the LCD uses. Closing this as thi sis not a firmware issue.

wolfmanjm, 2015-09-06:
SPI is called during ISR, this is very bad. fix ASAP

hakalan, 2015-09-08:
Hi, sorry about the SPI calls from ISR. My bad.

However, sampling a thermocouple or thermistor needs to be done at a fairly regular interval, and my assumption was that the main thread will block from time to time and cannot be used for this. 

If there is a solution to this which I'm missing, please advise and I will see if I can provide a fix.

wolfmanjm, 2015-09-08:
@hakalan Hi, Don't worry about it I'll fix it, i also need to update the way it is averaged and does noise elimination along the lines of what I did for thermistor.

Basically it must be read during on_idle, but the timer can tell it when to read. it only gets read at 20Hz so it is not really that important if it is late by a few milliseconds. It will not be allowed to do SPI during interrupt so there really is no way around that issue.
I'll do a moving average and also do some noise rejection..

hakalan, 2015-09-08:
Ok, thanks!
I agree timing is not critical, as long as there are no long interruptions
in the OnIdle calls.

I worry a bit about the main loop getting stuck and the heater turning on
forever.
Some protection against that scenario would be nice.
Let me know if I can help.

wolfmanjm, 2015-09-08:
on main loop has nothing to do with heaters
heaters are maintained in on_idle and during interrupts so there is no issue there.

arthurwolf, 2015-10-30:
Any progress on this ? @wolfmanjm did you need some hardware to test this or something ?

wolfmanjm, 2015-10-30:
I have the H/W, and I need to rewrite the entire thing to not use interrupts the way they are used. I just need to find some time, it seems to not be a very high priority.

arthurwolf, 2015-10-31:
Somebody came to the forums looking for help setting up multiple
thermocouples, I told them it was possible, completely forgot about this
bug, and now they are very upset it won't work. I kinda messed up there.

And in support emails I've had a few people recently ( since september )
asking about multiple thermocouples, not sure why exactly, but it's getting
popular for some reason.

theoo, 2016-01-05:
In my case I decided to switch the electonic from a spinderbot (spiderbot.eu) to smoothieboard.
It's pretty strait forward, but the printer is equipped with thermocouple. Therefor I ordered these thermocouple amplifier (https://www.adafruit.com/product/269) and managed to bind them with the board. It works well but not long.

It could be one explanation: migrating from a random electronic to smoothieboard.
Another case could be when one needs more than 4 thermistors, 2 heads + bed + heating chamber + coolant + the coffee machine...

oysteinkrog, 2016-02-09:
@arthurwolf: New E3D V6 uses a PT100 thermocouple instead of a thermistor, they have an adapter board but I imagine people want to use the thermocouple directly.

greyltc, 2016-03-31:
I hope this PR will solve this issue: https://github.com/Smoothieware/Smoothieware/pull/891
Thoughts?

tkurbad, 2016-11-11:
Just for clarification: Would PR #891 enable the use of more than one thermocouple on the same SPI channel?

wolfmanjm, 2016-11-11:
it should fix it yes

arthurwolf, 2017-01-23:
Anyone would please mind testing https://github.com/Smoothieware/Smoothieware/pull/891 ?
Even testing it with just one thermocouple would be useful as it would at least confirm it's not broken, and would enable us to merge it ( if a problem still exists with multiple thermocouple that can be worked on later )

tkurbad, 2017-01-23:
I can confirm that #891 works stable with one thermocouple. I've done 10 or so long running prints with Smoothie patched from PR #891.

I'll test with two thermocouples as soon as my dual extrusion upgrade is ready.

arthurwolf, 2017-01-23:
Thanks ! Please tell us the results.

tkurbad, 2017-01-23:
Will sure do, but it may take some time, before I'll come around to it. So, probably just merge the PR and I'll report back my findings regarding multiple thermocouples anyway...

arthurwolf, 2017-01-23:
@wolfmanjm ok with merging ? this *probably* fixes the issue, and it works
fine for a single thermocouple

wolfmanjm, 2017-01-23:
As it changes code related to normal thermistor too it needs to be tested that it does not break other sensors especially thermistor.

arthurwolf, 2017-01-23:
Can anyone please test the thermistor code ?

tkurbad, 2017-01-24:
It works well with my heated bed thermistor, if that counts.

arthurwolf, 2017-01-24:
I think it does.

tkurbad, 2017-02-07:
Can I be of any more help to get this merged?

theoreichel, 2017-02-10:
I can try the firmware as well, @tkurbad would you mind sending giving a link to the .bin so I don't have to patch and compile it?
My machine delta printer with two thermocouples via SPI for the heads, and a thermistor for the heated-bed.

tkurbad, 2017-02-10:
@theoreichel here you are: http://www.tk-webart.de/projekte/kossel/firmware.bin

theoo, 2017-02-11:
Before upgrading I was able to print with a firmware dated from July 2016.
Since I tried @tkurbad patched firmware (thanks!) I have the following error:

    Recv: ERROR: Temperature runaway on B (delta temp -20.095276), HALT asserted, TURN POWER OFF IMMEDIATELY - reset or M999 required
    Changing monitoring state from 'Printing' to 'Error:  Temperature runaway on B (delta temp -20.095276), HALT asserted, TURN POWER OFF IMMEDIATELY - reset or M999 required'

As it's a temperature-related issue, I guess it makes sens to describe my setup below and provide 
[my config file](https://gist.github.com/theoo/1a0f820db45e89af5b02c2b4171e2a09).

This error is correct, the "blower" of my printer (12cm above the head, you can see it on the [printer picture](http://spiderbot.eu/en/products/printer-kits-pb/spiderbot-v2-2-standard-kit.html)) starts around layer 10 (I tuned it in the slicer) and makes the bed temperature to drop from 105°C to around 85°C. This is a design flaw because the effector cannot carry more weight for a smaller and preciser fan.

I have temporary updated my config with the following lines, hoping to test the firmware:

    temperature_control.bed.runaway_heating_timeout 0
    temperature_control.module_name.runaway_range 60

Unfortunately the (exact same) error continues to pop, with a delta temp of roughly -20°C, I wasn't able to disable this "new feature" (it was obviously not available in my former firmware).

I tried with the "vanilla" firmware baked yesterday (the .bin), same error, unable to print an entire part.

Apart from that, if I turn off the blower and let the ABS flow in an misshapen heap of plastic, the SmoothieBoard doesn't freeze or crash. I did 3 prints of about 90 minutes each with the both heads used (support or bi-color).

As you can read in the config file above, I use two thermocouples for the head and a thermistor for the bed.
Let me know if I can be of any help.

theoo, 2017-02-11:
I printed [this thing](http://www.thingiverse.com/thing:729822/#files) tree times. The first time the temperature reading on one thermocouple went crazy low from one reading to another and stopped the printer exactly the same way as describe in my previous post. But on the T0 tool instead of the bed.

`Error: Temperature runaway on T (delta temp -65.566666), HALT asserted, TURN POWER OFF IMMEDIATELY - reset or M999 required`

I added the same `runaway_range 60` for both tool heads in the config and the printing succeeded. On the graph the temperature was steady.

On the hardware side, I'm wondering if the MAX31855 chip could have had a hiccups, the SmoothieBoard, the therocouple card from adafruit and the Raspberry PI are powered by a beefy 5v DC/DC converter from Meanwell. The cabling seams good and I never had such problem before. 
I will remove the `runaway_range` and see how it goes these next days.

Is it something that worth digging?

tkurbad, 2017-02-12:
@theoreichel I've had the same effects as you describe with runaway temperatures. For me it came down to two influencing factors:, which also had nothing to do with the patch:
* First, I've also had my fan blowing towards my bed / nozzle and cooling things down a lot.
* Secondly, after re-doing the PID calibration of both bed and nozzle, I've got much more stable temperatures.

In fact, my impression is that the way the patched version reads the temperatures from the various sensors makes design flaws and inaccurate configuration values surface more quickly/violently, which is a good thing both for safety and for print results.

Do you care to share how you connected the two thermocouple boards? Both on the same SPI bus, I assume? What pin did you choose as CS for the second thermocouple?

theoo, 2017-02-13:
Thanks @tkurbad for the details. It seams indeed that it's a matter of calibration, after a bit of tuning and few more prints I can stabilize it... If you say it's for my best I believe you.

In [my config](https://gist.github.com/theoo/1a0f820db45e89af5b02c2b4171e2a09) I set the following pins for _chip select_:

```
temperature_control.hotend.chip_select_pin   0.27
temperature_control.hotend2.chip_select_pin   0.16
```
And yes, both modules are in serial on the same bus.

arthurwolf, 2017-02-13:
So that's confirmation that using multiple thermocouples now works, right ?

theoo, 2017-02-13:
@arthurwolf I can confirm that after more than 6 prints with @tkurbad's firmware, the board didn't crashed.
Some prints crashed because of other factors, @tkurbad suggested that it's related to the PID setup and other printer specific factors.

arthurwolf, 2017-02-13:
Then I think unless @wolfmanjm has objections this is ready to be merged.

Thanks a lot for testing !

tkurbad, 2017-04-04:
Do we need to do more tests? I'd really like to see this merged...

greyltc, 2017-09-19:
This can probably be closed now.

closed 2017-09-19 by wolfmanjm

puzrin, 2017-11-30:
Could anyone summarize, how to attach MAX31865 (PT100) device, but keep sdcard/lcd work?