Smoothieware/Smoothieware #722: Unable to set hwpwm on pin 1.23

theoo, 2015-09-08:
Trying to attach a hobby servo on pin 1.23 I've been unable to generate a PWM signal with the hardware feature on pin 1.23, supposedly hwpwm capable.
The servo was not moving, so I checked with an oscilloscope on pin 1.23.
I can confirm that the same setup, servo or oscilloscope works fine on pin 3.25.

>
SENDING:version
Build version: edge-966961d, Build date: Sep  5 2015 04:05:19, MCU: LPC1769, System Clock: 120MHz

WIP config file: http://pastebin.com/TsgAesF3
M503 command: http://pastebin.com/huKvJ0Pu

I'm trying to replace the Megatronics board from a Spiderbot delta printer (http://spiderbot.eu/en) which uses a servo to switch the head.

Mikk36, 2015-09-08:
The list of pins being used depends on the board at hand.
On the Smoothieboard 4X, for example, this pin isn't connected to Q5, because it doesn't exist.

wolfmanjm, 2015-09-08:
This should work and the code works, so maybe it is because P1.23 is connected to a mosfet is interfering with the sgnal? 3rd large fet on a 5xC. are you on a 5X or a 3X.

Also the config does not seem to define the heater_pin for either of the hotends which is odd.

wolfmanjm, 2015-09-08:
http://smoothieware.org/lpc1769-pin-usage seems pretty clear to me which pins are free and which are not. 

P1.23	LED4	Pad6	3rd large fet	spare on 3 and 4 driver boards, h/w PWM capable

theoo, 2015-09-10:
Thanks for the help guys, I've been spending too much time searching so I created this page
http://smoothieware.org/pwm-capable
with enough keywords so the next ones won't bother with invalid issues.

Thanks again for the great work

closed 2015-09-10 by theoo