Smoothieware/Smoothieware #729: G54: Coordinate System Select

dancole, 2015-09-12:
Please consider implementing Coordinate System Select Commands. The G92 (Set Position) command works for setting a single WCS (Work Coordinate System), but it would be useful to have more than one point.

My understanding is that we switch to each WCS, then use G92 to move the origin for that WCS. Later, in a different script, we can run the same gcode for each WCS. 

G53 - Move in Machine Coordinates
G54 to G59 - Select Standard Coordinate Systems 1 through 6
G.59.x - Select Extended Coordinate Systems 7 and up.

References:
http://reprap.org/wiki/G-code#G54..59:_Coordinate_System_Select_.28CNC_specific.29
http://linuxcnc.org/docs/html/gcode/gcode.html#sec:G54-G59_3
http://linuxcnc.org/docs/html/gcode/coordinates.html#cha:coordinate-system

wolfmanjm, 2015-09-12:
Should be possible. Contributions to the code are welcome ;)

arthurwolf, 2015-09-12:
I'd love to implement this but can't at the moment. Anybody willing to give
it a go, I'll help as much as possible.

wolfmanjm, 2015-12-13:
FYI I am working on this. It is challenging to get it done correctly and work with Smoothie. Especially the G53 is very hard.
In linuxcnc the WCS are saved whenever changed, I propose for smoothie to only save them if you do M500, is this OK?
This will also affect a lot of the leveling strategies, now they will have to work in machine coordinate space and not the requested (WCS) space. Not sure how much this will break existing ABL if any G5x are in effect.
I also changed G92 to use an offset rather than reset axis as per Linuxcnc.
Here is the WIP https://github.com/wolfmanjm/Smoothie/tree/feature/wcs-and-g92-offsets

wolfmanjm, 2015-12-14:
This has been implemented by PR #804

closed 2015-12-14 by wolfmanjm

wolfmanjm, 2015-12-16:
@dancole this is now in edge, can you please test it and make sure it works the way you expect.