Smoothieware/Smoothieware #733: G92 broken on delta

OskarLinde, 2015-09-13:
It always seemed wrong to me that G92 changed actuator positions rather than just a adding soft coordinate system transform. A simple test proves that it is indeed wrong:

```
G28
G91 ; Relative coordinates just makes the example more clear
G Z-50 Y-50
G92
G Y50 ; Notice something weird?
G Z50 ; Oops, not back at starting location!
```

The fix is rather straight-forward but requires some refactoring. Might as well implement #729 at the same time.

OskarLinde, 2015-09-13:
The cleanest solution is to split Robot into two separate classes,

* Machine
* Robot

**Machine** operates in cartesian machine coordinates that have perfect correspondence to actuator coordinates. For simplicity, lets define machine coordinate (0,0,0) as the home position. When Machine homes, the actuator positions are reset. This is the only place in the code that actuator positions are modified. The logic for interacting with the Planner is moved here, but no G-code processing is done at this level as well as path segmentation.

**Robot**, on top of Machine handles and interprets G-Codes and keeps track of all virtual coordinate systems (G92, G54, compensationTransform, etc) including home_offset, homing_position, etc.

G53 would be used to move to raw machine coordinates ignoring any offsets or coordinate system transformations.

This would also simplify implementation of G68/69 - coordinate system rotation.

arthurwolf, 2015-09-13:
I'd see Machine more as a separate class that Robot owns, and that holds
current coordinates, coordinate offsets, and the method to read/modify all
that.
No need to make a new module.

arthurwolf, 2015-09-13:
@OskarLinde By the way, you really should stick to the delta calibration stuff for now.
If G92 is broken, make a note, leave it there, and come back to fix it in another separate PR.
Same thing for changing the inverse compensation stuff.

The rule is, a PR should contain one thing, and one thing only, and stay as small as possible.
While this is true for PRs, it's also true even more generally : You want to be working on one thing at a time, it's not only for you, but also so the rest of the community that makes comments/helps can stay focused and not loose track of what's going on.

I see a lot of great ideas for improvements here, but both the code and the discution should ideally, for now, stay on just that delta calibration stuff. If other stuff is broken, you'll want to come back after that's done, and fix it.
I know it's not as optimal as fixing everything at once, but doing things all together means we sometimes have to give some optimisation, for the sake of working together.

Awesome work overall, and thanks a lot for all you've done so far. I hope you understand what I'm saying, I'm not good at explaining this sort of stuff :)

wolfmanjm, 2015-09-13:
I documented a while back that G92 is NOT to be used on deltas this is a delta specific issue, just do not use G92 on a delta. This is true in Marlin and most other firmwares.
G92 Zxx works if X0 Y0 otherwise do not use it.

wolfmanjm, 2015-09-13:
However I am not totally against fixing G92 to work on deltas, I think it requires using forward kinematics.
However I am not in favor of the suggested refactor. That is way overkill for this.
Instead we just need to keep a coupe of transformation matrices to go from Cartesian to Work coordinates. also one to go from Work coordinates to Compensated coordinates.

The issue with fixing G92 for deltas is that would require Robot to be aware of deltas which breaks the current abstraction. So somehow that fix needs to be handled by arm solution, which I do not see an easy/clean way of doing ATM.

OskarLinde, 2015-09-19:
Fixing G92 doesn't require forward kinematics or anything delta-specific in Robot. All you need to do is change G92 to not use reset_axis, but instead use an offset transform. Like so:

Work Coordinates --[offset]--> Absolute Coordinates --[compensationTransform]--> Machine Coordinates --[ArmSolution]--> Actuator positions

To support G54...G59 you just switch between different offset transforms. To properly support it you also want to support rotations, so the more generic approach would be to implement offset as a rigid transformation (rotation matrix + translation).

The added advantage of this is that you can then rely on actuator positions always staying the same with G28 being the only exception which feels like the right way to do it.

OskarLinde, 2015-09-20:
To @arthurwolf's point, I'm not going to issue a PR to address this issue, but I will instead refer to a few diffs to illustrate my simple suggestion above:

5bd9af5 Make G92 reset the coordinate system without using reset_axis()
e64457c Speculative: Make G28 reset any G92 offset
c136c59 Implement G54..59
62132b9 Implement G53 - move absolute

These are not properly tested or analyzed, just quickly hacked together and briefly run (seemingly correctly). Just consider them a basis for discussion. There are definitely things open to discuss, such as how persistent G92 should be, and if the approach seems reasonable some things should be cleaned up and optimized.

wolfmanjm, 2015-12-12:
I am re assessing this stuff based on my recent use of LinuxCNC and GRBL for CNC work.
I agree G92 should not call ```reset_axis_*``` however there is an inconvenience in that the machine zero then would only be settable with homing, and homing switches are optional.

I propose that G92 be implemented as per GRBL/linuxcnc (as a global offset) and also the G54 offsets should be implemented.
.
Homing sets the machine zero, but for convenience I will invent a new G/Mcode that will set machine zero (much like G92 currently does) There is no current g/m code to do this as grbl decided to not allow this, and linuxcnc does have a manual set machine zero but it is not a M/G code.

Additionally I propose that the current compensation transform be applied immediately to any incoming G0/G1 this will avoid having to do the inverse compensation transform. This mirrors how GRBL/LinuxCNC do Bed leveling compensation, they basically modify the Gcode before it is sent to the machine, (See bCNC for an example). so I think the sequence of transforms would be...

Incoming G0/G1 -> compensation transform for Z (and maybe XY) -> G92 offset -> G54 offset -> arm solution -> actuator position.

Does that sound reasonable? Does it avoid the inverse compensation as I think it does?

If so I will implement this based on the speculative work @OskarLinde did above (with modifications).

wolfmanjm, 2015-12-13:
This was a lot more complicated than it at first seemed. and it may indeed require inverse compensation after all :( however If I add that it will not be the same way, as I already suggested I will add a flag to the compensation transform function to select inverse. Turns out the compensation transform needs to be done in machine coordinates so after the other offsets have been applied, so it can't be done first.
This will undoubtedly break various ABL efforts,

wolfmanjm, 2015-12-14:
This will be obsoleted by PR #804

closed 2015-12-14 by wolfmanjm