Smoothieware/Smoothieware #756: LCD Blanking out after print is done

WZ9V, 2015-10-29:
Using either Pronterface (USB) or exernal SD Card, my LCD display (ASMZ Mini) is blanking out after a print and requires a reset to bring it back.  This is on  the following version - 

Build version: edge-ed25d57, Build date: Oct 17 2015 22:04:52, MCU: LPC1768, System Clock: 100MHz

I was not seeing this behaviour on the late September/early October edge versions.

gddeen, 2015-10-29:
Is there a power off m code?
The USB keeps some electronics up?

wolfmanjm, 2015-10-29:
This is a hardware issue, as there is nothing in S/W that would do that. I suggest you check your wiring, if it is too long it could do that.

closed 2015-10-29 by wolfmanjm

WZ9V, 2015-10-29:
Same wiring that NEVER did it before.  I'll roll back to September and see if it persists.

WZ9V, 2015-10-29:
Yup, your right.  Despite it all working correctly yesterday, something weird is happening today.  All I can figure is I must have done something (like changed a wire position) when I swapped in the new Z-probe.