Smoothieware/Smoothieware #757: Feature Request: Support for AD8495

errolt, 2015-10-30:
Hi wolfmanjm,

Please consider adding support for Thermocouples connected via the AD8495 chip.

It is analog with a straight 5mV/C output. The circuit is very simple and easy to retrofit with a baord like this: https://www.adafruit.com/products/1778

Thank you,
Errol

wolfmanjm, 2015-10-30:
Thermocouples are already supported via the SPI interface available here... http://shop.uberclock.com/collections/smoothie/products/spi-thermocouple

I have no plans to add the above interface. I do not have one to test either.
If you want this you will need to add it yourself or get someone to add it for you.

errolt, 2015-10-30:
Last time I heard there was issues with multiple spi-thermocoules on the same bus, add to that mix an spi lcd and it adds even more fun. :)

Anyway, no worries, I'll look at adding it.

closed 2015-10-30 by errolt

arthurwolf, 2015-10-30:
@errolt If you need any help implementing this, don't hesitate to ask

wolfmanjm, 2015-10-30:
@errolt I will fix the SPI issues for the multiple thermocouple issue.
Note that when you add the support for the analog version, you will need to use a spare ADC port, you will not be able to use one of the thermistor ports. I think there maybe at least one spare ADC port you can use. The reason you need to use a raw ADC port is that the thermistor ports have external components designed to read a resistance not a voltage, so reading a voltage is made hard because there is a pullup resistor to AVCC which usually overpowers the amplifiers in these analog circuits.

errolt, 2015-10-30:
This amp claims to have a 7ma output drive, which should be fine driving the 4k7 resistor, which will draw 0.7ma. It also actively measures the amp's output and compensates for any "losses".

If the resistor do give me problems then I will just remove it...

errolt, 2015-11-06:
I have added it to my branch. You where right, it was very easy to add. Also works fine on the normal thermistor inputs.

arthurwolf, 2015-11-07:
Hey. Can you point me at your code in your branch ?

Cheers.

errolt, 2015-11-07:
Hi Arthur,

I don't know if I followed your coding standards.

Code: https://github.com/errolt/Smoothieware/commit/9cee995ddbe7c073958de722d49dc9fd3d052872

Thank you,
Errol

wolfmanjm, 2015-11-07:
looks good to me... if you want to do a pill request feel free to.

arthurwolf, 2015-11-07:
Yep looks very good, please do a pull request, and ping me so I can update
the documentation.

errolt, 2015-11-08:
Will do, as soon as I can figure out how to do a pull request on a single commit. Looks like a few commandline commands. Will try as soon as I can.

errolt, 2015-11-10:
https://github.com/Smoothieware/Smoothieware/pull/765