Smoothieware/Smoothieware #760: M104 support for T parameter

mdebski, 2015-11-05:
Hi,

what do you guys think about supporting Tn (toolhead number) parameter to M104/M109 (set extruder temp/set and wait) ? It seems, despite [0] description, that it is supported on may reprap printers. Also most known opensource slicers - slic3r and cura - generate gcode like that, so it became de-facto standard.

Currently gcode from slic3r like
T0
G1 ......
M104 S220 T1
G1 .....
causes unwanted toolhead change (as is interpreted as M104 ; T1) on temperature change (ex. after first layer) and breaks stuff.

Or is there any way in smoothie to change inactive toolhead's temperature?

If it receives positive feedback (and probably some implementation hints) I'm willing to work on this.
For reference, I guess it should be implemented somewhere around [1].

[0] http://reprap.org/wiki/G-code#M104:_Set_Extruder_Temperature
[1] https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/TemperatureControl.cpp#L297

arthurwolf, 2015-11-05:
Hi there.

As far as I know, this is currently supported. Maybe update your firmware
to the latest version ?

mdebski, 2015-11-05:
I can't see it anywhere in the mentioned [0] file. Any hints of where to look for it?

I'm using smoothie fork by my printer's producer (zmorph), which indeed is behind master, but that means I can't easily update :(

arthurwolf, 2015-11-05:
Toolmanager handles T0/T1 etc ... and temperaturecontrol checks if it's
currently active ( and therefore should listen to the gcode ) here :
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/TemperatureControl.cpp#L303

Yeah ZMorph is behind master, and *way* behind edge. I'm not sure why you'd
want to use their firmware instead of edge, what are the advantages ?

wolfmanjm, 2015-11-05:
This is already implemented and tool change works as a command Tn not as a parameter, however if Tn is used as a parameter it will change the toolhead. That is the way it is supposed to work as Tn is a modal command.

closed 2015-11-05 by wolfmanjm

mdebski, 2015-11-05:
I know what Tn as a command does, and obviously see it's implemented and working.

What I'm talking about is using Tn as parameter, argument to M104 or M109. This is what many other firmwares support, and what some slicers tend to use meaning "set nth extruder temperature, regardless of currently selected" - like in:
T0
...
; print some stuff using zeroth extruder
M104 T0 S190
M104 T1 S190
...
; print some more stuff, still using zeroth extruder.

And this does not work in Smoothie. This evokes two questions:
1. What is the recommended way of expressing "set nth extr temp" in Smoothie? 
2. What slicer do you use for multiextruder prints so that it does not use this syntax? I don't know of any such.

@arthurwolf I don't know of any advantages of zmorph's fork, I see they mostly add stuff to panel's menu :P. I'm just don't want to risk getting into compatibility issues, but probably it's not very rational fear.

wolfmanjm, 2015-11-05:
T is not a parameter it is a command word and is defined as such in the gcode standard. it is also modal so once set it stays set until otherwise.
It cannot be changed to a parameter in the current implementation.
I think you are incorrect about how other firmwares implement this.
AFAIK Tn anywhere on the line will set the tool to n until it is reset.

wolfmanjm, 2015-11-05:
If you want to use zmorphs fork then submit a request to them to add it to their fork.

mdebski, 2015-11-05:
What do you mean by gcode standard? 
Btw, [0] list firmwares which support T as parameter to M109 as at least Duet, Sprinter, MakerBot. As I said before, at least slic3r and Cura generates gcode as cited above.

I don't necessarily want to use zmorph's fork, and I believe it's better to fix stuff upstream.

[0] http://reprap.org/wiki/G-code#M109:_Set_Extruder_Temperature_and_Wait

wolfmanjm, 2015-11-05:
There is a NIST standard for GCode.

M109 T1 is supported and works and has worked for a year.
It does the same as 
T1
M109

What you are not seeing is whether the Tn is modal, which it is defined to be.
I think you are misinterpreting what is supposed to happen when you specify a Tn.

Tn anywhere on the line will select that tool for that line and any subsequent line, this is how it is supposed to work for reprap printers.

However technically speaking it is incorrect as the NIST standard says M6 does the tool change and Tn just selects the tool, however Reprap has decided to not follow the standard in this area.