Smoothieware/Smoothieware #771: No extrusion in G2/G3

mdebski, 2015-11-16:
It seems that "E" parameter to G2/G3 (arc moves) is ignored - no extrusion happens during move.
For example:
G92 E0
G2 X-100 Y-100 I20 J20 E5
generates no extruder motor movement at all.
It seems that Extruder.cpp module doesn't care about G2/3 at all:
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/extruder/Extruder.cpp#L547

Is this intended?

arthurwolf, 2015-11-16:
Hey.

No, actually Extruder was originally written to work with G2/G3, and
*confirmed* to work with them. It'd actually be a lot simpler if it didn't
work with G2/G3.

But as no slicer out ther actually generates g2/g3 that I know of, nobody
ever uses that, and it looks like that has been broken at some point and
nobody did notice.

What slicer are you using ?

Cheers.

mdebski, 2015-11-16:
It's hand-generated gcode for printing empty sphere with fancy support.

wolfmanjm, 2015-11-16:
G2/G3 are not supported by the extruder, only G0/G1...

https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/extruder/Extruder.cpp#L547

Given all the fairly recent changes I don't know what would happen if this were simply extended to include G2/G3, it may or may not work, you can try it.

mdebski, 2015-11-18:
So I added 2,3 to that "if" and seems that printing arcs works correctly.
See https://github.com/Smoothieware/Smoothieware/pull/772

mdebski, 2015-11-18:
Oh, fine, the bug was introduced by moving the change to my fork ("just one line, no need to git format-path..."). So the compiled and tested firmware was ok.

closed 2015-11-18 by wolfmanjm

Wolfs-gang, 2018-03-21:
Hi guys, sorry to reactivate this issue. I have 3d printer with smootieware and want to print an arc. The g-code is hand-generated. The problem is, that the extrusion is still not working. In my case the nozzle is doing the arc movement (both G2 and G3) but no extrusion during the path. The extrusion starts after finishing the arc. 
I'm a newby and not sure how to solve this problem. Does someone has an idea? 
Thanks

wolfmanjm, 2018-03-21:
Sorry this is quite involved to fix and due to the fact no slicers currently generate G2/G3 we do not support it, and have no schedule to add this anytime soon.

commit 01628f97 (2015-11-21): Merge pull request #772 from mdebski/fix-arcs-extrude

Pay attention to G2/G3 in extruder module. Fix #771.
commit 210b9843 (2015-11-21): Pay attention to G2/G3 in extruder module. Fix #771.