Smoothieware/Smoothieware #774: [ZGrid]After Z grid leveling, will not go Z=0

evarty, 2015-11-22:
Machine: Cartesian, XYZ gantry, stationary bed
Firmware: Nov 21 edge build, and some edge build from Spring 2015
Probe: mill bit electrically connected to pin, copper clad board connected to other pin, of Zmin port on board. 
Host: Pronterface on Debian Jessie

Description of Problem: After performing ZGrid compensation, probe will not reach Z=0 with G0 Z0. Specifics in attached Pronterface log, which is commented.

This happens under an old firmware edge build, gives heights that it will not move past of around .5mm. Under the Nov 21 edge build, it has dropped to about .25mm. 

If I define z height based on G30 prior to ZGrid bed leveling, then send G0 Z0, probe goes to bed as expected.

[ZGridBugLog.txt](https://github.com/Smoothieware/Smoothieware/files/40973/ZGridBugLog.txt)

qharley, 2015-11-22:
I will look into it today.  Would you mind posting a copy of your config, config-override and ZGrid files?

evarty, 2015-11-22:
My config is this (its not normally a .txt, but github wont accept it as it normally is):
[config.txt](https://github.com/Smoothieware/Smoothieware/files/41083/config.txt)

I have never had a Zgrid o config-override before, so I think I saved them properly. (Problem persists after saving and reloading file) Note that the Zgrid file is not actually a text file, I just appended a .txt to it. It appears to error midway through. Vim, emacs, and gedit all seem to give errors for a chunk of the text.

[config-override.txt](https://github.com/Smoothieware/Smoothieware/files/41084/config-override.txt)
[Zgrid.txt](https://github.com/Smoothieware/Smoothieware/files/41085/Zgrid.txt)

evarty, 2015-11-22:
Side point, this also happens with the 3 point leveling, which used to work for me, so it could easily be user error as well. I will closely recheck my procedures.

evarty, 2015-11-24:
So, upon further review, it was in fact user error. I was setting my Z height to 0 at a point where the compensation was not zero, so the compensation was taking over what I thought was supposed to happening. Sorry to waste your time.

closed 2015-11-24 by evarty