Smoothieware/Smoothieware #809: temp module event no trigger unless threshold is crossed

dposea, 2015-12-20:
if you define a switch.fan , but the hotend is above the trigger termp when the printer is powered up, the event never get triggered and the fan stays off.

wolfmanjm, 2015-12-20:
Are you using the very latest edge?

wolfmanjm, 2015-12-20:
Please provide your config.

wolfmanjm, 2015-12-20:
I cannot reproduce,I think this was fixed several weeks ago.
So either you are not using the latest edge, or your config is not setup properly.

closed 2015-12-20 by wolfmanjm

dposea, 2015-12-21:
I'm using the version suggested by the docs on turning of the microSD card.

https://github.com/Smoothieware/Smoothieware/blob/edge/FirmwareBin/firmware-disablemsd.bin?raw=true. I noticed there are menu changes too.

dposea, 2015-12-21:
I just noticed the no msd version is also in firmware.bin. I'll test on that version just to be sure, but will not post again unless there is stil a problem.

wolfmanjm, 2015-12-21:
use the posted firmware.bin or build the latest version. this is NOT a bug I tested it.