Smoothieware/Smoothieware #817: Low movement speed if redundand F speed setting in gcode

evgs, 2016-01-03:
I have a lot of G1 Xxxx Yxxx Fconst gcode sequences produced by Slic3r (sample below).
(This movements was generated with option "Avoid crossing perimeters")
Although desired moving path is very fine without XY jerks, this gcode prevents Smoothieware to do normal acceleration, so extruder moves very slowly.

Yes, I know this numerous F12000.000 is redundand, but I think Smoothieware should handle this gcode as expected 

```
...
G1 X-9.705 Y-11.903 F12000.000
G1 X-9.705 Y-11.903 F12000.000
G1 X-9.686 Y-11.880 F12000.000
G1 X-9.607 Y-11.787 F12000.000
G1 X-9.601 Y-11.781 F12000.000
G1 X-9.600 Y-11.780 F12000.000
G1 X-9.514 Y-11.689 F12000.000
G1 X-9.440 Y-11.612 F12000.000
G1 X-9.396 Y-11.564 F12000.000
G1 X-9.362 Y-11.529 F12000.000
G1 X-9.339 Y-11.506 F12000.000
G1 X-9.251 Y-11.420 F12000.000
...
```

arthurwolf, 2016-01-03:
If XY movement is very small, you'd expect it not to be able to do normal
acceleration with or without the extra F parameters, simply because it
can't look ahead far enough in it's planning queue.

The extra F parameters are wasteful, but shouldn't disturb Smoothie at all.

closed 2016-01-03 by wolfmanjm

wolfmanjm, 2016-01-03:
as Arthur says this has nothing to do with using Fxxx on everyline that is totally correct and accepted.
The issue is you need to understand how acceleration works, this may help...

http://prusaprinters.org/calculator/#speed

evgs, 2016-01-10:
Okay, I missed lookahead depth factor in my considerations.