Smoothieware/Smoothieware #818: Jogging after print aborted from Smoothie Web causes erroneous movment

W3DRK, 2016-01-05:
I've noticed that from within Smoothie Web if I abort a print and then use the jog controls to lift the Z axis or move X/Y, the first move after the abort will cause a random and often violent move of X or Y.

For example: During a print in progress, if I click the Abort button in Smoothie Web, once the printer stops if I lift click the jog control to lift the Z axis, X or Y will move as well, and usually a LOT faster than Z.  Many times the random move results in X or Y slamming into their mechanical limits, grinding stepper sounds, etc...potentially damaging behavior.

This only happens during the first move after using Smoothie Web to abort a print.  And the the jog commands can come from Smoothie Web, USB, wherever..the result is the same.  

Any moves after that work normally, assuming the erroneous move didn't hit the limit switch which requires an M999 or reset to recover from.

There is at least one other individual who has reported the same behavior in the Smoothieware forum, so it does not appear to be isolated to just me.

I have also noticed that many times, uploading a file in Smoothie Web makes the Z Axis move.  It occurs as soon as I click the upload button...the Z Axis will move up 10mm or so.

Build version: edge-10ee205, Build date: Jan  3 2016 04:05:28

wolfmanjm, 2016-01-05:
delta or cartesian? printing from sdcard? auto levelling enabled?

Mikk36, 2016-01-05:
Are the motors disabled after the abort by any chance?
If so, you should see the exact same behaviour without even aborting.

W3DRK, 2016-01-05:
Excellent questions, I should have provided these from the start.

It's a dual Z cartesian (Ord Bot Hadron) and I'm printing from the on-board SD card.  No auto bed leveling enabled.  I'm attaching my config below if it helps.

I believe it leaves the motors enabled after I abort it but I will double check tonight.  But it only happens on the very first move after aborting.  Any subsequent moves after that work normally, assuming it didn't slam one of the axis' into a limit switch and halt all movement.

> cat sd/config.txt
# NOTE Lines must not exceed 132 characters
# Robot module configurations : general handling of movement G-codes and slicing into moves
default_feed_rate                            4000             # Default rate ( mm/minute ) for G1/G2/G3 moves
default_seek_rate                            4000             # Default rate ( mm/minute ) for G0 moves
mm_per_arc_segment                           0.5              # Arcs are cut into segments ( lines ), this is the length for
                                                              # these segments.  Smaller values mean more resolution,
                                                              # higher values mean faster computation
#mm_per_line_segment                          5               # Lines can be cut into segments ( not usefull with cartesian
                                                              # coordinates robots ).

# Arm solution configuration : Cartesian robot. Translates mm positions into stepper positions

# 1/32 microsteps
alpha_steps_per_mm                           161.1047            # Steps per mm for alpha stepper (X)
beta_steps_per_mm                            161.1047            # Steps per mm for beta stepper  (Y)
gamma_steps_per_mm                           811.2               # Steps per mm for gamma stepper (Z)

# Planner module configuration : Look-ahead and acceleration configuration
planner_queue_size                           32               # DO NOT CHANGE THIS UNLESS YOU KNOW EXACTLY WHAT YOU ARE DOING
acceleration                                 3000             # Acceleration in mm/second/second.
#z_acceleration                              500              # Acceleration for Z only moves in mm/s^2, 0 uses acceleration which is the default. DO NOT SET ON A DELTA
acceleration_ticks_per_second                1000             # Number of times per second the speed is updated
junction_deviation                           0.02             # Similar to the old "max_jerk", in millimeters,
                                                              # see https://github.com/grbl/grbl/blob/master/planner.c
                                                              # and https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.8
                                                              # Lower values mean being more careful, higher values means being
                                                              # faster and have more jerk
#z_junction_deviation                        0.0              # for Z only moves, -1 uses junction_deviation, zero disables junction_deviation on z moves DO NOT SET ON A DELTA
#minimum_planner_speed                       0.0              # sets the minimum planner speed in mm/sec

# Stepper module configuration
microseconds_per_step_pulse                  1                # Duration of step pulses to stepper drivers, in microseconds
base_stepping_frequency                      100000           # Base frequency for stepping

# Cartesian axis speed limits
x_axis_max_speed                             18000            # mm/min
y_axis_max_speed                             18000            # mm/min
z_axis_max_speed                             600              # mm/min

# Stepper module pins ( ports, and pin numbers, appending "!" to the number will invert a pin )
alpha_step_pin                               2.0              # Pin for alpha stepper step signal
alpha_dir_pin                                0.5!             # Pin for alpha stepper direction
alpha_en_pin                                 0.4              # Pin for alpha enable pin
alpha_current                                1.2              # X stepper motor current
alpha_max_rate                               30000.0          # mm/min

beta_step_pin                                2.1              # Pin for beta stepper step signal
beta_dir_pin                                 0.11             # Pin for beta stepper direction
beta_en_pin                                  0.10             # Pin for beta enable
beta_current                                 1.2              # Y stepper motor current
beta_max_rate                                30000.0          # mm/min

gamma_step_pin                               2.2              # Pin for gamma stepper step signal
gamma_dir_pin                                0.20!             # Pin for gamma stepper direction
gamma_en_pin                                 0.19             # Pin for gamma enable
gamma_current                                1.2              # Z stepper motor current
epsilon_current                              1.2              # Second Z stepper motor current
gamma_max_rate                               300.0            # mm/min

# Serial communications configuration ( baud rate default to 9600 if undefined )
uart0.baud_rate                              115200           # Baud rate for the default hardware serial port
[... cut ...]
# uncomment the 3 below to manually setup ip address
network.ip_address                           192.168.1.10     # the IP address
network.ip_mask                              255.255.255.0    # the ip mask
network.ip_gateway                           192.168.1.1      # the gateway address
network.hostname                                                         smoothie         # optionally set this hostname for dhcp
#network.mac_override                        xx.xx.xx.xx.xx.xx  # override the mac address, only do this if you have a conflict

wolfmanjm, 2016-01-05:
FYI the abort command never guarantees the position is correct, and homing is always recommended after you use the abort command, as it is really hard to recover the actual position of the actuators after you force the queued commands to abort. It is a lot safer to pause and let the queue drain.

wolfmanjm, 2016-01-05:
What board are you using? as smoothieboards do not have 1/32 and the config is not an azteeg X5.

W3DRK, 2016-01-05:
I wanted 1/32 micro stepping so I went with an SBASE.  I also donated to Smoothieware.  In any event I can't imagine it being a hardware issue...

There is a problem with homing after an abort, at least in my case.  If there's a half-printed object stuck to the glass, obviously bad things are going to happen if I try to home the printer.  

Generally my workflow is if I need to abort a print for whatever reason, I click abort, then click to lift Z 10mm-20mm to lift the hot nozzle off the print, then shut off the motors so I can slide the Y axis out and wait for the print to cool so it can be removed from the glass.  I can't manually lift Z since it's on lead screws.

Instead of aborting, you suggest pausing?  I don't recall seeing a pause option in Smoothie Web but I may just be overlooking that.  It doesn't seem to only be a loss of positioning issue, the speed in which these random movements occur can be EXTREMELY fast and violent.  Further, why would instructing Z to move 10mm cause X or Y to go slam into its mechanical limits?  No mater were Smoothie thinks the axis are, I can't imagine 2 axis moving simultaneously when only one was instructed to do so.

Another thing to note, if I tell Z to lift only .1mm, the corresponding X or Y move will likely not be far enough to hit anything.  But if I lift Z 1mm or 10mm, that's when things get scary.  It almost seems as if those moves are a multiple of the Z move, or somehow getting the higher steps/mm needed for the Z lead screws.

wolfmanjm, 2016-01-05:
I have confirmed that abort is not calculating the correct current position, so I suggest using either abort 1 command or M25 which will pause the print once the queue is empty.

I'm not sure how long it will take to fix this issue. But I'm lookin ginto it.

Thanks

wolfmanjm, 2016-01-05:
ok seems I fixed this a long time ago, then unfixed it recently because I forgot why I fixed it originally :)

wolfmanjm, 2016-01-05:
@W3DRK please test this when you get a chance and confirm it is fixed. I still recommend homing after an abort though.

closed 2016-01-05 by wolfmanjm

W3DRK, 2016-01-05:
Great!  So edge-fc8cfc6 should have the fix reapplied?  I will try and report back shortly...

wolfmanjm, 2016-01-05:
No you will need to build latest edge cbbd33b1e3eab0e19f4ad20f63ad8d090deb68c4
the firmware.bin is a few weeks old

W3DRK, 2016-01-05:
> No you will need to build latest edge cbbd33b
> the firmware.bin is a few weeks old

Yep I figured that out a few seconds after I asked. Thanks!

jjcolletta, 2016-01-06:
W3DRK - Thanks for submitting this and Wolf thanks for the fix (re-fix) fix.  I thought I was loosing my mind.  I will use the pause command and not the abort button on the web or from the GLCD for awhile.

W3DRK, 2016-01-06:
Cool, I'm testing cbbd33b now and will report back in a few minutes as to whether or not it takes care of it.

W3DRK, 2016-01-06:
OK the problem appears to be gone!