Smoothieware/Smoothieware #834: false endstops trigger while moving: debounced_get(n)

PaoloBi, 2016-01-30:
Hi, I sometimes get an "endstop triggered error" while moving (and obviously NOT triggering endstops).

I checked endstops.cpp, Endstops::on_idle:

........
    for( int c = X_AXIS; c <= Z_AXIS; c++ ) {
        if(this->limit_enable[c] && STEPPER[c]->is_moving()) {
            std::array<int, 2> minmax{{0, 3}};
            // check min and max endstops
            for (int i : minmax) {
                int n= c+i;
                if(debounced_get(n)) {
                    // endstop triggered
                    ...
                    this->status= LIMIT_TRIGGERED;
                    // disables heaters and motors, ignores incoming Gcode and flushes block queue
                    THEKERNEL->call_event(ON_HALT, nullptr);
                    return;
                }
            }
        }
    }

debounced_get(n) is a simple loop, and returns true if pin X[Y,Z] is ON endstop_debounce_count times.

In my opinion this is a) a waste of time b) not correct. In my experience a correct debounce time is between 1 and 10ms; i guess debounced_get(n) takes less than 1ms to complete.
Wouldn't be better to change debounced_get and call it periodically (every 100us...1ms) ?

Thanks

wolfmanjm, 2016-01-30:
your problem is caused by noise on the enstop line, and debounce is not designed to filter out noise, although it can help in some cases. I suggest you change your endstops to normally closed to ground, this helps with noise, also make sure the endstop cables are not runnign next to motor or power cables.

Your analysis of the code is incorrect. there is a call to ON_IDLE in between each loop (this keeps smoothie alive) and can take anywhere from 1ms to several ms sometimes even more, so in reality the debounce time is around 100ms with default settings.

closed 2016-01-30 by wolfmanjm

triffid, 2016-01-30:
The debounce  works great for everyone else, if your printer's wiring is
poor then no amount of software fixes will help.

Keep in mind that smoothieboard has an RC noise filter on the endstop
inputs which is typically adequate to remove most induced noise.

If you haven't twisted your motor cables and/or run your endstop cables
close to the motor cables, neither the RC filter nor software debounce will
help as the induced noise will simply be too strong - the motor cables
carry medium currents with high frequency and sharp edges, which is almost
impossible to filter out if it's allowed to propagate.

Twisting the pairs greatly reduces emitted noise as the magnetic and
electric fields from each wire cancel out from even a short distance away -
this is why Ethernet uses twisted pairs.

Twisted pairs aren't just for aesthetics, they have an important electrical
purpose too!

In short, this is most likely a hardware problem, not a firmware issue.

PaoloBi, 2016-01-30:
Are you sure ? ;-)

1. My endstops are all NC
2. Take a look to Endstops::on_idle (old code...new code introduced 
debounced_get(n), but behaviour is the same)

*uint8_t debounce= 0*; // THIS SHOULD BE STATIC AND INCREMENTED EVERY 
ON_IDLE RUN....
*while(this->pins[n].get())* {
*if ( ++debounce >= debounce_count )* {
                         // endstop triggered
                         THEKERNEL->streams->printf("Limit switch %s was 
hit - reset or M999 required\n", endstop_names[n]);
                         this->status= LIMIT_TRIGGERED;
                         // disables heaters and motors, ignores 
incoming Gcode and flushes block queue
                         THEKERNEL->call_event(ON_HALT, nullptr);
                         return;
                     }
                 }

This is called within a single on_idle run, and actually does not 
debounce anything....

in effect, I get "endstop hit error" very rarely, but this does not mean 
the bug isn't there !

Regards,
Paolo

wolfmanjm, 2016-01-30:
Yes we are sure it is noise, if you get it triggered when it is not hit then it has absolutely nothing to do with debounce. the debounce is only needed for the homing cycle not the limit switches. I am not even sure why there is even a debounce in there for the limit switches. If a limit switch triggers it should stop immediately there is no need to debounce a limit switch. If your limit switch gets triggered before it is even touched then it is a H/W issue, probably noise. How on earth could that be S/W?

PaoloBi, 2016-01-30:
Hi,
I am sure it is noise, too ! The endstop isn't "really" triggered. 
Perhaps word "debounce" is measleading here.
The "debounce" code I use to write normally checks if a "condition" is 
stable for X time before accepting it.
This is the rule in industrial control.
In my setup probably a noise (a short spike from motors ?) gets captured 
from endstop input, but it's quite easy to filter it out if "debounce" 
code is well written.
To check if a digital input has a valid level for 2-300us, in my 
opinion, is not robust programming.
Regards

wolfmanjm, 2016-01-30:
I have already said the debounce_get() is irrelevant and incorrect. I agree there. But first it is not supposed to mitigate noise, and second debounce is not needed for limit switches, only homing switches (which is done correctly). If you want to submit a PR to add noise mitigation for limit switches feel free to do so. Try not to add any more RAM usage if you do so. (That is the main reason the code is the way it is, adding 6 shorts to count the debounce for each limit switch is not a preferred option when the noise can be mitigated in H/W with correct wiring. I suspect as it is for noise you could use a single count for when any limit switch is triggered.

PaoloBi, 2016-01-31:
You are right, I added just a global debounce counter. Tomorrow will check with oscilloscope what kind of noise is detected by endstop input (almost certainly X-motor, his cables must run together with min and max X endstop wires).
Anyway, I suggest to improve digital input handling. Right now, a short spike on X endstop input can waste a 12 hours print....