Smoothieware/Smoothieware #841: Enhancement: plasma Pierce and Arc OK + delay

openhardwarecoza, 2016-02-08:
Hi guys.  I have my Smoothieboard based plasma Cnc up and running,  but I could use an enhancement:   a plasma mode to M3:

In a plasma the start of each cut goes like this:
IHS phase: initial height sense:   uses a z endstop to touch off on the material to be cut,  homing z between moves (steel warps all the while while cutting) 

After touching off you do a G0 move to lift off a knowm fistance (from you machines manual)  called the Pierce Height.  

At the pierce height you send M3 to enable the tool (plasma) 

After this comes the difference from other machines.  Any other machine can then start moving. But with plasma we have an external input to wait for: Arc OK

I tried dwell delays but its not predictable enough.  I sometimes the arc takes 0.1sec to establish,  if the material is rusty in a spot it can take 1 sec.  And that changes along the position of the material.  Pausing with arc running in pierce height for 1 second while it already started 0.8s ago,  eats your consumables (tips) 

The plasmas and torch height controls have an output I can tie to Smoothie,  which we can wait for.  

So basically:

G28 Z0 ; touch off
G0 Z2.8 ; pierce height
M3 D2 ; start plasma,  max delay 2 second (if no arc ok in 2 second,  pause and wait for uset interaction (torch knocled off,  not on plate,  etc) 
Pause further execution until we get a signal on Arc OK pin
G4 P300;  from arc ok,  to allow torch to pierce through -  longer for thicker materials. Arc ok signals start if successful arc not 'pierced'  - so delay to let torch melt through 
G1 Z0.8 ; move to cut height
G1 X... Y....  ; toolpath

Can we add a plasma module that handles the arc ok / pause after m3 until arc ok happens (if yes, continue,  if no after delay,  pause and turn off)

wolfmanjm, 2016-02-08:
yes you can add a module to handle this case. However you will have to do that, as none of the developers have an arc cutter to play with :)

Maybe a more generic option would be to add a switch or a custom delay that delays until a specific switch goes true. This could be a module or built into switch. It would be similar to the temperature controls wait for a certain temp to be reached. Or the temperature switch control which does the opposite it monitors the temp and turns a pin on when the temp is reached. I think this would be easier than writing a plasma cutter module.
Basically you would insert a custom gcode similar to G4 Pxxx but one that waits on a specific pin to go high (or low).

openhardwarecoza, 2016-02-08:
I like the G4 with a wait for pin! That would be an ideal generic use case!

openhardwarecoza, 2016-02-08:
Could use some pointers on where to begin then I wouldn'ty mind taking a stab at it:

https://github.com/Smoothieware/Smoothieware/blob/be0230e2c3ded025e72f7d6ae661bf213984a46b/src/modules/robot/Robot.cpp#L341 ?

(If has letter D ....)?

wolfmanjm, 2016-02-08:
No, it would need to go in the Switch module, this is not a core function so cannot go in robot.

You would define a switch with a new option, wait_until_on for instance. and you would define the M code it uses (you could use G4.1 for instance), it cannot be G4 though as that is a well defined gcode that does not have that function.

the switch module could be extended to know that the new option wait_until_on means enter an on_idle_loop until the switch goes on or a timeout occurs. It also needs to be interruptable too on an ON_HALT

triffid, 2016-02-08:
In the meantime you could conceivably connect your signal to a thermistor
input and use the existing temperature wait stuff, probably safest to go
via a mosfet and a couple of resistors

qharley, 2016-02-08:
I would also try to use the existing temperature based wait stuff

fpvmustang, 2017-04-09:
wish it had an ok to move output. it takes away control of z axis in thc then returns it when not in use.
http://eagleplasma.com/proma-thc-sd.html#main

fpvmustang, 2017-04-09:
the torchmate at work has a hypertherm thc
the pierce delay is controlled in it.
so the only thing smoothie would have to do is
have a torch on signal M3 or M50 and then wait to move.
so only thing were really missing is a wait to move input.

fpvmustang, 2017-04-09:
arduino torch height control
[https://www.youtube.com/watch?v=Nif3uZHhUF0](url)

fpvmustang, 2017-04-09:
I think the pause button could be used pin 2.12 or 2.11 
use the m3 to start plasma but it also triggers the pause button.
use the thc to release the pause button.

wolfmanjm, 2017-04-09:
I am sorry but this is not something any of the core developers is likely to do. So unless you do it yourself it is not going to be done.

dandumit, 2017-04-09:
I am interested as well about subject.
But I think that it's needed more than a simple timer and a voltage reading.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

arthurwolf, 2017-04-09:
Can you describe precisely what new Gcodes you would want to implement and
exactly how they would work ?

fpvmustang, 2017-04-09:
I'm not sure we need to change anything with smoothie.
 with the currently available proma thc.
basically m3 activates plasma and a relay that activates smoothie pause. 
The arc ok from plasma thc releases smoothie pause.

So in our gcode we touch off go to pierce height.
 m3 plasma on g code pause (could include pierce delay).
 THC reads arc voltage takes away control of z axis releases smoothie pause.
The Z axis external stepper signals go through THC. Allowing THC to drive z axis directly.
at end of cut m5 kills plasma and pause relay. THC senses no arc and releases z axis.
So 
m5 pause ( long enough for THC to release Z and plasma to cool tip)
G0 Z2. (rapid height)
g0 X Y next pierce
Probe Z move to pierce height.

Ideally smoothie would have input for external z up down. From THC
But the torchmate at work the Z is completely ran off hypertherm THC.

want2bet, 2017-08-08:
Sorry to revive and old thread.  I'm building a CNC plasma table.  I'm a bit out of my league here on all aspects (CNC, C++, electronics, etc.) but I'm learning fast and haven't let the smoke out of anything (yet).  My lack of knowledge means I'll probably screw up the design the first 5 times.  I would appreciate any design guidance.
 
1.  Is it feasible for me to design the whole THC control aspect into a Module?  (I can  find plasma cutter integration circuits I could leverage. )  How would I tell the "conveyor" or "robot" not to try to execute any Z axis commands?  Or just ensure that my CAM post processor will not give any Z axis commands?

2.  What would be a "best practice" associated with starting a cut?  Should I create the module to expand M3 to something like M3.1 like this?
    M3.1  D2.1 H2.2 P1.4 C1.2  V104
    Arc Delay Timeout   D <seconds>
    Arc Pierce Height   H <default units (mm)>
    Arc Pierce Time     P <seconds>
    Arc Cut Height      C <default units (mm)>
    Target Cut Voltage  V <volts>

3.  If I do something like the above, then I can probably look at some of the post processing scripts in Fusion360 and others and find and M3 commands modify for plasma.  Is that a reasonable approach?

4.  Is this the best place to ask questions like this?

fpvmustang, 2018-02-23:
I went with a stand alone thc and 24dc lifter. So torch height is all self contained.
  start arc input and ok to move THC output. are my current problems.
I was trying to use switched out on G1 G2 G3 commands off on G0 for arc control
with M600 suspend and switched M601 resume for ok to move
 but bcnc does not support M600.
So thermister input like m190 or m191

fpvmustang, 2018-02-24:
From pc web interface or bcnc m109 activates plasma but from sd card g code is skipped
Also when M109 is read when running from pc it doesn't wait for thermistor 
%
 G20
 G90
 G54
 G00 X0.8973 Y1.4058 Z2.
 Z0.01
 M109 S60
 G01 Z0. F27.0 S0
 M109 S60
 X0.7687 Y1.559 F54.0
 G03 X0.6625 Y1.375 I0.1063 J-0.184
 X0.875 Y1.1625 I0.2125 J0.
 X1.0875 Y1.375 I0. J0.2125
 X0.875 Y1.5875 I-0.2125 J0.
 X0.7687 Y1.559 I0. J-0.2125
 M109 S60
 G01 X0.7722 Y1.5496
 G00 Z2.
 X-0.2345 Y0.184
 Z0.01
 M109 S60
 G01 Z0. F27.0
 M109 S60
 X-0.0375 Y0.2187 F54.0
 M109 S60
 Y3.
 G02 X0. Y3.0375 I0.0375 J0.
 M109 S60
 G01 X2.
 G02 X2.0375 Y3. I0. J-0.0375
 M109 S60
 G01 Y0.
 G02 X2. Y-0.0375 I-0.0375 J0.
 M109 S60
 G01 X0.
 G02 X-0.0375 Y0. I0. J0.0375
 M109 S60
 G01 Y0.2187
 G00 Z2.
 (END TOOL)
 M5
%

fpvmustang, 2018-02-24:
probably a grbl mode issue

wolfmanjm, 2018-02-24:
M109 is not implemented in the CNC builds. (The temperature control is omitted from a default CNC build). you will need to build your own CNC version and include the temperature control. or use the regular 3D build.

fpvmustang, 2018-02-25:
went to edge dec 22 2017  3d build. Same problem but I can see my temp is not being set from sd card.
But it is from web interface. i have 2 resistors panel reads 22c normaly 
and 73c when ok to move signal is sent. 
I'll post my config on google +

fpvmustang, 2018-02-25:
https://pastebin.com/PrJFnZY3
[https://pastebin.com/PrJFnZY3](url)

fpvmustang, 2018-02-25:
Thanks Guys we have a working machine

config example
`# First hotend configuration
temperature_control.hotend.enable            true             # Whether to activate this ( "hotend" ) module at all.
temperature_control.hotend.thermistor_pin    0.21             # Pin for the thermistor to read
temperature_control.hotend.heater_pin        2.5              # Pin that controls the heater, set to nc if a readonly thermistor is being defined
temperature_control.hotend.thermistor        EPCOS100K        # See http://smoothieware.org/temperaturecontrol#toc5
#temperature_control.hotend.beta             4066             # Or set the beta value
temperature_control.hotend.set_m_code        104              # M-code to set the temperature for this module
temperature_control.hotend.set_and_wait_m_code 109            # M-code to set-and-wait for this module
temperature_control.hotend.designator        T                # Designator letter for this module
#temperature_control.hotend.max_temp         300              # Set maximum temperature - Will prevent heating above 300 by default
#temperature_control.hotend.min_temp 0 # Set minimum temperature - Will prevent heating below if set

# Safety control is enabled by default and can be overidden here, the values show the defaults
# See http://smoothieware.org/temperaturecontrol#runaway
temperature_control.hotend.runaway_heating_timeout      0  # How long it can take to heat up, max is 2040 seconds.
temperature_control.hotend.runaway_cooling_timeout      0  # How long it can take to cool down if temp is set lower, max is 2040 seconds
temperature_control.hotend.runaway_range                0   # How far from the set temperature it can wander, max setting is 63°C

# PID configuration 
# See http://smoothieware.org/temperaturecontrol#pid
temperature_control.hotend.p_factor         13.7             # P ( proportional ) factor
temperature_control.hotend.i_factor         0.097            # I ( integral ) factor
temperature_control.hotend.d_factor         24               # D ( derivative ) factor

## plasma fire
switch.misc.enable                           true             #
switch.misc.input_on_command                 M811              # PLASMA FIRE
switch.misc.input_off_command                M812              #  PLASMA OFF RAPID
switch.misc.output_pin                       2.7              #
switch.misc.output_type                      digital          # just an on or off pin
`
post processor adds
M811
M109 S60
before G01
and before
M109 S20
M812
G00

example G code
 G20
 G90
 M109 S20
 M812
 G00 X1.028 Y0.9653 Z2. F100
 M109 S20
 M812
 Z0.11
 M811
 M109 S60
 G01 Z0. F100.0 S18000
 M811
 M109 S60
 X1.225 Y1. F250.0
 G03 X1. Y1.225 I-0.225 J0.
 X0.775 Y1. I0. J-0.225
 X1. Y0.775 I0.225 J0.
 X1.225 Y1. I0. J0.225
 M811
 M109 S60
 G01 X1.2053 Y1.0035
 M109 S20
 M812
 G00 Z2.

wolfmanjm, 2018-04-18:
After 2 years of inaction I am closing this issue.

closed 2018-04-18 by wolfmanjm