Smoothieware/Smoothieware #845: File not saved via USB on Ubuntu

dposea, 2016-02-13:
I'm not sure this should be a new bug, or included with bug #615. It always takes multiple attempts to edit the config file via USB. 

Steps taken:
Plug USB cable into laptop.
When explorer window opens click on file config, opens using Kate. 
Edit the file.
Save the changes.
Click the menu icon to eject the SD card.

No errors are received , and the drive shows as not mounted, but the changes are not saved, yet a *.kate.swp file is saved. 

I need to investigate more to find out if the card is being remounted RO, if the changes ever make it to a cache for the SD card on the smoothieboard side, if the PC cache is being purged on eject. and a few more things. I'll be glad to try to fix some of the USB SD card 
issues unless someone else already has a plan for a better implementation. I'm no USB guru but I do understand a good bit of the operation when dealing with SD Flash cards, including the need for wear leveling that complicates writing new sectors by a factor of about 100, even though the concept is simple.

It took me 7 attempts to make a change to my config file via USB and have that change saved. 

If this is a known issue and belongs with issue #615 please excuse me for opening a new issue for an old problem. If you would like me to dig into this deeper please let me know.

Thanks,

David Posea

arthurwolf, 2016-02-13:
Can you try using a different text editor ?

dposea, 2016-02-13:
Do you mean a different editor besides Kate??  Sure.

wolfmanjm, 2016-02-13:
how are you checking to see if the file is changed? I suspect this is an issue with your host, or user error, as I have never seen this problem.
You really need to reset the smoothie after changing any file via the MSD.

wolfmanjm, 2016-02-13:
FWIW MSD is totally broken, and concurrent access to the file system is not supported and never will be. The plan is to implement MTP and get rid of MSD as android has already done for similar reasons.

closed 2016-02-13 by wolfmanjm

wolfmanjm, 2016-02-13:
Also it is much safer to copy the file locally, edit it, copy it back, unmount the sdcard, then reset smoothie.
This is what I do from Ubuntu, and it never fails.

dposea, 2016-02-13:
I'm unpluging USB, then plugging it back in. I suspect the changes are still only present in the host cache and never make it to the card at all. Any attempt to edit the file without resetting smoothie says the filesystem is RO. I'll try your copy, edit, and write back method. Also, Authur may have a point and Kate is part of the problem because files from various slicers get written ok.
There are plenty of work arounds, but I did want to report the problem at least once. I'm content to continue editing files locally, removing and replacing the card.
Thanks,
David Posea

wolfmanjm, 2016-02-13:
rather than just unmount, try sync first, then unmount, then reset smoothie. See if that works. it may be that for whatever reason your version of unmount is not syncing. I use Ubuntu and it seems to always sync as it can take along time for unmount to return. But I usually do a sync first anyway.

triffid, 2016-02-15:
I use kate/linux and never have any problems like this as long as I wait
for unmount/eject to complete before smoothie reset.

It's possible that your OS doesn't wait for eject to complete before saying
it's ok to unplug; smoothie is fairly slow to write due to a number of
factors including the single-block-at-a-time write code and only using USB2
full speed, so it can take several seconds even if you only change a few
lines in config.

An exacerbating factor for the above issue is that various hosts (esp OSX)
likes to scan the entire disk, which takes ages due to smoothie's rather
limited read/write speed and massively bogs down any write operations, as
well as causing problems during printing. You may have some sort of daemon
on your host that does something like this.

Perhaps you have one of those bad cards that likes to freak out and do
weird stuff occasionally - filesystem remounted readonly is a common
symptom of this, as the host gets read or timeout errors from smoothie. Try
a different SD card, and check dmesg.

Fwiw I don't mount smoothie's MSD unless I'm actually planning to use it
for something, perhaps you could simply turn off auto-mount for that disk?

dposea, 2016-02-22:
I have learned to wait until udisk2 falls off the top list before doing any else with the SD card. I'm going to switch card types and see if that helps. SD card controllers are NOT created equal :)